MichaIng/DietPi #3064: Webmin - Error installing update

Sopor, 2019-08-19:
I was updating my four Webmin today and the only two that gave me error was DietPi. Do you know why?

```
(Reading database ... 54532 files and directories currently installed.)
Preparing to unpack .../395543_3013_2_upgrade.cgi ...
Unpacking webmin (1.930) over (1.920) ...
dpkg: dependency problems prevent configuration of webmin:
 webmin depends on shared-mime-info; however:
  Package shared-mime-info is not installed.

dpkg: error processing package webmin (--install):
 dependency problems - leaving unconfigured
Processing triggers for systemd (232-25+deb9u11) ...
Errors were encountered while processing:
 webmin
```

DietPi v6.25.3

I have sent Jamie an email so i'm waiting for a reply from him.

MichaIng, 2019-08-20:
@Sopor 
> Package shared-mime-info is not installed.

```
apt install shared-mime-info
```

Btw the easiest method, which we also use throughout dietpi-software, when installing deb packages manually, is to run `apt -f install` afterwards, when running into dependency errors. `dpkg` itself is not able to pull dependency packages automatically, but APT can detect missing ones and pull them to finish the install/upgrade.

Sopor, 2019-08-20:
```
$ sudo apt install shared-mime-info
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 shared-mime-info : Depends: libglib2.0-0 (>= 2.35.9) but it is not going to be installed
                    Depends: libxml2 (>= 2.7.4) but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

Should i only run `apt --fix-broken install` ?

MichaIng, 2019-08-20:
@Sopor 
Okay, that is an issue due to mixed APT repositories it seems. Which SBC?
And can you paste:
```
apt policy libglib2.0-0
apt policy libxml2
apt policy shared-mime-info
```

Sopor, 2019-08-20:
```
$ apt policy libglib2.0-0
libglib2.0-0:
  Installed: (none)
  Candidate: 2.50.3-2
  Version table:
     2.50.3-2 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages

$ apt policy libxml2
libxml2:
  Installed: (none)
  Candidate: 2.9.4+dfsg1-2.2+deb9u2
  Version table:
     2.9.4+dfsg1-2.2+deb9u2 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages

$ apt policy shared-mime-info
shared-mime-info:
  Installed: (none)
  Candidate: 1.8-1+deb9u1
  Version table:
     1.8-1+deb9u1 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
```

MichaIng, 2019-08-20:
@Sopor 
Okay no repo mix, only one package available each with sufficient versions. Strange... What happens if you explicitly install those two libraries? Just be careful before confirming any install, since if APT runs into such errors, probably some package conflicts are the case and forcing one install can force the removal of another package then. So check the output first about what would be done 😉:
```
apt install libglib2.0-0 libxml2
```

Sopor, 2019-08-20:
```
$ sudo apt install libglib2.0-0 libxml2
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 webmin : Depends: shared-mime-info but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

MichaIng, 2019-08-20:
@Sopor 
😠 dump APT denies to install any package as long as the prior install has not finished, even the dependencies that would solve it...

`apt -f install` does not solve it as well?
Probably then this works: `apt -m --allow-change-held-packages install libglib2.0-0 libxml2`

If all does not work, the only idea, besides removing webmin, trying to solve dependencies in the first place and install again, is to hack away those two dependencies for `shared-mime-info` by editing `/var/lib/dpkg/status`. This then hopefully allows to finish webmin install, so you can take care the dependencies afterwards.

Sopor, 2019-08-22:
```
$ sudo apt --fix-broken install
Reading package lists... Done
Building dependency tree
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  libglib2.0-0 libxml2 shared-mime-info
Recommended packages:
  libglib2.0-data xdg-user-dirs xml-core
The following NEW packages will be installed:
  libglib2.0-0 libxml2 shared-mime-info
0 upgraded, 3 newly installed, 0 to remove and 6 not upgraded.
1 not fully installed or removed.
Need to get 4,061 kB of archives.
After this operation, 11.4 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://raspbian.playstar.se/raspbian stretch/main armhf libglib2.0-0 armhf 2.50.3-2 [2,527 kB]
Get:2 http://raspbian.playstar.se/raspbian stretch/main armhf libxml2 armhf 2.9.4+dfsg1-2.2+deb9u2 [806 kB]
Get:3 http://raspbian.playstar.se/raspbian stretch/main armhf shared-mime-info armhf 1.8-1+deb9u1 [728 kB]
Fetched 4,061 kB in 0s (4,605 kB/s)
Selecting previously unselected package libglib2.0-0:armhf.
(Reading database ... 54870 files and directories currently installed.)
Preparing to unpack .../libglib2.0-0_2.50.3-2_armhf.deb ...
Unpacking libglib2.0-0:armhf (2.50.3-2) ...
Selecting previously unselected package libxml2:armhf.
Preparing to unpack .../libxml2_2.9.4+dfsg1-2.2+deb9u2_armhf.deb ...
Unpacking libxml2:armhf (2.9.4+dfsg1-2.2+deb9u2) ...
Selecting previously unselected package shared-mime-info.
Preparing to unpack .../shared-mime-info_1.8-1+deb9u1_armhf.deb ...
Unpacking shared-mime-info (1.8-1+deb9u1) ...
Setting up libglib2.0-0:armhf (2.50.3-2) ...
No schema files found: doing nothing.
Setting up libxml2:armhf (2.9.4+dfsg1-2.2+deb9u2) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Setting up shared-mime-info (1.8-1+deb9u1) ...
Setting up webmin (1.930) ...
Webmin install complete. You can now login to https://DietPi:10000/
as root with your root password, or as any user who can use sudo
to run commands as root.
```

MichaIng, 2019-08-22:
@Sopor 
Okay so sticking with `dpkg -i package.deb || apt -f install` for deb package upgrades/installs it is 😄.

Sopor, 2019-08-23:
It seems that `sudo dietpi-update` couldn't continue because of this error so i was forced to use Exit. The problem here is that the services that was shut down wasn't started again. Should it really be like this?

MichaIng, 2019-08-23:
@Sopor 
Indeed currently a failing script will not autostart the services. But actually good idea. E.g. dietpi-update and dietpi-software could have `dietpi-services start` added to exit trap, once services were stopped. But when exiting regularly, and a reboot is done, this needs to be removed again.

MichaIng, 2019-08-27:
I mark this issue as closed. Feel free to reopen if required. Adding the service start exit trap to our services is something I will think through for v6.27.

closed 2019-08-27 by MichaIng