MichaIng/DietPi #3065: Service cannot start after restoring backup

x011, 2019-08-20:
### Creating a bug report/issue

#### Required Information
- DietPi version |  `6.25.3`
- Distro version | `stretch`
- Kernel version |` Linux DietPi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux`
- SBC device |  `RPi 3 Model B (armv7l)`
- Power supply used | `1.5A`
- SDcard used | `SanDisk`

#### Additional Information (if applicable)
- Software title |  `Owncloud `
- Was the software title installed freshly or updated/migrated?
`Freshly`
- Can this issue be replicated on a fresh installation of DietPi?
I haven't tried it.

#### Steps to reproduce
After deleting /mnt/dietpi_userdata by mistake, I restored a backup which completed successfully, but redis-server , mariadb and lighttpd cannot start.

#### Expected behaviour
All services to be active.

#### Actual behaviour
lighttpd
mariadb
redis-server

Cannot start

#### Extra details

- `systemctl status redis-server.service `

```
Process: 5384 ExecStart=/usr/bin/redis-server /etc/redis/redis.conf (code=exited, status=203/EXEC)
  Process: 5381 ExecStartPre=/bin/run-parts --verbose /etc/redis/redis-server.pre-up.d (code=exited, status=203/EXEC)

Aug 20 09:17:56 DietPi systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Aug 20 09:17:56 DietPi systemd[1]: Stopped Advanced key-value store.
Aug 20 09:17:56 DietPi systemd[1]: redis-server.service: Start request repeated too quickly.
Aug 20 09:17:56 DietPi systemd[1]: Failed to start Advanced key-value store.
Aug 20 09:17:56 DietPi systemd[1]: redis-server.service: Unit entered failed state.
Aug 20 09:17:56 DietPi systemd[1]: redis-server.service: Failed with result 'exit-code'.
```

- `systemctl status mariadb.service `

 ```
Active: failed (Result: exit-code) since Tue 2019-08-20 09:18:52 BST; 9s ago
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
  Process: 5508 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=203/EXEC)
  Process: 5416 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemct
l set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 5412 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 5409 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 5508 (code=exited, status=203/EXEC)
```

**systemctl status lighttpd.service**

```
lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; disabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/lighttpd.service.d
           └─dietpi-process_tool.conf, dietpi-services_edit.conf
   Active: failed (Result: exit-code) since Tue 2019-08-20 09:13:21 BST; 7min ago

Aug 20 09:13:21 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Aug 20 09:13:21 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Aug 20 09:13:21 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Aug 20 09:13:21 DietPi systemd[1]: Stopped Lighttpd Daemon.
Aug 20 09:13:21 DietPi systemd[1]: lighttpd.service: Start request repeated too quickly.
Aug 20 09:13:21 DietPi systemd[1]: Failed to start Lighttpd Daemon.
Aug 20 09:13:21 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Aug 20 09:13:21 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
```

MichaIng, 2019-08-20:
@x011
Many thanks for your report.
Can you try to start the binaries manually and paste the output here?
```
killall -qw redis-server mysqld lighttpd
redis-server /etc/redis/redis.conf
mysqld
lighttpd -f /etc/lighttpd/lighttpd.conf
```

MichaIng, 2019-10-18:
I close this issue due to missing reply. Feel free to reopen if required.

closed 2019-10-18 by MichaIng