MichaIng/DietPi #3068: Asus Tinker Board and RPi Touchscreen support

Strat00s, 2019-08-21:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

Don't know how to call this or if it's even a problem. Let's say that the problem is lack of supported "stuff" on Asus Tinker Board. It is supposed to be a "straight-up" replacement for RPis, with same component placement and an actual DSI and CSI connectors (but upgraded CPU/GPU and more RAM). The problem is mainly the lack the DSI tourchscreen support (such as the original 7' RPi Touchscreen).

#### Describe the solution you'd like:

Add support for it. I have no idea how, but I guess it can't be that difficult considering Armbian and Tinker OS have it and you can add it yourself (more on Describe alternatives).

#### Describe alternatives you've considered:

Alternatives I actually tried: If you get the Armbian Buster image for Tinker Board, it has the support built-in. And you can later "upgrade" to dietpi and keep the touchscreen support. That's the work-around I am running right now. Don't know how stable it is going to be and what the performance is going to be.

#### Additional context

I had to skip/remove this line from "PREP_SYSTEM_FOR_DIETPI.sh" because it was never able to remove the packages (idk why and idk which one actually caused the problem, so I just made it skip the line).
...
`# Purging additional packages, that (in some cases) do not get autoremoved:`
`# - dbus: Not required for headless images, but sometimes marked as "important", thus not autoremoved.`
`# - dhcpcd5: https://github.com/MichaIng/DietPi/issues/1560#issuecomment-370136642`
`# - mountall: https://github.com/MichaIng/DietPi/issues/2613`
`#G_AGP dbus dhcpcd5 mountall`--- comment this line or remove it
`G_AGA`
...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2019-08-27:
@Strat00s 
Many thanks for your request. By "built-in" you mean on ARMbian and TinkerOS you can attach the display and it works OOTB by simply attaching it? Or can you enable it with some integrated setup tool?

We have to dig a bid into the GPIO/DSI/CSI connector support for this board. Probably some additional dtoverlays/kernel modules need to be enabled or some firmware or library package(s).

Strat00s, 2019-08-29:
> @Strat00s
> Many thanks for your request. By "built-in" you mean on ARMbian and TinkerOS you can attach the display and it works OOTB by simply attaching it? Or can you enable it with some integrated [...]

Exactly. If you install TinkerOS or Armbian, it works right from the box. On Armbian however, you have to either rotate the image or rotate the touch layer. But it does work.

MichaIng, 2019-08-31:
@Strat00s 
Okay, so should be not too hard to find. However I will not be able to have a look until v6.26. We have to finish this update and there are still some bug fixes to do.