MichaIng/DietPi #3069: Dietpi-Services | disable autostart

svh1985, 2019-08-22:
Is it possible to disable autostart for qBittorrent?
I want to manually control the state of the service.

I installed qBittorrent with dietpi-software.

UNiXMIT, 2019-08-22:
Yes you can. Check out the following file and add qBittorrent to the excluded services:

`cat /DietPi/dietpi/.dietpi-services_include_exclude`

MichaIng, 2019-08-23:
@svh1985 @UNiXMIT 
This can be done with GUI now as well since v6.25:
`dietpi-services`, select `qbittorrent`, select `Include/Exclude` to exclude it. But this simply adds the line to `/DietPi/dietpi/.dietpi-services_include_exclude` as well.

svh1985, 2019-08-23:
But it seems that the service, even though excluded, still auto starts on boot. How do I prevent that?

MichaIng, 2019-08-23:
@svh1985 
Hmm, then it must be enabled for systemd. In `dietpi-services` assure it is set to "DietPi controlled" or run: `systemctl disable qbittorrent`

I know it might be a bid confusing:
- systemctl enable/disable only controls whether it is started at boot by systemd (native init system + service handler) or not.
- In dietpi-services these states are called "systemd controlled"/"DietPi controlled", since if systemd does not start the service at boot, DietPi will at a later boot stage. But actually that does not really fit. Even if systemd starts it at boot, DietPi will still loop through it as well (although nothing happens since it already runs), and DietPi will start/stop/restart it as well during updates/installs/backups etc. So known services are in fact always controlled by DietPi as long as not excluded.

So what you usually want is either of these:
- DietPi controlled + included: DietPi will start the service save at a later boot stage, well ordered with other services, respecting dependencies. And all critical tasks, like updates, installs, backups, benchmarks, ..., will stop the service and restart it once done.
- systemd controlled + excluded: systemd will start the service at boot, earlier then DietPi would do, but respecting dependencies defined in the service file itself, which works well in most cases. DietPi will not stop it on critical maintenance tasks. Not really recommended but probably one might have a reason to keep a certain service running that handles a long lasting task that you definitely want to keep running.
- DietPi controlled + excluded: Nothing will start the service, neither at boot, nor after maintenance tasks.

svh1985, 2019-08-24:
I found the issue;
I had `/DietPi/dietpi/dietpi-services start qbittorrent` in openvpn **up**. But it seems like that the UP command still gets executed even if the VPN fails, which it sometimes does for no obvious reason. 😑

So, maybe we need to change **up** to **route-up** in PR: https://github.com/MichaIng/DietPi/pull/3047
I'll test in the coming days and report back here.

MichaIng, 2019-08-24:
@svh1985 
Ah great. I just read the man page about `route-up`. Indeed makes sense. Usually one would expect that these scripts/commands run after the VPN has completely initiated (not just the devs).
`down` should be likewise replaced by `route-pre-down`, so it runs when the VPN is still fully up?
This then allows to add to both scripts that require the VPN connection to be still accessible.

MichaIng, 2019-08-27:
Okay I mark this issue as closed. Feel free to reopen if required.

closed 2019-08-27 by MichaIng