MichaIng/DietPi #3070: Unable to access desktop via VNC

Trunkzeh, 2019-08-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | buster
- Kernel version | Linux RPI4 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux
- SBC device | RPi 4 Model B (armv7l)
- Power supply used | Official RPI4 USB-C Adapter
- SDcard used | Sandisk Extreme 32GB

#### Additional Information (if applicable)
- Software title | XFCE & VNC4 / RealVNC
- Was the software title installed freshly or updated/migrated? Fresh 6.25 install & then XFCE VNC4 on first attempt. Then re-install 6.25 and XFCE and RealVNC.
- Can this issue be replicated on a fresh installation of DietPi? Yes.
- Bug report ID | 883d2f7b-94e0-4a7e-90e2-2d7b2bf59b16

#### Steps to reproduce

1. Install latest version of DietPi
2. Install XFCE desktop
3. Install either VNC4 or RealVNC
4. Dietpi-Autostart wizard will appear; select either Automatic Login (2) or LightDM login mask (16)
5. Restart RPI4.
6. Attempt to connect via VNC viewer on client; VNC will connect however will show 'Cannot currently show the desktop'

#### Expected behaviour

- VNC session should show local desktop.

#### Actual behaviour

- VNC session shows 'Cannot currently show the desktop' message on black background.

#### Extra details

- cat .xsession-errors:

root@RPI4:~# cat .xsession-errors
Xsession: X session started for  at Thu 22 Aug 11:38:14 BST 2019
localuser:root being added to access control list
dbus-update-activation-environment: systemd --user not found, ignoring --systemd argument
dbus-update-activation-environment: setting VNCDESKTOP=x11
dbus-update-activation-environment: setting HOSTNAME=RPI4
dbus-update-activation-environment: setting XDG_SESSION_TYPE=x11
dbus-update-activation-environment: setting SHLVL=1
dbus-update-activation-environment: setting HOME=/root
dbus-update-activation-environment: setting DBUS_SESSION_BUS_ADDRESS=unix:path=/tmp/.vnc-0/run/session-75b57e20c88642b3/bus
dbus-update-activation-environment: setting JOURNAL_STREAM=8:15860
dbus-update-activation-environment: setting XDG_SESSION_CLASS=user
dbus-update-activation-environment: setting PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
dbus-update-activation-environment: setting INVOCATION_ID=7f3cb648eff9456db909c839e69d3ada
dbus-update-activation-environment: setting XDG_RUNTIME_DIR=/tmp/.vnc-0/run
dbus-update-activation-environment: setting DISPLAY=:1
dbus-update-activation-environment: setting LANG=en_GB.UTF-8
dbus-update-activation-environment: setting XAUTHORITY=/root/.Xauthority
dbus-update-activation-environment: setting SHELL=/bin/bash
dbus-update-activation-environment: setting DISPLAYNUM=1
dbus-update-activation-environment: setting GPG_AGENT_INFO=/run/user/0/gnupg/S.gpg-agent:0:1
dbus-update-activation-environment: setting PWD=/root
dbus-update-activation-environment: setting XDG_DATA_DIRS=/usr/share/xfce4:/usr/local/share/:/usr/share/
/usr/bin/x-session-manager: X server already running on display :1
xfce4-session: No SSH authentication agent found
gpg-agent[933]: WARNING: "--write-env-file" is an obsolete option - it has no effect
gpg-agent: a gpg-agent is already running - not starting a new one

(xfce4-session:922): xfce4-session-WARNING **: 11:38:15.653: gpg-agent returned no PID in the variables
[... 78 lines cut ...]
(xfsettingsd:908): xfsettingsd-WARNING **: 11:41:34.233: Failed to get the _NET_NUMBER_OF_DESKTOPS property.

** (xfdesktop:901): WARNING **: 11:41:34.244: Thumbnailer failed calling GetFlavors

(xfwm4:893): xfwm4-WARNING **: 11:41:34.804: Error opening /dev/dri/card0: No such file or directory

(Thunar:899): dbind-WARNING **: 11:41:35.443: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
xfce4-session: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.
xfwm4: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.0.

(xfsettingsd:908): libxfce4ui-WARNING **: 11:45:30.669: ICE I/O Error

(xfsettingsd:908): libxfce4ui-WARNING **: 11:45:30.669: Disconnected from session manager.

(xfdesktop:901): libxfce4ui-WARNING **: 11:45:30.670: ICE I/O Error

(xfdesktop:901): libxfce4ui-WARNING **: 11:45:30.671: Disconnected from session manager.
xfsettingsd: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.0.
xfdesktop: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.0.
migrate: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.0.
Gdk-Message: 11:45:30.681: Thunar: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.0.

(xfce4-panel:897): libxfce4ui-WARNING **: 11:45:30.694: ICE I/O Error

(xfce4-panel:897): Gtk-CRITICAL **: 11:45:30.694: IA__gtk_main_quit: assertion 'main_loops != NULL' failed

Trunkzeh, 2019-08-22:
As a site-note I've set the following hdmi flags in /Dietpi/config.xt:

hdmi_group=2
hdmi_mode=82

This is a requirement to simulate an HDMI signal from a monitor as per various articles on the raspberrypi forums.

olem10, 2019-08-24:
I have just been into this one using realvnc, even though I did not get the same error messages. 
If you are going to vnc into your desktop (:0) you need to enable vncserver-x11-serviced
% sudo systemctl enable vncserver-x11-serviced.service
You also need to set booting into your desktop using dietpi-config
If you want a virtual desktop enable vncserver-x11-virtuald.service.

MichaIng, 2019-08-24:
@Trunkzeh 
Many thanks for your report.

Did you set `SOFTWARE_VNCSERVER_SHARE_DESKTOP=1` in `dietpi.txt`, so attempt to connect to an existing "physical" desktop session? In this case did you login (in case LightDM autostart), so that the desktop is active on physical screen, before attempting to connect via VNC?
- Only in case of shared desktop mode, the `config.txt` HDMI settings have an effect. The settings you chose are DMT (usually regular PC minor) 1080p.
- AFAIK for the VNC clients it does not matter which one you choose, as long as the physical system + monitor supports this mode. So I would try to leave the settings commented/removed, to allow the system auto-detect the correct mode, or a most compatible one. As long as the desktop shows up well on physical monitor, and it is not some extreme resolution, like 4k, the clients should show it as well.

If you did not set `SOFTWARE_VNCSERVER_SHARE_DESKTOP=1`, the VNC server will create a virtual desktop, thus the physical HDMI settings will have no effect. Instead `SOFTWARE_VNCSERVER_*` settings in `dietpi.txt` can be used to define virtual screen size.

Since it should be less conflicting in general, please try it with virtual desktop mode first, thus: `SOFTWARE_VNCSERVER_SHARE_DESKTOP=0`

For general debugging on service level:
```
systemctl status vncserver
```
With RealVNC additionally:
```
systemctl status vncserver-x11-virtuald
systemctl status vncserver-x11-serviced
```

@olem10
Thanks for sharing. I just checked the code and both services are enabled within the dietpi-software install script: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software#L9205-L9206
However actually I am not to happy with the way we implement this. We have a shared service for both, RealVNC and TigerVNC, which makes it hard to think through for me (the code was created before I joined DietPi). And since RealVNC has its own services, e.g. our service does simply nothing when shared desktop mode is chosen, leaving it to `vncserver-x11-serviced`. But with virtual desktop mode chosen, our service will create an instance, using the `/usr/bin/vncserver` binary. Is this even present with RealVNC? I just see `vncserver-x11` in the package. And also does `vncserver-x11-virtuald` probably already initiate a virtual desktop, which could then double or conflict with the one we initiate?

I probably find time to run some tests with RealVNC. Possibly this has simply changed a bid with Buster, so our implementation requires an update. Most likely I will then separate the install scripts, so our own service is for TigerVNC only, while we use+configure RealVNC native services separately.

> /usr/bin/x-session-manager: X server already running on display :1

This indeed looks like a doubled/conflicting VNC session on display 1 (default for virtual desktop mode).

Trunkzeh, 2019-08-26:
@MichaIng 

No I hadnt changed any settings on Dietpi.txt; I checked the SOFTWARE_VNCSERVER_SHARE_DESKTOP string nonetheless and it is set to 0. I've removed RealVNC; rebooted; installed TigerVNC; rebooted and then removed the two hdmi strings in config.txt and rebooted one last time:

vncserver.service - Manage VNC Server (DietPi)
   Loaded: loaded (/etc/systemd/system/vncserver.service; enabled; vendor preset: enabled)
   Active: active (exited) since Mon 2019-08-26 20:31:10 BST; 6min ago
 Main PID: 620 (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4525)
   Memory: 73.9M
   CGroup: /system.slice/vncserver.service

Aug 26 20:31:11 RPI4 vncserver[620]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserve
r line 357.
Aug 26 20:31:11 RPI4 vncserver[620]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserve
r line 357.
Aug 26 20:31:11 RPI4 vncserver[620]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserve
r line 357.
Aug 26 20:31:11 RPI4 vncserver[620]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserve
r line 357.
Aug 26 20:31:11 RPI4 vncserver[620]: New 'RPI4:1 (root)' desktop at :1 on machine RPI4
Aug 26 20:31:11 RPI4 vncserver[620]: Starting applications specified in /root/.vnc/xstartup
Aug 26 20:31:11 RPI4 vncserver[620]: Log file is /root/.vnc/RPI4:1.log
Aug 26 20:31:11 RPI4 vncserver[620]: Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/
passwd RPI4:1 to connect to the VNC server.
Aug 26 20:31:12 RPI4 dbus-daemon[814]: [session uid=0 pid=812] Activating service name='org.xfce.Xfconf' r
equested by ':1.0' (uid=0 pid=810 comm="xfce4-session ")
Aug 26 20:31:12 RPI4 dbus-daemon[814]: [session uid=0 pid=812] Successfully activated service 'org.xfce.Xf
conf'

MichaIng, 2019-08-27:
I am running some test currently. TigerVNC on RPi2. VNC connection from client fails immediately:
```
Aug 27 03:52:28 DietPi vncserver[501]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserver line 357.
Aug 27 03:52:28 DietPi vncserver[501]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserver line 357.
Aug 27 03:52:28 DietPi vncserver[501]: /usr/bin/xauth:  file /root/.Xauthority does not exist
Aug 27 03:52:28 DietPi vncserver[501]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserver line 357.
Aug 27 03:52:28 DietPi vncserver[501]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserver line 357.
Aug 27 03:52:28 DietPi vncserver[501]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserver line 357.
Aug 27 03:52:28 DietPi vncserver[501]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserver line 357.
Aug 27 03:52:28 DietPi vncserver[501]: Use of uninitialized value $proto in socket at /usr/bin/tigervncserver line 357.
Aug 27 03:52:28 DietPi vncserver[501]: New 'DietPi:1 (root)' desktop at :1 on machine DietPi
Aug 27 03:52:28 DietPi vncserver[501]: Starting applications speci```
fied in /root/.vnc/xstartup
Aug 27 03:52:28 DietPi vncserver[501]: Log file is /root/.vnc/DietPi:1.log
Aug 27 03:52:28 DietPi vncserver[501]: Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd DietPi:1 to connect to the VNC server.
Aug 27 03:52:34 DietPi dbus-daemon[620]: [session uid=0 pid=614] Activating service name='ca.desrt.dconf' requested by ':1.1' (uid=0 pid=611 comm="light-locker ")
Aug 27 03:52:34 DietPi dbus-daemon[620]: [session uid=0 pid=614] Successfully activated service 'ca.desrt.dconf'
```
- `Use of uninitialized value $proto` I know this error from perl scripts: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=928314
- `getprotobyname()` depends on `netbase`: `G_AGI netbase`
- Those error messages are solved by this, but connection still fails.
- But to be true, no display attached, framebuffer at 16x16.

Will try tomorrow after setting up display and local desktop.

Trunkzeh, 2019-08-27:
@MichaIng And I thought I was going mad.. thanks :-) I'll continue fettling with it today to see if I can get it working. I've moved the branch on it to 'dev' so if you push any changes to that branch I can test.

Trunkzeh, 2019-08-27:
Log from ~/.vnc/xstartup:

`Tue Aug 27 11:57:08 2019
 Connections: accepted: 192.168.128.225::52846
 SConnection: Client needs protocol version 3.8
 SConnection: Client requests security type VncAuth(2)
terminate called after throwing an instance of 'rdr::Exception'
terminate called recursively
(EE)
(EE) Backtrace:
(EE)
(EE)
Fatal server error:
(EE) Caught signal 6 (Aborted). Server aborting
(EE)
migrate: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.0.
Gdk-Message: 11:57:14.408: Thunar: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.0.

xfdesktop: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.0.
xfsettingsd: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.0.
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":1.0"
      after 46 requests (38 known processed) with 0 events remaining.
xfwm4: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.0.
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":1"
      after 175 requests (175 known processed) with 0 events remaining.
xfce4-session: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.`

Out of curiosity; whats the default vnc password on a fresh VNC server install? I assume its dietpi? I ran vncpasswd (and set it to dietpi) and rebooted and still the same issue.

Trunkzeh, 2019-08-27:
Fixed the bloody problem. Issue is reported https://github.com/TigerVNC/tigervnc/issues/800 and https://github.com/TigerVNC/tigervnc/issues/812; workaround is to ensure vncpasswd sets the readonly password as well as a full-control password. Current behavior is only full-control.

MichaIng, 2019-08-28:
@Trunkzeh 
Many thanks for this find. Actually I was not able to identify a way to apply the workaround. Since `vncpasswd` has not further command options, how do you do that?

The issues you linked mention some required code change. Also the linked Debian bug report it looks like some code change in the mentioned library is required. Discussion there ended in April 🤔.

Trunkzeh, 2019-08-29:
@MichaIng Running vncpasswd will first ask you to set the full-control password; it will then ask if you want to set a read-only password as well (select Y) then enter and confirm the read-only pass.

Trunkzeh, 2019-08-29:
An idea to automate the workaround is to copy a manually created passwd that you have previous ran vncpasswd and had set two passwords ('dietpi' can be the full-control and 'viewonly' can be the view-only for example) and copy this to ~/.vnc/passwd as part of the TigerVNC install.

MichaIng, 2019-08-31:
@Trunkzeh 
PR up: https://github.com/MichaIng/DietPi/issues/3070
Resolved a bunch of other minor issues I recognised.

Many thanks for finding the workaround, the `-f` prevents to pass password doubled to STDIN. I used now just the same global password for view-only. Not sure what it's used for? I can tell the client to connect in view-only mode, but I still need to enter the control password.

Trunkzeh, 2019-09-02:
@MichaIng Perfect. I'll do a fresh install tonight and feedback.

MichaIng, 2019-09-02:
@Trunkzeh 
Note that you need to switch the `dev` branch for that:
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=dev' /DietPi/dietpi.txt
dietpi-update 1
dietpi-software install 28
```

Trunkzeh, 2019-09-02:
@MichaIng: Worked like a charm. Couple of things to note:

1. Installing TigerVNC using dietpi-software install 28 automatically installs LXDE as the desktop environment. There should be an option to select which desktop environment you wish to use as opposed to defaulting to LXDE. I'm going to test uninstalling LXDE via dietpi-software and installing another desktop environment such as MATE (I prefer XFCE but will wait till XFCE 4.14 is rolled into DietPi - I'll set up a ticket on github with any work required to upgrade from XFCE .12 to .14)

2. After TigerVNC is installed; DietPI needs a restart before accepting VNC connections. I'm not sure if this is because the VNC daemon isn't running or the desktop environment isn't running.

MichaIng, 2019-09-02:
@Trunkzeh 
Thanks for testing.

> There should be an option to select which desktop environment you wish to use as opposed to defaulting to LXDE.

Jep, that is an outstanding task. I plan to implement a desktop preference system like it exists for webservers. So you select a preferred desktop within `dietpi-software` menu, and as fast as a desktop is required, instead of defaulting to LXDE, the preference is respected. A related `dietpi.txt` setting will be added as well then, to choose this for automated first run installs.

> I'll set up a ticket on github with any work required to upgrade from XFCE .12 to .14

We will most likely not gonna offer some custom install paths for desktops. The configuration and maintenance effort is simply too high. But no harm to have a HowTo available for users which want to switch/test.
But Debian Bullseye ships XFCE 4.14: https://packages.debian.org/bullseye/xfce4
With some luck, it is backported to Buster, lets hope for it.

> After TigerVNC is installed; DietPI needs a restart before accepting VNC connections.

This is since VNC servers are not controlled by DietPi-Services, so they are not started after installs have finished. This is required since, e.g. if you run installs from within a VNC session, dietpi-services would kill your session and itself by this 😉. Those services (like SSH and all network related services as well) are `enabled`, so started by systemd on boot.
However you can start it manually after installs: `systemctl start vncserver`
And we can add it to be started separately at the end of installs, when VNC server was just installed.

Trunkzeh, 2019-09-03:
@MichaIng The separate ticket was just to see what work would be required for DietPi to use the latest XFCE release; which could then maybe be merged into a PR. I initially thought the .14 release would stay in debian sid for awhile but if its in bullseye then it should trickle down into buster-backports.

Removing LXDE and installing another desktop environment works. However the VNC server is inaccessible; i suspect this is due to the .vncsession file points to the old desktop environment? In any case; doing dietpi-software reinstall 28 does the trick. My RPI4 is now hovering around the 62c mark so will switch to XFCE4 next and see how that gets on :-)

MichaIng, 2019-09-03:
@Trunkzeh 
> However the VNC server is inaccessible; i suspect this is due to the .vncsession file points to the old desktop environment?

Ah yeah that's true, not the `.vncsession`, but `/root/.vnc/xstartup`, where the last line need to be replaced with `xfce4-session` session.
Of course a reinstall covers this.

Trunkzeh, 2019-09-03:
I thought so :-) Found some builds for Raspbian 10 and XFCE 4.14; for those that want to experiment with the new builds:

Pre-requisite: XFCE 4.12 already installed on your system

```
echo 'deb http://download.opensuse.org/repositories/home:/stevenpusser:/xfce-4.14/Raspbian_10/ /' > /etc/apt/sources.list.d/home:stevenpusser:xfce-4.14.list
rm -f Release.key && wget -nv https://download.opensuse.org/repositories/home:stevenpusser:xfce-4.14/Raspbian_10/Release.key -O Release.key
apt-key add - < Release.key && rm -f Release.key
apt update
apt full-upgrade
```

Builds are from an openSUSE maintainer; and there are now armhf debs so works on our Pi platform. This is untested so please make sure you backup before you make any changes :)

Trunkzeh, 2019-09-03:
@MichaIng The original reported issue has been fixed by your PR. Should I close this issue or leave it open for if/when the desktop preference system is implemented?

MichaIng, 2019-09-03:
@Trunkzeh 
Lets leave it open for the beta phase. There have been some more changes done, and it makes sense that beta testers have a special eye on VNC server installs and runs. So we can fine tune the solution/code in case, or even fix the one or the other additional issue recognised.

Self reminder: Start vncserver after it has been installed => General approach: Collect non-controlled service name during installs, start those explicitly before exit, since `dietpi-services start` will not do it.

PR #3086 (2019-09-01): DietPi-Software | TigerVNC: Resolve immediate coredump on connect + other minor issues
**Status**: Ready
- [x] Changelog

**Testing**:
- [x] Stretch
- [x] Buster
- [x] Bullseye

Fixes: #3070

**Commit list/description**:
+ DietPi-Software | Fix wrong shown pre-req software name being installed, when desktop is auto-selected. $software_id is set to SQLite before, which is shown instead of correct LXDE.
+ DietPi-Software | TigerVNC: Install the netbase package, which is required to solve a failing perl function due to missing dependency.
+ DietPi-Software | TigerVNC: Resolve "No session for PID  [... cut]
closed 2019-10-15 by MichaIng