MichaIng/DietPi #3073: Choosing language for keyboard when installing looks wierd, the buster build for Rpi

bobby276070, 2019-08-22:
Hi, please have a look at the menue for selecting language for keyboard, what I see when scrolling is in most cases a lot of black squares instead for specific language, in my case I was looking for Swedish, found a line named "Sweed" - Sweed is neither English or German for Swedish.

Another important point is that if I like to give a good password for root/dietpi and SW-setup, this has to take place after changing to the proper language of keyboard. Maybe have he choise of keyboard to use and activate, before the question of changing the passwords .. ?!

Many thanks Robert

MichaIng, 2019-08-23:
@bobby276070 
Many thanks for your report.

Indeed a known issue with Debian Buster: https://github.com/MichaIng/DietPi/issues/2912
Sadly they did not fix it yet... I am wondering why this didn't get some priority BEFORE official Debian release. If one cannot read the keyboard menu and keys are assigned wrong, then this can break the whole Debian console usage...

MichaIng, 2019-08-27:
I close this issue in favour of the other one. We cannot do much besides hoping for a soon solution from Debian. A custom workaround would be much work, which is rendered obsolete with a Debian-wise fix, we at least me not gonna do that 😉.

closed 2019-08-27 by MichaIng