MichaIng/DietPi #3074: Use another text color when updating

Sopor, 2019-08-23:
Could you use another color for the text? It is even harder to see it real on the monitor than in the screenshot below. 

[image]

Sopor, 2019-08-23:
Same at the login banner

[image]

And again, it is much easier to see it here on a screenshot than it is on the monitor.

MichaIng, 2019-08-23:
@Sopor 
Many thanks for your request. The plan is to make all those colour customisable, since it depends on the terminal and screen, how they look like and how well they are readable. The idea generally was to separate important outputs from less important ones, e.g. informational mess vs results and such.

See here: https://github.com/MichaIng/DietPi/issues/2651

This can be expanded for all DietPi outputs generally. Not too much work since we used shared functions from `dietpi-globals` mostly.

MichaIng, 2019-08-27:
I close this issue on favour of the existing, however cross-linked it here: https://github.com/MichaIng/DietPi/issues/2651
A general colour code customisation makes sense, as mentioned, to tune colours for the local terminal/screen.
Moreover actually, to make it 100% effective, it should be separate sets for local terminal, serial console and SSH. This is not too hard, since we can get the attached tty via `tty` command. SSH use `/dev/pts/[0-9]`, local console `/dev/tty[1-9]` and serial console would be everything else, usually `/dev/ttyS[0-9]` (software-based), `/dev/ttyAMA[0-9]` (ARM hardware serial device), `/dev/ttySAC[0-9]` (hardware serial device on some Odroids AFAIK), but `/dev/ttyFIQ[0-9]` or something like that is also seen 😄.

closed 2019-08-27 by MichaIng