MichaIng/DietPi #3075: Possible to choose date format on login banner?

Sopor, 2019-08-23:
I wondering if you could make the `dietpi-banner` also change date format? Now it seems to use dd/mm/yy. In Sweden we use yyyy-mm-dd and in US they use mm/dd/yyyy. Now it will show:  `DietPi v6.25.3 : 17:44 - Fri 23/08/19`

The year should have 4 digits and it would be nice if it was possible to choose from these three formats:
1. mm/dd/yyyy
2. dd.mm.yyyy (maybe you should add dd/mm/yyyy too?)
3. yyyy-mm-dd

There are others too but these are the most common ones https://en.wikipedia.org/wiki/Date_format_by_country

Even if you don't add this i believe you should at least add 20 so the year will show 2019.

MichaIng, 2019-08-23:
@Sopor 
Many thanks for your request.

Jep, either a custom date format or the one that matches the chosen locale for console, could be used.

Does `date +"%R - %a %x"` show it in the format you like? The order is given, but the format should match you console locale then. `%x` although I think always shows year with only two digits, at least with dots then?

I couldn't find a way to show the whole date with 4-digits year, without the need to manually add the whole format with day, month, year and separators: https://www.gnu.org/software/coreutils/manual/html_node/Date-conversion-specifiers.html#Date-conversion-specifiers
I am not keen to add a bunch of typical formats for selection, which then cover some users, but not all. Would be easier to show on chosen locale (with two digits year then) automatically. Or we allow to define the whole format manually, so one can read the docs and fully customise.

MichaIng, 2019-08-26:
@Sopor 
Okay I did a quick fix: https://github.com/MichaIng/DietPi/commit/db47a9466ff452bf20c48bf0bd474aa85f3fd6e4
It now prints date/time in the format of the system locale you chose.

Year in 4-digits requires more effort, a choice menu with then all wider used formats added indeed. But this is out of scope for v6.26. A larger banner customisation with sub menus is something for the next version of if someone else want to contribute 🙂.

closed 2019-10-15 by MichaIng