MichaIng/DietPi #3076: dietpi-wifi.txt is missing

olem10, 2019-08-24:
### Creating a bug report/issue
Dear,

I just downloaded the DietPi_v6.25_RPi-ARMv6-Buster.img image for raspberry pi.

The dietpi-wifi.txt is missing from the boot folder preventing configuration by WiFi.

Regards
Ole

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2019-08-24:
@olem10 
Many thanks for your report.

However, I just checked and the file exists in the archive:
[image]

This file is created, even in an error handled way, so impossible this fails: https://github.com/MichaIng/DietPi/blob/dev/PREP_SYSTEM_FOR_DIETPI.sh#L1606

PeterLacknase, 2019-08-24:
The file gets removed during initial boot setup, since it contains unencrypted wifi credentials It‘s no bug, it‘s a (very good) feature.

MichaIng, 2019-08-24:
@PeterLacknase 
That is true. Although I guess @olem10 meant the situation before booting the first time? Probably wrong assumption.

`dietpi-wifi.txt` indeed is only meant to configure WiFi BEFORE first boot. Since on most images this is a dedicated FAT partition, this allows pre-configuring WiFi from a Windows system that cannot mount the ext4 root partition.

After first boot one can use `dietpi-config` to setup WiFi. Credentials are then applied to wpasupplicant settings and stored to a safer place where only root can read.

olem10, 2019-08-24:
@PeterLacknase @MichaIng 
I read a tutorial where I got the impression that the dietpi-wifi.txt was present in the boot folder before booting the first time. You can also get this impression by reading the comments in dietpi.txt.

However I was able to solve the problem by using  dietpi-config, so no worries.

MichaIng, 2019-08-24:
@olem10 
> I got the impression that the dietpi-wifi.txt was present in the boot folder before booting

Ah did you check the /boot dir on the second ext4 partition? That is indeed misleading and in case I will fix the tutorials about this: The file is present on the first FAT partition, which is only mounted to /boot on running system. On the raw SDcard, the /boot dir is empty, since it is only a mount point.

MichaIng, 2019-08-26:
Okay I mark this issue as closed. Feel free to reopen if required.

closed 2019-08-26 by MichaIng