MichaIng/DietPi #3077: MD5 and SHA1 don't match the hash.txt

mbonne, 2019-08-24:
### Creating a bug report/issue

Downloaded the current release of the VMware version from site: 
https://dietpi.com/downloads/images/DietPi_VMware-x86_64-Buster.7z

Ran a comparison check against the included files using the hash.txt in .7z file.

Comparing SHA1 values: DietPi_VMware-x86_64-Buster.vmdk - doesn't match.
The SHA1 from txt: a5444a3b425f578a905a5879a0ebe3d4be19856a
Your files Hash:  844d87bda28f04d640c15c79460fe23bba36a868

DietPi_VMware-x86_64-Buster.vmx matches up.
 Correct Hash:    ba42a6ea3ea158e423f836fd61fc1d010014107f
 Your files Hash: ba42a6ea3ea158e423f836fd61fc1d010014107f

Please advise.

==================================================
Filename          : DietPi_VMware-x86_64-Buster.vmdk
MD5               : 13e92936452e6f0e8b4d68cccab580d7
SHA1              : a5444a3b425f578a905a5879a0ebe3d4be19856a
CRC32             : bfb6bb2e
SHA-256           : a5a9d6d8214e5a569bb4f5169568c54f8cf43e71c44987119833e123496e9070
SHA-512           : 1df82d4d4cfcf395ec1ca4bbd29be4e2dd71a65493b98fa8547c4493d6a2bb5589e1ddfe2092a8aa023d6108212f775cd35f94a1911b7a157198061d983a2eba
SHA-384           : 38a273e0616c35be4939c47d00016b7dab9ff46886d74a9c397b3c7f5c0701331698d70fcc807131f1742a1681d5953a
Modified Time     : 11.07.2019 00:41:33
Created Time      : 10.07.2019 23:28:11
Entry Modified Time: 11.07.2019 00:41:33
File Size         : 791.937.024
File Version      : 
Product Version   : 
Identical         : 
Extension         : vmdk
File Attributes   : A
==================================================

==================================================
Filename          : DietPi_VMware-x86_64-Buster.vmx
MD5               : 64f5d7ca3390cb7f1239cdf0fdf721a0
SHA1              : ba42a6ea3ea158e423f836fd61fc1d010014107f
CRC32             : d3f90b73
SHA-256           : 14ef584c566b2a4264bc8e34bca17bdf92bb0efcad8be81c3067526e8f39fcef
SHA-512           : 363b1ad515177879262fd503711b61ef82816bb7ab6b19a5fc0eca4946aaec9573f732519cd89aba9c0e4da87384b6d40380b9bdaf0f51920679ad2e783229b9
SHA-384           : 1485158b1b98aa7cdb27bc69105a76967b2c4b460da6d203e9c8d4ece81abbb090d8f14464e4a3ec630cebc92fbcb892
Modified Time     : 11.07.2019 00:40:07
Created Time      : 11.07.2019 00:40:07
Entry Modified Time: 11.07.2019 00:40:07
File Size         : 2.434
File Version      : 
Product Version   : 
Identical         : 
Extension         : vmx
File Attributes   : A
==================================================

MichaIng, 2019-08-25:
@mbonne 
Many thanks for your report.

I just checked and verified that vmdk hashes in `hash.txt` are wrong and I get the same SHA1 you got. Also checked the archive on server. Obviously a packaging mistake, perhaps a last-minute check/change and forgotten to redo the hashes.

This means you can safely use the image. I am currently repacking the archive with the correct hashes.

mbonne, 2019-08-25:
Thanks for the prompt response.
🍻

closed 2019-08-25 by mbonne