MichaIng/DietPi #3078: Docker Compose is missing in the Software Cataloge

Gill-Bates, 2019-08-25:
Dietpi offers docker inside dietpi-software catalog. But Docker-compose is missing. IMHO should be added.

MichaIng, 2019-08-26:
@Gill-Bates 
Many thanks for your request.

Simple binary install, talking care the dependencies manually: https://docs.docker.com/compose/install/
Debian package install: https://packages.debian.org/buster/docker-compose

ModischFabrications, 2019-10-07:
I had the same expectation that DietPi-Docker installed docker-compose with it and arrived here while researching the problem. It might be sensible to add it to the docker installation or at least add a prompt for "optional extras: ...".

` sudo apt install docker-compose` worked flawless for anyone with the same problem.

Joulinar, 2020-12-27:
@MichaIng 
I was thinking to look into the implementation of `docker-compose`. Easiest way is to go with the `apt` package. However it's 2.5 years old `docker-compose` version (1.21.0-3). Probably better to have it installed via `pip` version (1.27.4)?? Even if it will take longer to compile. What do you think?

MichaIng, 2020-12-27:
I think since we pull docker-ce from official upstream sources as well, we should go with the up-to-date PyPI (pip) composer. There seems to be a pre-compiled wheel available for "any" architecture: https://pypi.org/project/docker-compose/#files
Let's see how long the pip install really takes.

Joulinar, 2020-12-27:
I will do a comparison between RPi1,3,4 and VM Stretch/Buster this evening

Joulinar, 2020-12-27:
Below my test result

**Buster**
- RPi B (armv6l)
- RPi 3 Model B+ (armv7l)
- RPi 4 Model B (aarch64)
- Virtual Machine 1 (x86_64)

**Stretch**
- Virtual Machine 2 (x86_64)

**DietPi version**
- v6.34.3

**Software installed**
- ID 130 - Python 3 pip

**Additional dependencies**
On RPi 1/3
- make gcc libffi-dev libssl-dev

On RPi 4
- make gcc

VM 1/2
- none

**Install command**
- `pip3 install docker-compose`

**Timing**
- `time` command was used to calculate installation

- VM 1: 1m6.168s
- VM 2: 1m14.006s
- RPi4: 4m35.200s
- RPi3: 28m43.226s
- RPi1: still compiling

MichaIng, 2020-12-28:
Puhh, yes I remember pynacl being a major time consumer in other cases as well. The question is if many users will run a multi-container application on RPi 1/Zero anyway 🤔. I cannot imagine this being a good idea 😄.

Strange that you didn't need libffi-dev + libssl-dev on RPi 4. Was it probably the 64-bit image so that arm64 vs armhf makes the difference here, i.e. there are pre-compiled pynacl wheels for arm64 but not for armhf?

Joulinar, 2020-12-28:
I was surprised as well that I don't need `libffi-dev libssl-dev` on the RPi4B 64bit. Will do another test tomorrow as it was not that painful like RPi1/3

Looks like there is no pre-compiled pynacl wheels for arm64 https://pypi.org/project/PyNaCl/#files

MichaIng, 2020-12-28:
This is not the only repository. non-64-bit RPis pull from piwheels.org as well, and very strange is that there are in fact PyNaCl wheels for ARMv7 and ARMv6 available for Python 3: https://www.piwheels.org/project/pynacl/
I never saw this wheel being pulled, it seems to be not detected correctly. Would be a huge time saver if it was, a task for tomorrow:  https://github.com/piwheels/piwheels

I do not remember where arm64/ARMv8 wheels are additionally pulled from.

Joulinar, 2020-12-28:
OK I was able to reduce runtime on my RPi3B+ from 28 minutes down to 30 seconds 😄 

The trick is to specify `piwheels.org` as source
`pip3 install docker-compose -i https://www.piwheels.org/simple`

RPi OS 64bit is not supported as it is still BETA

<details>
  <summary>Logfile attached. Click to expand!</summary>

```
root@DietPi3:~# time pip3 install docker-compose -i https://www.piwheels.org/simple
Looking in indexes: https://www.piwheels.org/simple
Collecting docker-compose
  Downloading https://www.piwheels.org/simple/docker-compose/docker_compose-1.27.4-py2.py3-none-any.whl (110 kB)
     |████████████████████████████████| 110 kB 381 kB/s
Collecting cached-property<2,>=1.2.0
  Downloading https://www.piwheels.org/simple/cached-property/cached_property-1.5.2-py2.py3-none-any.whl (7.6 kB)
Collecting distro<2,>=1.5.0
  Downloading https://www.piwheels.org/simple/distro/distro-1.5.0-py2.py3-none-any.whl (18 kB)
Collecting docker[ssh]<5,>=4.3.1
  Downloading https://www.piwheels.org/simple/docker/docker-4.4.1-py2.py3-none-any.whl (146 kB)
     |████████████████████████████████| 146 kB 749 kB/s
Collecting dockerpty<1,>=0.4.1
  Downloading https://www.piwheels.org/simple/dockerpty/dockerpty-0.4.1-py3-none-any.whl (16 kB)
Collecting docopt<1,>=0.6.1
  Downloading https://www.piwheels.org/simple/docopt/docopt-0.6.2-py2.py3-none-any.whl (13 kB)
Collecting jsonschema<4,>=2.5.1
  Downloading https://www.piwheels.org/simple/jsonschema/jsonschema-3.2.0-py2.py3-none-any.whl (56 kB)
     |████████████████████████████████| 56 kB 450 kB/s
Requirement already satisfied: setuptools in /usr/local/lib/python3.7/dist-packages (from jsonschema<4,>=2.5.1->docker-compose) (51.1.0)
Collecting attrs>=17.4.0
  Downloading https://www.piwheels.org/simple/attrs/attrs-20.3.0-py2.py3-none-any.whl (49 kB)
     |████████████████████████████████| 49 kB 345 kB/s
Collecting paramiko>=2.4.2
  Downloading https://www.piwheels.org/simple/paramiko/paramiko-2.7.2-py2.py3-none-any.whl (206 kB)
     |████████████████████████████████| 206 kB 932 kB/s
Collecting bcrypt>=3.1.3
  Downloading https://www.piwheels.org/simple/bcrypt/bcrypt-3.2.0-cp37-cp37m-linux_armv7l.whl (57 kB)
     |████████████████████████████████| 57 kB 466 kB/s
Collecting cffi>=1.1
  Downloading https://www.piwheels.org/simple/cffi/cffi-1.14.4-cp37-cp37m-linux_armv7l.whl (346 kB)
     |████████████████████████████████| 346 kB 506 kB/s
Collecting cryptography>=2.5
  Downloading https://www.piwheels.org/simple/cryptography/cryptography-3.3.1-cp37-cp37m-linux_armv7l.whl (782 kB)
     |████████████████████████████████| 782 kB 1.0 MB/s
Collecting pynacl>=1.0.1
  Downloading https://www.piwheels.org/simple/pynacl/PyNaCl-1.4.0-cp37-cp37m-linux_armv7l.whl (605 kB)
     |████████████████████████████████| 605 kB 902 kB/s
Collecting pyrsistent>=0.14.0
  Downloading https://www.piwheels.org/simple/pyrsistent/pyrsistent-0.16.0-cp37-cp37m-linux_armv7l.whl (100 kB)
     |████████████████████████████████| 100 kB 1.8 MB/s
Collecting python-dotenv<1,>=0.13.0
  Downloading https://www.piwheels.org/simple/python-dotenv/python_dotenv-0.15.0-py2.py3-none-any.whl (18 kB)
Collecting PyYAML<6,>=3.10
  Downloading https://www.piwheels.org/simple/pyyaml/PyYAML-5.3.1-cp37-cp37m-linux_armv7l.whl (44 kB)
     |████████████████████████████████| 44 kB 781 kB/s
Collecting requests<3,>=2.20.0
  Downloading https://www.piwheels.org/simple/requests/requests-2.25.1-py2.py3-none-any.whl (61 kB)
     |████████████████████████████████| 61 kB 1.4 MB/s
Collecting certifi>=2017.4.17
  Downloading https://www.piwheels.org/simple/certifi/certifi-2020.12.5-py2.py3-none-any.whl (147 kB)
     |████████████████████████████████| 147 kB 868 kB/s
Collecting chardet<5,>=3.0.2
  Downloading https://www.piwheels.org/simple/chardet/chardet-4.0.0-py2.py3-none-any.whl (178 kB)
     |████████████████████████████████| 178 kB 857 kB/s
Collecting idna<3,>=2.5
  Downloading https://www.piwheels.org/simple/idna/idna-2.10-py2.py3-none-any.whl (58 kB)
     |████████████████████████████████| 58 kB 1.4 MB/s
Collecting six>=1.4.0
[... 4 lines cut ...]
  Downloading https://www.piwheels.org/simple/urllib3/urllib3-1.26.2-py2.py3-none-any.whl (136 kB)
     |████████████████████████████████| 136 kB 1.1 MB/s
Collecting websocket-client<1,>=0.32.0
  Downloading https://www.piwheels.org/simple/websocket-client/websocket_client-0.57.0-py2.py3-none-any.whl (200 kB)
     |████████████████████████████████| 200 kB 3.4 MB/s
Collecting importlib-metadata
  Downloading https://www.piwheels.org/simple/importlib-metadata/importlib_metadata-3.3.0-py3-none-any.whl (10 kB)
Collecting typing-extensions>=3.6.4
  Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-3.7.4.3-py3-none-any.whl (22 kB)
Collecting zipp>=0.5
  Downloading https://www.piwheels.org/simple/zipp/zipp-3.4.0-py3-none-any.whl (5.2 kB)
Collecting pycparser
  Downloading https://www.piwheels.org/simple/pycparser/pycparser-2.20-py2.py3-none-any.whl (112 kB)
     |████████████████████████████████| 112 kB 3.1 MB/s
Installing collected packages: pycparser, urllib3, six, idna, chardet, cffi, certifi, zipp, websocket-client, typing-extensions, requests, pynacl, cryptography, bcrypt, pyrsistent, paramiko, importlib-metadata, docker, attrs, texttable, PyYAML, python-dotenv, jsonschema, docopt, dockerpty, distro, cached-property, docker-compose
[... cut ...] 1.5.0 docker-4.4.1 docker-compose-1.27.4 dockerpty-0.4.1 docopt-0.6.2 idna-2.10 importlib-metadata-3.3.0 jsonschema-3.2.0 paramiko-2.7.2 pycparser-2.20 pynacl-1.4.0 pyrsistent-0.16.0 python-dotenv-0.15.0 requests-2.25.1 six-1.15.0 texttable-1.6.3 typing-extensions-3.7.4.3 urllib3-1.26.2 websocket-client-0.57.0 zipp-3.4.0

real    0m30.268s
user    0m17.245s
sys     0m1.202s
root@DietPi3:~#
```
</details>

Question is, if we still need to install build dependencys `make gcc libffi-dev libssl-dev`

MichaIng, 2020-12-28:
Awesome. I wonder why it is not selected automatically as source, AFAIK it was, at least I remember certain bugs with the piwheels.org repo when Python 3 wheels were pulled by pip2 (resolved now). Looks like we should apply that source then for all ARMv6+ARMv7 boards, maybe there is a pip config for that. Who knows how much time was already waste to install dependencies and build wheels for modules which were pre-build on piwheels.org for a long time 😄.

> Question is, if we still need to install build dependencys

Your log does not show a single source build but all were pre-built wheels, hence no build tools required 👍.

Joulinar, 2020-12-28:
> You should have a file at /etc/pip.conf containing:

```
[global]
extra-index-url=https://www.piwheels.org/simple
```

> If` that's there, pip should use piwheels. If not, create the file and add those lines to it.
Note that pip version 9 or above is required.

https://www.piwheels.org/faq.html

Using option `-i` allow pip to fall back to PyPI if the requested package (or one of its dependencies) is not available on piwheels

Joulinar, 2020-12-28:
Did one more test on my RPi B (without make dependency). Installation time is down to 2m35.960s from 160m24.768s. Even faster than the RPi4B now 🤣

MichaIng, 2020-12-28:
I'm still wondering why piwheels was already used as source in the past, like here: #3346
This reminds me of the fact that piwheels.org supports Python 3 only, hence we need to make sure that `pip2` does not use it when creating that config file. Is far as I understood piwheels.org has not some check integrated to not ship a Python 3 wheel for a Python 2 instance, so it should be skipped automatically, but better to test it.

Joulinar, 2020-12-28:
why not staying with option `-i`? Means, no need to create config files and we are in control to use pip3. Otherwise we could run into issues for user who have Python2 still installed and try doing `pip install.`

MichaIng, 2020-12-28:
On the other hand we suggest `pip` calls to update software as well, and telling user to use that option on ARMv6 and ARMv7, else not, makes is a bid complicated. If there is already such a repository, it makes sense to use it by default for all pip calls where possible. If something is not available in this repo, PyPI is still used, if I understand correctly, so it should be safe. Just pip2 needs to be tested. Will run some tests.

Joulinar, 2020-12-28:
so we need to find a software not being available on `piwheels` to test fallback to `PyPI`

MichaIng, 2020-12-28:
You see the fallback already, `pythonhosted.org` is the hosting for PyPI. But I'll test it on RPi now, non-available package might behave different than non-matching architecture.

MichaIng, 2020-12-28:
Just installed OctoPrint on RPi without a single source build:
<details>

```
root@Building-ARMv6-Buster:~# pip install octoprint
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting octoprint
  Downloading https://www.piwheels.org/simple/octoprint/OctoPrint-1.5.2-py2.py3-none-any.whl (3.8 MB)
     |████████████████████████████████| 3.8 MB 258 kB/s
Collecting rsa==4.0
  Downloading https://www.piwheels.org/simple/rsa/rsa-4.0-py2.py3-none-any.whl (38 kB)
Collecting sarge==0.1.5post0
  Downloading https://www.piwheels.org/simple/sarge/sarge-0.1.5.post0-py3-none-any.whl (19 kB)
Collecting tornado==5.1.1
  Downloading https://www.piwheels.org/simple/tornado/tornado-5.1.1-cp37-cp37m-linux_armv7l.whl (463 kB)
     |████████████████████████████████| 463 kB 1.0 MB/s
Collecting watchdog==0.10.4
  Downloading https://www.piwheels.org/simple/watchdog/watchdog-0.10.4-py3-none-any.whl (76 kB)
     |████████████████████████████████| 76 kB 1.4 MB/s
Collecting blinker<2,>=1.4
  Downloading https://www.piwheels.org/simple/blinker/blinker-1.4-py3-none-any.whl (14 kB)
Collecting cachelib<1,>=0.1
  Downloading https://www.piwheels.org/simple/cachelib/cachelib-0.1.1-py3-none-any.whl (13 kB)
Collecting Click<8,>=7.1.2
  Downloading https://www.piwheels.org/simple/click/click-7.1.2-py2.py3-none-any.whl (84 kB)
     |████████████████████████████████| 84 kB 1.0 MB/s
Collecting emoji<1,>=0.5.4
  Downloading https://www.piwheels.org/simple/emoji/emoji-0.6.0-py3-none-any.whl (50 kB)
     |████████████████████████████████| 50 kB 1.8 MB/s
Collecting feedparser<7,>=6.0.2
  Downloading https://www.piwheels.org/simple/feedparser/feedparser-6.0.2-py3-none-any.whl (80 kB)
     |████████████████████████████████| 80 kB 2.3 MB/s
Collecting filetype<2,>=1.0.7
  Downloading https://www.piwheels.org/simple/filetype/filetype-1.0.7-py2.py3-none-any.whl (15 kB)
Collecting flask<2,>=1.1.2
  Downloading https://www.piwheels.org/simple/flask/Flask-1.1.2-py2.py3-none-any.whl (94 kB)
     |████████████████████████████████| 94 kB 1.1 MB/s
Collecting Flask-Assets<3,>=2.0
  Downloading https://www.piwheels.org/simple/flask-assets/Flask_Assets-2.0-py3-none-any.whl (8.5 kB)
Collecting Flask-Babel<2,>=1.0
  Downloading https://www.piwheels.org/simple/flask-babel/Flask_Babel-1.0.0-py3-none-any.whl (9.5 kB)
Collecting Babel>=2.3
  Downloading https://www.piwheels.org/simple/babel/Babel-2.9.0-py2.py3-none-any.whl (8.8 MB)
     |████████████████████████████████| 8.8 MB 3.9 MB/s
Collecting Flask-Login<0.6,>=0.5
  Downloading https://www.piwheels.org/simple/flask-login/Flask_Login-0.5.0-py2.py3-none-any.whl (16 kB)
Collecting frozendict<2,>=1.2
  Downloading https://www.piwheels.org/simple/frozendict/frozendict-1.2-py3-none-any.whl (3.6 kB)
Collecting future<1,>=0.18.2
  Downloading https://www.piwheels.org/simple/future/future-0.18.2-py3-none-any.whl (491 kB)
     |████████████████████████████████| 491 kB 1.1 MB/s
Collecting itsdangerous<2,>=1.1.0
  Downloading https://www.piwheels.org/simple/itsdangerous/itsdangerous-1.1.0-py2.py3-none-any.whl (16 kB)
Collecting Jinja2<3,>=2.11.2
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-2.11.2-py2.py3-none-any.whl (127 kB)
     |████████████████████████████████| 127 kB 2.6 MB/s
Collecting markdown<3.2,>=3.1
  Downloading https://www.piwheels.org/simple/markdown/Markdown-3.1.1-py2.py3-none-any.whl (87 kB)
     |████████████████████████████████| 87 kB 1.8 MB/s
Requirement already satisfied: setuptools>=36 in /usr/local/lib/python3.7/dist-packages (from markdown<3.2,>=3.1->octoprint) (51.0.0)
Collecting markupsafe<2.0,>=1.1
  Downloading https://www.piwheels.org/simple/markupsafe/MarkupSafe-1.1.1-cp37-cp37m-linux_armv7l.whl (33 kB)
Collecting netaddr<1,>=0.7.19
  Downloading https://www.piwheels.org/simple/netaddr/netaddr-0.8.0-py3-none-any.whl (1.9 MB)
     |████████████████████████████████| 1.9 MB 3.5 MB/s
Collecting netifaces<1,>=0.10.9
  Downloading https://www.piwheels.org/simple/netifaces/netifaces-0.10.9-cp37-cp37m-linux_armv7l.whl (38 kB)
Collecting OctoPrint-FileCheck>=2020.08.07
  Downloading https://www.piwheels.org/simple/octoprint-filecheck/OctoPrint_FileCheck-2020.8.7-py2.py3-none-any.whl (19 kB)
Collecting OctoPrint-FirmwareCheck>=2020.09.23
[... 52 lines cut ...]
  Downloading https://www.piwheels.org/simple/websocket-client/websocket_client-0.57.0-py2.py3-none-any.whl (200 kB)
     |████████████████████████████████| 200 kB 2.5 MB/s
Collecting werkzeug<2,>=1.0.1
  Downloading https://www.piwheels.org/simple/werkzeug/Werkzeug-1.0.1-py2.py3-none-any.whl (298 kB)
     |████████████████████████████████| 298 kB 1.1 MB/s
Collecting wrapt<2,>=1.12.1
  Downloading https://www.piwheels.org/simple/wrapt/wrapt-1.12.1-cp37-cp37m-linux_armv7l.whl (68 kB)
     |████████████████████████████████| 68 kB 802 kB/s
Collecting zeroconf<0.25,>=0.24
  Downloading https://www.piwheels.org/simple/zeroconf/zeroconf-0.24.5-py3-none-any.whl (65 kB)
     |████████████████████████████████| 65 kB 1.3 MB/s
Collecting ifaddr
  Downloading https://www.piwheels.org/simple/ifaddr/ifaddr-0.1.7-py2.py3-none-any.whl (10 kB)
Collecting sgmllib3k
  Downloading https://www.piwheels.org/simple/sgmllib3k/sgmllib3k-1.0.0-py3-none-any.whl (7.1 kB)
Collecting six
  Downloading https://www.piwheels.org/simple/six/six-1.15.0-py2.py3-none-any.whl (10 kB)
[... cut ...] t-0.57.0 werkzeug-1.0.1 wrapt-1.12.1 zeroconf-0.24.5
```
</details>

The same with `pip2` correctly pulls everything from pythonhosted.org, including sources in some cases, doing time-consuming compiling 😄.

I'd say it's safe to apply this on all ARMv6 + ARMv7 when installing Python 3 pip, then we can as well go through some software installs and likely are able to remove build dependencies in most cases. Would be awesome if there was some resource for arm64 wheels, but so far couldn't find any.

Joulinar, 2020-12-28:
Yes it could save a lot of time for our users doing this change. I guess arm64 will follow is soon as RPi OS 64bit will reach stable status.

MichaIng, 2020-12-28:
Yes

Joulinar, 2020-12-28:
how did you config file looks like?

MichaIng, 2020-12-28:
Just your suggestion (=the one suggested by piwheels.org): https://github.com/MichaIng/DietPi/issues/3078#issuecomment-751819418

Joulinar, 2020-12-28:
ok I thought it could prevent still using `pypi.org`.

MichaIng, 2020-12-28:
As the settings name says, it's an "extra" index, no override 😃.

wisehackermonkey, 2020-12-29:
bump would love to see this added!

wisehackermonkey, 2020-12-29:
I'm running dietpi on rpi 3,
the 
>pip3 install docker-compose -i https://www.piwheels.org/simple
didnt work for me. (sadface emoji)
pip2 install took too long and i gave up.
Ended up going with apt install instead. :/
What did work for me was 
```bash
apt update
apt install docker-compose
```

Joulinar, 2020-12-29:
On what is the issue using `pip3 install docker-compose -i https://www.piwheels.org/simple`? Can you share the error message? Are you running 64 or 32bit OS?

wisehackermonkey, 2020-12-30:
> And what is the issue using `pip3 install docker-compose -i https://www.piwheels.org/simple`? Can you share the error message? Are you running 64 or 32bit OS?

sorry for being vauge
```
 uname -a
Linux dietpi3 5.4.72-v8+ #1356 SMP PREEMPT Thu Oct 22 13:58:52 BST 2020 aarch64 GNU/Linux
Dietpi rpi 3, 32bit.
```
I think i needed to apt update inorder for it to work.

Joulinar, 2020-12-30:
looks like 64Bit as your kernel is `aarch64`. On a 64bit system `pip3 install docker-compose -i https://www.piwheels.org/simple` will not work as currently `piwheels` did not support Raspberry OS 64bit BETA.

There you have 2 options

1. wait for `pip3` finish compiling up-to-date docker compose (around 30 minutes)
2. use `apt` to install 2.5 years old docker compose

wisehackermonkey, 2021-01-03:
thanks @Joulinar 
Thanks for the support! 
ill have to try V next time...
```bash
pip3 install docker-compose -i https://www.piwheels.org/simple
```

here's what i ended up doing in the end
```bash
# [Install Docker on Raspberry Pi: Installing Docker and Docker Compose on the Raspberry Pi in 5 Simple Steps - DEV](https://dev.to/rohansawant/installing-docker-and-docker-compose-on-the-raspberry-pi-in-5-simple-steps-3mgl)
sudo apt-get install -y libffi-dev libssl-dev
sudo apt-get install -y python3 python3-pip python3-setuptools

#check if python3 installed
type -P python3 >/dev/null 2>&1 && echo Python 3 is installed 

# actuall install docker compose
pip3 install docker-compose -i https://www.piwheels.org/simple

docker-compose -v 
```

Joulinar, 2021-01-03:
Basically Python could have been installed using `dietpi-software`
`libffi-dev libssl-dev` should not be required as using `-i https://www.piwheels.org/simple` will pull already compiled modules.

wisehackermonkey, 2021-02-22:
# so would this work?
```
# install python3 dependencies
dietpi-software libffi-dev libssl-dev

#check if python3 installed
python3 -V

# actuall install docker compose
pip3 install docker-compose 

docker-compose -v 
```

Joulinar, 2021-02-22:
no that's not how it is working

it depends if you are running 32 or 64bit OS version as this would make a difference in packages needed  😉 
As an alternative, you could try out our new BETA 7.0. This would include Docker-Compose and would allow installation via `dietpi-software`

wisehackermonkey, 2021-02-25:
# two things:
#### 1) How do i get my hands on `BETA 7.0`?

#### 2) side note:
oh yeah now that i think about it this is wrong `dietpi-software libffi-dev libssl-dev`

I think it done by package numbers internally?
like: 
```bash
dietpi-software 73
Note this is a arbitray number for example only
```
link.
https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software

Joulinar, 2021-02-25:
Beta can be applied as follow. https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

closed 2021-02-28 by MichaIng