MichaIng/DietPi #3079: Dietpi-Banner | Show Nord VPN connection state

svh1985, 2019-08-26:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Not really, would be a nice improvement tho

#### Describe the solution you'd like:

- To make the NordVPN implementation complete, it would be nice to show the connection state in the Dietpi-Banner
- Have the ability to add it to the list of items printed when loggin into dietpi

#### Additional context

- I think we can reuse some code from the dietpi-nordvpn script itself

MichaIng, 2019-08-27:
@svh1985 
Great idea. Here is the part of the scripts that estimates the in-menu state info: https://github.com/MichaIng/DietPi/blob/dev/dietpi/misc/dietpi-nordvpn#L160-L187

Since WAN IP is a separate entry, `connected: <server>`/`disconnected: <server>`/`not configured: <subscription URL>` + probably RX/TX traffic in a separate line could be shown?

You might want to play around with the custom banner entry for now, e.g. create a script that creates the output line/lines. We can add a function/input mode to `dietpi-nordvpn` script as well, which only prints the current status instead of starting "G"UI.

MichaIng, 2019-08-31:
Ref: https://packages.debian.org/buster/apparmor-utils

PR #3084 (2019-09-08): NordVPN improvements | Change up and down script and add item to DietPi-Banner
**Status**:  Testing
- [x] Change up and down to run on route change
- [x] Add NordVPN entry to DietPi-Banner
- [x] Add function to NordVPN to pass status to DietPi-Banner 

Fixes: #3079

**Commit list/description**:
- DietPi-Nordvpn | Changed up to route-up and down to route-pre-down. This will ensure that the scripts will only run after the VPN successfully connected.
closed 2019-09-10 by svh1985