MichaIng/DietPi #3082: Buster: upgrade from Stretch does not enable Buster features like AppArmor?

GvY85, 2019-08-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.25.x
- Distro version | Buster
- Kernel version | ?
- SBC device | Native PC (UEFI) -> Z83-II mini pc

#### Additional Information (if applicable)
- Software title | All
- Was the software title installed freshly or updated/migrated? Migrated from Stretch
- Can this issue be replicated on a fresh installation of DietPi? Have not tried that yey

#### Steps to reproduce
Migrate from Stretch to Buster by updating sources and running apt-get update && apt-get upgrade && apt-get dist-upgrade

#### Expected behaviour
You should end up with a full fletched Buster install with all new features such as AppArmor enabled?

#### Actual behaviour
It seems that if i upgrade from an excisting Dietpi Stretch installation to Buster I dont get Buster features like AppArmor installed/enabled by default like it should with Buster? Is this on purpose or a bug? 
And what happens when or do I need to install a fresh Buster using 
https://dietpi.com/downloads/images/DietPi_NativePC-UEFU-x86_64-Buster.7z?

MichaIng, 2019-08-31:
@GvY85 
Many thanks for your report.

DietPi is no fully fledged Debian Buster, but it comes very lightweight, so that you need to install certain features, like AppArmor, manually. But that should be easy via APT and we might consider to add it to DietPi-Software as well, since some things need to be adjusted, to e.g  allowing a symlink for MariaDB database files to dietpi_userdata for easier external drive transfer.

GvY85, 2019-08-31:
Aha, so it is a deliberate thing. Good to know.
Then perhaps indeed it would be nice to add it to Dietpi-Software. It seems like a nice feature

GvY85, 2019-08-31:
On the other hand: since it is automatically turned on on a regular or minimal Buster and seem lightweight perhaps it should be default on with Dietpi as well? To not skimp on security?

MichaIng, 2019-08-31:
To collect some info: https://packages.debian.org/buster/apparmor-utils
- Available on all distro versions
- Depends on Python 3
- Known is that builtin MariaDB rules need to be adjusted to allow database symlink which we place by default. Other software titles might need rework as well, either the way we install it or the AppArmor rules.

I am still not 100% sure about doubled features with systemd access and security settings. Once can limit access to certain general or special directories, on file-access basis via run user anyway and limit kernel/device capabilities as well. So at first we will ship it as optional install only, marked as experimental/beta to allow intensive testing with out software installs, before shipping it as fully DietPi compatible. Only if there is any real benefit (security-wise) over simply using stricter systemd unit settings, we might implement it as regular part of DietPi images.

mrbluecoat, 2021-06-12:
Not sure if I'm missing something obvious:

```
$ apt install -y apparmor apparmor-profiles apparmor-utils apparmor-profiles apparmor-profiles-extra
$ aa-status
apparmor module is loaded.
apparmor filesystem is not mounted.
```

Mount message persists after reboot.

mrbluecoat, 2021-06-12:
Tried a random guess but it didn't work:

```
cat > boot.conf <<EOF
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
APPARMOR=1
SECURITY=apparmor
EOF

rpi-eeprom-config --apply boot.conf
reboot
```

MichaIng, 2026-10-27:
Kernel-wise, AppArmor can be explicitly enabled via `apparmor=1` kernel command-line argument, e.g. in `/boot/firmware/cmdline.txt` on RPi. But it should be actually enabled OOTB in any modern distro version.