MichaIng/DietPi #3083: Dietpi-NordVPN: randomly connnect to different server on a user settable time interval

GvY85, 2019-08-30:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
When using Dietpi-NordVPN you can only select one server and you have to stick with that one unless you manually run Dietpi-NordVPN again and select a different server.

#### Describe the solution you'd like:
For me it would be great if you could select a range of servers from the list and set an interval so that it connects randomly to one of them at a set interval (reboot/1h/1d/etc).
I like to change servers once in a while for more anonimity and now the only way to do it is manually.
Next step could be that Dietpi-NordVPN automatically selects a new server if the current one is/gets down and/or test the servers initially selected to see if they are up before connecting.

#### Describe alternatives you've considered:
Curently i am using a script that randomly selects a .ovpn file from the DietPi-NordVPN directory and copies it to a new file name "my random server.ovpn".
I call this script every ones in a while with Cron and select "my random server.opvn" in the serverlist of DietPi-NordVPN to connect to. Then I reboot my server once a week so at reboot i have a new random server to connect to.
This is all very hacked together and not really nice. Would be much greater to have it integrated intp DietPi

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2019-08-31:
@GvY85 
Many thanks for your request.

Indeed sounds reasonable. So a list of severs can be chosen and js cycled through on every connect and/or time based. But it should not change during one session, I think?
Also if one server fails, it could try the next until connection succeeds. Although probably it should be checked if it is due to connection issue or authentication (wrong/invalid user/password) to not trigger some possible brute-force protection mechanism.

About the selection, easy would be to have a checkbox selection menu, a bid more effort wound be to allow entering e.g. country/countries and an index range.

So far my thought.

And btw. you don't require a dedicated account for FeatHub, you can login via your GitHub account. This is even the only possible way, AFAIK.

GvY85, 2019-08-31:
Exactly my thoughts, sounds good to me :).

Ill see if i get it up on Feathub