MichaIng/DietPi #3085: Dietpi software installation for Sonarr fails

anachronox99, 2019-08-31:
#### Details:
- Date           | Sat 31 Aug 10:26:04 AEST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1253 SMP Thu Aug 15 11:49:46 BST 2019
- Distro         | buster (index=5)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
Launch dietpi-software
Try to install sonarr. 
After about 3 minutes this error message displays and the program prompts me to raise an issue

#### Expected behaviour:
The software should install successfully

#### Actual behaviour:
Doesn't install. See logs

#### Additional logs:
```
Log file contents:
Hit:1 https://download.mono-project.com/repo/debian raspbianstretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:3 https://apt.sonarr.tv develop InRelease [13.3 kB]
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Err:3 https://apt.sonarr.tv develop InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
Reading package lists...
W: GPG error: https://apt.sonarr.tv develop InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
E: The repository 'https://apt.sonarr.tv develop InRelease' is not signed.
```

--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

anachronox99, 2019-08-31:
More information. If I try to install anything else, I now also get the same failure message. It's still trying to install Sonarr whether I try to or not, so I'm effectively blocked from installing anything

MichaIng, 2019-08-31:
@anachronox99 
Many thanks for your report. Please see for solution: https://github.com/MichaIng/DietPi/issues/3029

anachronox99, 2019-08-31:
Thanks - this works.

closed 2019-08-31 by anachronox99