MichaIng/DietPi #3088: upgrade dietpi

dnj0104, 2019-09-01:
#### Details:
- Date           | Sun  1 Sep 11:07:40 BST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1253 SMP Thu Aug 15 11:49:46 BST 2019
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:
Trying to update dietpi
1. ...
2. ...

#### Expected behaviour:
Just a update
- ...

#### Actual behaviour:
updates stopped
- ...

#### Extra details:
maybe sdcard is damaged,  dietpi has been running on same sdcard couple of months
with no problem until now 
- ...

#### Additional logs:
```
Log file contents:
Setting up libssl1.0.2:armhf (1.0.2s-1~deb9u1) ...
Unrecognized character ▒; marked by <-- HERE after postderef'<-- HERE near column 30 at /usr/share/perl/5.24/warnings.pm line 113.
Compilation failed in require at /usr/share/perl/5.24/warnings/register.pm line 4.
Compilation failed in require at /usr/share/perl/5.24/vars.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl/5.24/vars.pm line 7.
Compilation failed in require at /usr/share/perl/5.24/base.pm line 5.
BEGIN failed--compilation aborted at /usr/share/perl/5.24/base.pm line 5.
Compilation failed in require at /usr/share/perl5/Debconf/Log.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Log.pm line 7.
Compilation failed in require at /usr/share/perl5/Debconf/Db.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Db.pm line 7.
Compilation failed in require at /usr/share/debconf/frontend line 6.
BEGIN failed--compilation aborted at /usr/share/debconf/frontend line 6.
dpkg: error processing package libssl1.0.2:armhf (--configure):
 subprocess installed post-installation script returned error exit status 255
Setting up libssl1.1:armhf (1.1.0k-1~deb9u1) ...
Unrecognized character ▒; marked by <-- HERE after postderef'<-- HERE near column 30 at /usr/share/perl/5.24/warnings.pm line 113.
Compilation failed in require at /usr/share/perl/5.24/warnings/register.pm line 4.
Compilation failed in require at /usr/share/perl/5.24/vars.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl/5.24/vars.pm line 7.
Compilation failed in require at /usr/share/perl/5.24/base.pm line 5.
BEGIN failed--compilation aborted at /usr/share/perl/5.24/base.pm line 5.
Compilation failed in require at /usr/share/perl5/Debconf/Log.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Log.pm line 7.
Compilation failed in require at /usr/share/perl5/Debconf/Db.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Db.pm line 7.
Compilation failed in require at /usr/share/debconf/frontend line 6.
BEGIN failed--compilation aborted at /usr/share/debconf/frontend line 6.
dpkg: error processing package libssl1.1:armhf (--configure):
 subprocess installed post-installation script returned error exit status 255
dpkg: dependency problems prevent configuration of openssl:
 openssl depends on libssl1.1 (>= 1.1.0); however:
  Package libssl1.1:armhf is not configured yet.

dpkg: error processing package openssl (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of proftpd-basic:
 proftpd-basic depends on libssl1.0.2 (>= 1.0.2d); however:
  Package libssl1.0.2:armhf is not configured yet.

dpkg: error processing package proftpd-basic (--configure):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Errors were encountered while processing:
 tzdata
 libssl1.0.2:armhf
 libssl1.1:armhf
 openssl
 proftpd-basic
E: Sub-process /usr/bin/dpkg returned an error code (1)

MichaIng, 2019-09-01:
@dnj0104 
Looks like file corruption. I had this myself two times within the perl module files...
Could you paste: `ls -Al /usr/share/perl/5.24/`

dnj0104, 2019-09-01:
here it is .

dietpi@DietPi:~$ ls -Al /usr/share/perl/5.24/
total 1500
-rw-r--r--  1 root root   2618 Nov 29  2018 AnyDBM_File.pm
drwxr-xr-x  3 root root   4096 May  1 18:49 App
drwxr-xr-x  3 root root   4096 May  1 18:49 Archive
drwxr-xr-x  2 root root   4096 May  1 18:49 Attribute
drwxr-xr-x  4 root root   4096 May  1 18:49 autodie
-rw-r--r--  1 root root  12886 Nov 29  2018 autodie.pm
-rw-r--r--  1 root root   5487 Nov 29  2018 AutoLoader.pm
-rw-r--r--  1 root root  19637 Nov 29  2018 AutoSplit.pm
-rw-r--r--  1 root root   4238 Nov 29  2018 autouse.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 B
-rw-r--r--  1 root root   8931 Nov 29  2018 base.pm
-rw-r--r--  1 root root  31025 Nov 29  2018 Benchmark.pm
-rw-r--r--  1 root root  22957 Nov 29  2018 bigint.pm
-rw-r--r--  1 root root  20200 Nov 29  2018 bignum.pm
-rw-r--r--  1 root root  15646 Nov 29  2018 bigrat.pm
-rw-r--r--  1 root root   2086 Nov 29  2018 blib.pm
-rw-r--r--  1 root root    758 Nov 29  2018 bytes_heavy.pl
-rw-r--r--  1 root root    447 Nov 29  2018 bytes.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 Carp
-rw-r--r--  1 root root  20138 Nov 29  2018 Carp.pm
-rw-r--r--  1 root root  33484 Nov 29  2018 _charnames.pm
-rw-r--r--  1 root root  20989 Nov 29  2018 charnames.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 Class
drwxr-xr-x  2 root root   4096 May  1 18:49 Compress
drwxr-xr-x  3 root root   4096 May  1 18:49 Config
-rw-r--r--  1 root root   5737 Nov 29  2018 constant.pm
-rw-r--r--  1 root root   3267 Nov 29  2018 CORE.pod
drwxr-xr-x 10 root root   4096 May  1 18:49 CPAN
-rw-r--r--  1 root root 140192 Nov 29  2018 CPAN.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 DBM_Filter
-rw-r--r--  1 root root  14385 Nov 29  2018 DBM_Filter.pm
-rw-r--r--  1 root root  18922 Nov 29  2018 DB.pm
-rw-r--r--  1 root root   3510 Nov 29  2018 deprecate.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 Devel
-rw-r--r--  1 root root  18696 Nov 29  2018 diagnostics.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 Digest
-rw-r--r--  1 root root  10701 Nov 29  2018 Digest.pm
-rw-r--r--  1 root root   1556 Nov 29  2018 DirHandle.pm
-rw-r--r--  1 root root  17556 Nov 29  2018 Dumpvalue.pm
-rw-r--r--  1 root root  15555 Nov 29  2018 dumpvar.pl
drwxr-xr-x  2 root root   4096 May  1 18:49 Encode
drwxr-xr-x  2 root root   4096 May  1 18:49 encoding
-rw-r--r--  1 root root   4761 Nov 29  2018 English.pm
-rw-r--r--  1 root root   5524 Nov 29  2018 Env.pm
-rw-r--r--  1 root root   5795 Nov 29  2018 experimental.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 Exporter
-rw-r--r--  1 root root   2367 Nov 29  2018 Exporter.pm
drwxr-xr-x  9 root root   4096 May  1 18:49 ExtUtils
-rw-r--r--  1 root root  58280 Nov 29  2018 Fatal.pm
-rw-r--r--  1 root root   4313 Nov 29  2018 feature.pm
-rw-r--r--  1 root root   5022 Nov 29  2018 fields.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 File
-rw-r--r--  1 root root   5572 Nov 29  2018 FileCache.pm
-rw-r--r--  1 root root   2105 Nov 29  2018 FileHandle.pm
-rw-r--r--  1 root root   4003 Nov 29  2018 filetest.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 Filter
-rw-r--r--  1 root root   4561 Nov 29  2018 FindBin.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 Getopt
drwxr-xr-x  2 root root   4096 May  1 18:49 HTTP
drwxr-xr-x  3 root root   4096 May  1 18:49 I18N
-rw-r--r--  1 root root   3340 Nov 29  2018 if.pm
-rw-r--r--  1 root root    172 Nov 29  2018 integer.pm
drwxr-xr-x  5 root root   4096 May  1 18:49 IO
drwxr-xr-x  2 root root   4096 May  1 18:49 IPC
drwxr-xr-x  3 root root   4096 May  1 18:49 JSON
-rw-r--r--  1 root root   3204 Nov 29  2018 less.pm
[... 31 lines cut ...]
-rw-r--r--  1 root root    848 Nov 29  2018 subs.pm
-rw-r--r--  1 root root   2099 Nov 29  2018 Symbol.pm
drwxr-xr-x  5 root root   4096 May  1 18:49 TAP
drwxr-xr-x  2 root root   4096 May  1 18:49 Term
drwxr-xr-x  5 root root   4096 May  1 18:49 Test
-rw-r--r--  1 root root  30052 Nov 29  2018 Test.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 Text
drwxr-xr-x  2 root root   4096 May  1 18:49 Thread
-rw-r--r--  1 root root   8287 Nov 29  2018 Thread.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 Tie
drwxr-xr-x  2 root root   4096 May  1 18:49 Time
drwxr-xr-x  3 root root   4096 May  1 18:49 Unicode
drwxr-xr-x  4 root root   4096 May  1 18:49 unicore
-rw-r--r--  1 root root   6594 Nov 29  2018 UNIVERSAL.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 User
-rw-r--r--  1 root root  31615 Nov 29  2018 utf8_heavy.pl
-rw-r--r--  1 root root    342 Nov 29  2018 utf8.pm
-rw-r--r--  1 root root   1149 Nov 29  2018 vars.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 version
-rw-r--r--  1 root root   1715 Nov 29  2018 version.pm
-rw-r--r--  1 root root   9860 Nov 29  2018 version.pod
-rw-r--r--  1 root root   4313 Nov 29  2018 vmsish.pm
drwxr-xr-x  2 root root   4096 May  1 18:49 warnings
-rw-r--r--  1 root root  21639 Nov 29  2018 warnings.pm
-rw-r--r--  1 root root   3899 Nov 29  2018 XSLoader.pm

MichaIng, 2019-09-01:
@dnj0104 
Okay meta data are fine, and it is only this `warnings.pm` file that contains an unrecognised character. I downloaded the `perl-modules-5.24_5.24.1-3+deb9u5_all.deb` package from https://archive.raspbian.org/raspbian/pool/main/p/perl/, extracted the `/usr/share/perl/5.24.1/warnings.pm`, and found the line that contains a `<tab>` character at the mentioned line 113 column 30:
```
    'experimental::postderef'		=> "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x40\x00\x00\x00", # [55]
```

Usually I would reinstall the APT or DPkg package to repair broken files, but as long as there are unconfigured packages, APT stops with error right from the beginning. So I would manually replace the file, either download/extract it manually, or take it here, as zip since GitHub does not allow upload of `.pm` files 😉: 
[warnings.zip](https://github.com/MichaIng/DietPi/files/3563894/warnings.zip)

dnj0104, 2019-09-02:
replaced the warnings.pm with the zipped one above, and then run command apt -f install

this happend:

`root@DietPi:~#` apt -f install
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
5 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Array found where operator expected at /usr/lib/arm-linux-gnueabihf/perl/5.24/IO                                                                                                                               /Handle.pm line 387, near "prhnt @_9"
        (Do you need to predeclare prhnt?)
debconf: Perl may be unconfigured (Unrecognized character \x08; marked by <-- HE                                                                                                                               RE after        prhnt @_9<-- HERE near column 11 at /usr/lib/arm-linux-gnueabihf                                                                                                                               /perl/5.24/IO/Handle.pm line 387.
Compilation failed in require at /usr/lib/arm-linux-gnueabihf/perl/5.24/IO/Seeka                                                                                                                               ble.pm line 9.
BEGIN failed--compilation aborted at /usr/lib/arm-linux-gnueabihf/perl/5.24/IO/S                                                                                                                               eekable.pm line 9.
Compilation failed in require at /usr/lib/arm-linux-gnueabihf/perl/5.24/IO/File.                                                                                                                               pm line 11.
BEGIN failed--compilation aborted at /usr/lib/arm-linux-gnueabihf/perl/5.24/IO/F                                                                                                                               ile.pm line 11.
Compilation failed in require at /usr/share/perl/5.24/FileHandle.pm line 9.
Compilation failed in require at (eval 1) line 3.
BEGIN failed--compilation aborted at (eval 1) line 3.
) -- aborting
Setting up tzdata (2019b-0+deb9u1) ...
Number found where operator expected at /usr/share/perl/5.24/warnings/register.p                                                                                                                               m line 3, near "$VERSION 5"
        (Missing operator before 5?)
String found where operator expected at /usr/share/perl/5.24/warnings/register.p                                                                                                                               m line 3, near "5 '1.04'"
        (Missing operator before  '1.04'?)
Illegal declaration of subroutine vars::mkMask at /usr/share/perl/5.24/warnings/                                                                                                                               register.pm line 9.
Compilation failed in require at /usr/share/perl/5.24/vars.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl/5.24/vars.pm line 7.
Compilation failed in require at /usr/share/perl/5.24/base.pm line 5.
BEGIN failed--compilation aborted at /usr/share/perl/5.24/base.pm line 5.
Compilation failed in require at /usr/share/perl5/Debconf/Log.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Log.pm line 7.
Compilation failed in require at /usr/share/perl5/Debconf/Db.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Db.pm line 7.
Compilation failed in require at /usr/share/debconf/frontend line 6.
BEGIN failed--compilation aborted at /usr/share/debconf/frontend line 6.
dpkg: error processing package tzdata (--configure):
 subprocess installed post-installation script returned error exit status 255
Setting up libssl1.0.2:armhf (1.0.2s-1~deb9u1) ...
Number found where operator expected at /usr/share/perl/5.24/warnings/register.p                                                                                                                               m line 3, near "$VERSION 5"
        (Missing operator before 5?)
String found where operator expected at /usr/share/perl/5.24/warnings/register.p                                                                                                                               m line 3, near "5 '1.04'"
        (Missing operator before  '1.04'?)
Illegal declaration of subroutine vars::mkMask at /usr/share/perl/5.24/warnings/                                                                                                                               register.pm line 9.
Compilation failed in require at /usr/share/perl/5.24/vars.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl/5.24/vars.pm line 7.
Compilation failed in require at /usr/share/perl/5.24/base.pm line 5.
BEGIN failed--compilation aborted at /usr/share/perl/5.24/base.pm line 5.
Compilation failed in require at /usr/share/perl5/Debconf/Log.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Log.pm line 7.
Compilation failed in require at /usr/share/perl5/Debconf/Db.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Db.pm line 7.
Compilation failed in require at /usr/share/debconf/frontend line 6.
BEGIN failed--compilation aborted at /usr/share/debconf/frontend line 6.
dpkg: error processing package libssl1.0.2:armhf (--configure):
[... cut ...]
dpkg: dependency problems prevent configuration of proftpd-basic:
 proftpd-basic depends on libssl1.0.2 (>= 1.0.2d); however:
  Package libssl1.0.2:armhf is not configured yet.

dpkg: error processing package proftpd-basic (--configure):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Errors were encountered while processing:
 tzdata
 libssl1.0.2:armhf
 libssl1.1:armhf
 openssl
 proftpd-basic
E: Sub-process /usr/bin/dpkg returned an error code 

> (1)

I think I have to run dietpi-recover or something. :)

MichaIng, 2019-09-02:
@dnj0104 
Hmm, can you try: `dpkg-reconfigure perl`

Does `dmesg` actually report any I/O errors? `dmesg -l emerg,alert,crit`

dnj0104, 2019-09-04:
now it looks better, this is my rpi with your `warnings.pm` and after `apt -f install`

`root@DietPi:~# apt -f install
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.`

after that I run `dmesg` and this ; warnings appeard amongst alot of other things

` 2.237930] systemd[1]: systemd-logind.service: Cannot add dependency job, ignoring: Unit systemd-logind.service is masked.
[    2.241999] systemd[1]: getty-static.service: Cannot add dependency job, ignoring: Unit getty-static.service is masked.`

`0.755116] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.755124] [vc_sm_connected_init]: failed to initialize shared memory service
[    0.758416] [vc_sm_connected_init]: end - returning -1`

`9.194306] sd 0:0:0:0: [sda] No Caching mode page found
[    9.194317] sd 0:0:0:0: [sda] Assuming drive cache: write through`

` WARN::dwc_otg_hcd_init:1046: FIQ DMA bounce buffers: virt = 0xbdd04000 dma = 0xfdd04000 len=9024
[    0.676775] FIQ FSM acceleration enabled for :
               Non-periodic Split Transactions
               Periodic Split Transactions
               High-Speed Isochronous Endpoints
               Interrupt/Control Split Transaction hack enabled
[    0.684950] dwc_otg: Microframe scheduler enabled
[    0.684999] WARN::hcd_init_fiq:459: FIQ on core 1 at 0x805ed94c
[    0.686582] WARN::hcd_init_fiq:460: FIQ ASM at 0x805edcb4 length 36
[    0.688122] WARN::hcd_init_fiq:486: MPHI regs_base at 0xf0006000`

some things about the external drive and other things but nothing about the sd-card what I can see,

So I try make an update of dietpi again.

Cheers.
`

dnj0104, 2019-09-04:
update worked now like a charm, BUT, it´s a bit strange, when I tried to do updates the firsts couple of time the onboard wifi was disable, then before the last update, the update that wnt thru I enabled the onboard wifi and that was then it worked, my pi is not connected to my wifi, its only connected with ethernet.

MichaIng, 2019-09-04:
@dnj0104 
Your dmesg entries are btw expected. Those are due to limitations of the graphics driver, when `gpu_mem` is lower then 16M and limitations of the USB driver `dwc_otg`, but nothing that would cause problems AFAIK. Still not sure how those perl error can occur then, since it is obviously not due to file corruption.

> I also tried command dpkg-reconfigure perl after i pressed enter and it took some seconds and I get a new command row. so nothing happend here.

Jep, that is expected to not show output in every case. However you initial error message contained some line that perl package might not be configured, so at least to it a try and check if afterwards apt upgrade works.

Strange that WiFi played any role here, especially since we did not touch this fields with the updates from v6.22 to v6.25, however I'm glad that finallly you have successfully upgraded to v6.25?

dnj0104, 2019-09-05:
Yes, it works great now, thanks for your support, 

Case Closed

closed 2019-09-05 by dnj0104