MichaIng/DietPi #3089: Wont connect to internet

JirkaKrivanek98, 2019-09-01:
#### Required Information:
- DietPi Version | v6.2
- SBC Device     | NanoPi Neo (armv7l) (index=60)
- Distro         | stretch (index=4)
- Command        | Connection test: http://ftp.cn.debian.org/debian/
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2018-03-05 18:09:08--  http://ftp.cn.debian.org/debian/
Resolving ftp.cn.debian.org (ftp.cn.debian.org)... failed: Connection timed out.
wget: unable to resolve host address ‘ftp.cn.debian.org’
```

MichaIng, 2019-09-01:
@JirkaKrivanek98 
Many thanks for your report.

DNS resolving times out, thus takes longer than 10 seconds, where it only should take a fraction of one second.

You can check how long it takes:
```
time wget --spider http://ftp.cn.debian.org/debian/
```

Which DNS server do you use? A custom one or do you use DHCP?

You can increase the timeout to e.g. 20 seconds via:
```
sed -i 's/local timeout=10/local timeout=20/' /DietPi/dietpi/func/dietpi-globals
```
Then rerun `dietpi-software`.

When updated to v6.25, you can adjust this timeout and connection attempts via:
`dietpi-config` > `Network Options: Misc`

JirkaKrivanek98, 2019-09-01:
It wont help. I use DHCP. I am net to this and dont know much what to do.

MichaIng, 2019-09-01:
@JirkaKrivanek98 
Please run the following command from console and paste output here:
```
time wget --spider http://ftp.cn.debian.org/debian/
```

Trunkzeh, 2019-09-02:
@JirkaKrivanek98 Have you confirmed you have internet connectivity to the NanoPi? 

1. Are you connected via ethernet or WIFI? Please output the appropriate ifconfig / iwconfig
2. Are you able to ping 8.8.8.8 / 1.1.1.1 / a suitable DNS ?
3. Are you able to ping google.com / a suitable domain ?

MichaIng, 2019-09-02:
@JirkaKrivanek98 @Trunkzeh 
> ifconfig

Btw we don't ship the deprecated `net-tools` (which includes `ifconfig`) with DietPi by default. If the `wget` command I pasted above fails, `ip a` is the modern replacement for `ifconfig`.

Trunkzeh, 2019-09-02:
My Linux-fu is severely lacking..

JirkaKrivanek98, 2019-09-02:
Spider mode enabled. Check if remote file exists.
--2018-03-05 17:44:12--  http://ftp.cn.debian.org/debian/
Resolving ftp.cn.debian.org (ftp.cn.debian.org)... failed: Temporary failure in name resolution.
wget: unable to resolve host address ‘ftp.cn.debian.org’

real    0m40.107s
user    0m0.016s
sys     0m0.044s

I use ethernet.

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.101  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::81:46ff:fe7f:a7a5  prefixlen 64  scopeid 0x20<link>
        ether 02:81:46:7f:a7:a5  txqueuelen 1000  (Ethernet)
        RX packets 3116  bytes 225103 (219.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 570  bytes 117444 (114.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 40

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2  bytes 168 (168.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2  bytes 168 (168.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

MichaIng, 2019-09-02:
@JirkaKrivanek98 
Okay so it fails even without set timeout after 40 seconds.

Please try the following (the first command is just to see the current entry):
```
cat /etc/resolv.conf
echo 'nameserver 8.8.8.8' > /etc/resolv.conf
time wget --spider http://ftp.cn.debian.org/debian/
```

If this works now, then you can make this permanent by applying a static IP instead of DHCP (`dietpi-config` > `Network Options: Adapters` > `Ethernet`). Most routers should deal well with this when just the same IP is used that was assigned before by DHCP. Then you can change the DNS server. `8.8.8.8` is Google DNS which is a mostly well functional fallback. You can try out other local public DNS servers from your region.

Usually using the router itself as DNS server (choosing "Gateway" from the options) works best. The router uses an upstream DNS and caches entries for local requests. However in your case this somehow did not seem to succeed. But worth to play around with this a bid, probably there was simply something gone wrong with the DHCP assignment.

MichaIng, 2019-09-13:
Great you solved it, I will mark this issue as closed then. Feel free to reopen if required.

closed 2019-09-13 by MichaIng