MichaIng/DietPi #3090: RetroPie not working (only shows splashscreen at boot followed by consolle login credentials

garret, 2019-09-02:
I installed RetroPie on a fresh dietpi install (v6.25.1) by following this guide https://github.com/MichaIng/DietPi/issues/1653  (the SD card is of 32GB class 10). Before installing RetroPie I had also installed through the dietpi-software transmission, sonarr, radarr and Plex media server but did not touch them to configure them. The installation of RetroPie seems succesful as no error appeared during the installation. Additionally, a window confirming that RetroPie was correctly installed showed and I just had to reboot. 

By rebooting, I can see the RetroPie splashcreen appears for several seconds but then I am presented with the classical screen of dietpi consolle asking to login with a username and password. 

Do you know what could be the issue?

MichaIng, 2019-09-02:
@garret 
Hmm, long time not visiting this code. Did you enable RetroPie autostart in `dietpi-autostart`?
The guide you linked contains RetroPie being installed to `/mnt/dietpi_userdata/RetroPie`, but our autostart entry attempts to run `/opt/retropie/supplementary/emulationstation/emulationstation.sh`. Not 100% sure if binaries are indeed in `/opt` while userdata contain settings and games or such?

garret, 2019-09-02:
@MichaIng I did not enable RetroPie in autostart. The only thing I did was, just as the guide suggests, to give this command: 
`echo 3 > /DietPi/dietpi/.dietpi-autostart_index`
Is that what you meant? But if I see the RetroPie splashscreen is this already not a sign that is enabled in the autostart? 

I have not my raspberry pi here with me but can check later which types of files and folders are created and let you know :)

MichaIng, 2019-09-02:
@garret 
Ah yeah, this command sets the autostart index correctly. But what it does not do, is enabling autologin. So the login mask will be still shown, after login RetroPie should start.
Use `dietpi-autostart` and re-select RetroPie there to enable autologin. You are asked as well which user should be used to login (thus run RetroPie). Just note that it has not been tested with something different than root. Probably permissions need to be changed. Best more secure attempt would be `dietpi`, since it has write access to userdata.

If it fails, please try to run manually:
```
/opt/retropie/supplementary/emulationstation/emulationstation.sh
```
And paste output in case of failure. Probably the RetroPie installer has changed things that we need to adopt.

garret, 2019-09-02:
@MichaIng 
I will check later when I come home about the autologin/autostart and let you know :)

In the meantime, I think an answer for your last question could be found here https://github.com/MichaIng/DietPi/issues/1653#issuecomment-377610411 though I really think RetroPie would be still a really nice addition to DietPi!

MichaIng, 2019-09-02:
No fully automated install, but it would not be the only interactive installer that we use, e.g. same as Pi-hole.
Will have a look after v6.26 release, when I can.

garret, 2019-09-02:
@MichaIng 
I run the dietpi-autostart command, selected RetroPie and then dietpi as user. However, every time at the boot I get the screen I attached after the RetroPie splashscreen. I also tried to give manually the command you had suggested. 
[image]

By selecting root as user does not change the situation. I am thinking if it is not more convenient to have a multiboot system with retropie and dietpi like by using something like NOOBS or berry boot.

MichaIng, 2019-09-02:
@garret 
Interesting, actually `dietpi` is added to `video` group on image creation: https://github.com/MichaIng/DietPi/blob/master/dietpi/func/dietpi-set_software#L611
Also then when running as `root`, this error should not show up.

Just to rule it out, when running with `sudo` or logging in as root, the errors are the same `lvl 0:`?

A multiboot system is not something we gonna implement, but we should be able to find the issue.

garret, 2019-09-03:
@MichaIng 
Yes, also with root as user I get the same errors.

MichaIng, 2019-09-03:
Okay, have to run some tests here, also consider to check the RetroPie docs about which binary to use in which cases. AFAIK there is a direct RetroPie executable as well, but we used emulationstation since it allows to be called as root. But since one can use the `dietpi` user for autologin now, the direct executable could be used.

garret, 2019-09-04:
@MichaIng 
Ok so essentially I should abandon my hopes to use RetroPie on Dietpi, right? :P 
I think at this point I will try to use PINN and have a dual boot with dietpi and another one with retropie.

MichaIng, 2019-09-04:
@garret 
The issue should not be hard to solve, but requires some tests, which I can do after v6.26 release.

CupOfTea696, 2019-10-21:
Just stumbled upon this when encountering the same issue, this is just an FYI for anyone reading this.

It looks like the RetroPie setup installs a boot splash screen, but doesn't actually enable autostart. Best way to set this up is through `dietpi-autostart` and enabling it there. The splash screen itself can be [enabled/disabled](https://retropie.org.uk/docs/Splashscreen/) from the RetroPie Menu in EmulationStation or through the setup script under option 3 (`RetroPie-Setup/retropie_setup.sh` › `Configuration / tools` › `splashscreen - Configure Splashcreen`), but does not actually affect whether or not RetroPie / EmulationStation will autostart.

MichaIng, 2019-10-23:
@CupOfTea696 
Thanks for your info. That is indeed strange, so good to know that our RetroPie autostart entry still has a use 😄. However makes sense to check for reintegration to automate/enhance the whole setup a bid, e.g. disable the splash screen by default 😉.

MichaIng, 2020-09-27:
Additional information: #3790