MichaIng/DietPi #3093: deluge and couchpotato failed

yopyop2019, 2019-09-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`

G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

hi, 
I have installed deluge and couchpotato but I can not connect them, also whenever I use dietpi-services restart couchpotato tells me FAILED and if I make a "stop" it tells me that it stops but continues to run the couchpotato service please check if the couchpotato installation is configured correctly since I've never had these problems before, thanks

(I Download and install the version from the dietpi.com page on 08-31-2019)

MichaIng, 2019-09-02:
@yopyop2019 
Many thanks for your report.

Can you try:
```
systemctl stop couchpotato
rm -R /var/run/couchpotato
systemctl start couchpotato
```
If it still fails:
```
journalctl -u couchpotato
```

The first error is due to world-writable PID file, why ever this is. But the systemd unit reports "INVALIDARGUMENT", which does not really fit. However lets see if it is really only the PID file.

yopyop2019, 2019-09-02:
thanks for your kind reply
I deal with the commands but I still can't connect deluge with couchpotato

journalctl -u couchpotato

[image]

MichaIng, 2019-09-02:
@yopyop2019 
Hmm strange, so it's the sysvinit service itself that seems to create this PID file with wrong permissions...

What happens if you run the script manually:
```
CouchPotato.py --daemon --quiet --data-dir /mnt/dietpi_userdata/couchpotato
```

Will run some tests tomorrow as well.

yopyop2019, 2019-09-02:
when i run the command couchpotato log

[image]

MichaIng, 2019-09-02:
@yopyop2019 
There seem to be another instance still running:
```
killall -qwr [cC]ouch[pP]otato*
CouchPotato.py --daemon --quiet --data_dir=/mnt/dietpi_userdata/couchpotato
```

Or is there an unrelated other process using this port?
```
fuser 5050/tcp
```

yopyop2019, 2019-09-02:
@MichaIng 
same error with the two commands

fuser 5050/tcp
     output :         
            5050/tcp:              824

MichaIng, 2019-09-02:
@yopyop2019 
And which process this is?
```
ps 824
```

yopyop2019, 2019-09-02:
sorry
is the couchpotato

[image]

MichaIng, 2019-09-02:
@yopyop2019 
Ah, the kill command was from I think because the actual binary is python...
```
kill 824
CouchPotato.py --daemon --quiet --data_dir=/mnt/dietpi_userdata/couchpotato
```

yopyop2019, 2019-09-03:
@MichaIng 
hi, execute the command but I still can't connect to the deluge, so install transmission and it worked perfect, so it must be a problem of the deluge

MichaIng, 2019-09-08:
@yopyop2019 
Okay, so I got it wrong. CouchPotato generally works, besides service stop/restart does not really stop the old process, so the restart then fails. However the old process is still up.

But your issue is that CouchPotato cannot connect to Deluge, right? Does the Deluge web UI work well?

yopyop2019, 2019-09-09:
@MichaIng 

Yes, both separate applications work fine, but when trying to connect the deluge is when the problem occurred

MichaIng, 2019-09-09:
@yopyop2019 
Okay, I have to run some local test. If you face any error message or log entry, please paste it here.

MichaIng, 2020-03-18:
@yopyop2019 
Took a while but I made some tests now. Basically it works, connection test from CouchPotato to Deluge succeeds, no error logs. But I have not idea how to trigger the actual download now. I added some movies for testing, two were found, but how do I start the download now?

MichaIng, 2020-06-01:
Marking as closed due to missing reply. Feel free to reopen if the issue is still present.

closed 2020-06-01 by MichaIng