MichaIng/DietPi #3096: webmin does not start after install

nnovaes, 2019-09-03:
dietpi version 6.25.3

after install of webmin through dietpi-software:

root@DietPi:~# service webmin status
● webmin.service - Webmin (DietPi)
   Loaded: loaded (/etc/systemd/system/webmin.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-09-03 10:39:52 EDT; 5min ago
  Process: 882 ExecStart=/etc/webmin/start (code=exited, status=203/EXEC)

Sep 03 10:39:52 DietPi systemd[1]: Starting Webmin (DietPi)...
Sep 03 10:39:52 DietPi systemd[1]: webmin.service: Control process exited, code=exited status=203
Sep 03 10:39:52 DietPi systemd[1]: Failed to start Webmin (DietPi).
Sep 03 10:39:52 DietPi systemd[1]: webmin.service: Unit entered failed state.
Sep 03 10:39:52 DietPi systemd[1]: webmin.service: Failed with result 'exit-code'.
root@DietPi:~# 

root@DietPi:/var/log# cat daemon.log | grep web
Sep  3 10:39:52 DietPi systemd[882]: webmin.service: Failed at step EXEC spawning /etc/webmin/start: No such file or directory
Sep  3 10:39:52 DietPi systemd[1]: webmin.service: Control process exited, code=exited status=203
Sep  3 10:39:52 DietPi systemd[1]: webmin.service: Unit entered failed state.
Sep  3 10:39:52 DietPi systemd[1]: webmin.service: Failed with result 'exit-code'.
root@DietPi:/var/log#

MichaIng, 2019-09-03:
@nnovaes
Many thanks for your report.
Looks like they switched the binary name or location.

Also there is a repository available now which we might apply instead of installing the package manually: https://doxfer.webmin.com/Webmin/Installation

nnovaes, 2019-09-03:
> Can you paste: `ls -l /etc/webmin`**
> 
> Also there is a repository available now which we might apply instead of installing the package manually: https://doxfer.webmin.com/Webmin/Installation

there's no such directory ...

root@DietPi:/# ls -l /etc/webmin
ls: cannot access '/etc/webmin': No such file or directory

nnovaes, 2019-09-03:
i've also tried to uninstall and reinstall through dietpi-software, no success.

MichaIng, 2019-09-08:
@nnovaes 
Very strange, I could not replicate. Install works very well here. Especially strange, since download of the Webmin package and dpkg install are both error handled, so there is not really a chance this can fail without interactive error prompt.
I would need a full dietpi-software install output to check for suspicious/missing/error output.

However I switched our install script to use the official Webmin APT repo: https://github.com/MichaIng/DietPi/commit/ec55d49a488fd9cb1dbb18cbfa067b5d1d1a974e
Before attempting to debug the old install, please try first the new way manually:
```
curl -sSL 'http://www.webmin.com/jcameron-key.asc' | apt-key add -
echo 'deb http://download.webmin.com/download/repository sarge contrib' > /etc/apt/sources.list.d/webmin.list
G_AGUP
G_AGI webmin
```

I'll add a reinstall on next DietPi-Update as well to switch all users to the APT repo. It ships a newer version as well and there is no dependency reinstalled with it, so it's a pretty fast and safe reinstall/update.

nnovaes, 2019-09-16:
same issue... wheni run

> 
> curl -sSL 'http://www.webmin.com/jcameron-key.asc' | apt-key add -
> echo 'deb https://download.webmin.com/download/repository sarge contrib' > /etc/apt/sources.list.d/webmin.list
> G_AGUP
> G_AGI webmin

i get

> 
> ecting previously unselected package libapt-pkg-perl.^M
> paring to unpack .../4-libapt-pkg-perl_0.1.32_amd64.deb ...^M
> acking libapt-pkg-perl (0.1.32) ...^M
> ecting previously unselected package apt-show-versions.^M
> paring to unpack .../5-apt-show-versions_0.22.7_all.deb ...^M
> acking apt-show-versions (0.22.7) ...^M
> paring to unpack .../6-webmin_1.930_all.deb ...^M
> acking webmin (1.930) ...^M
> g: error processing archive /tmp/apt-dpkg-install-fgVF4k/6-webmin_1.930_all.deb
> unpack):^M
> able to open '/usr/share/webmin/quota/uninstall.pl.dpkg-new': Operation not
> mitted^M
> ors were encountered while processing:^M
> mp/apt-dpkg-install-fgVF4k/6-webmin_1.930_all.deb^M
> Sub-process /usr/bin/dpkg returned an error code (1)
> 
>

MichaIng, 2019-09-17:
@nnovaes 
> Operation not
mitted

Did you run those commands as root user or with `sudo`/`G_SUDO` prefix?

nnovaes, 2019-09-17:
yes, i did run as root

MichaIng, 2019-09-17:
Could you try to purge the old package first: `apt purge webmin`

nnovaes, 2019-09-17:
same issue after running "apt purge webmin"

MichaIng, 2019-09-18:
@nnovaes 
Not the same issue, but now a different file: `/usr/share/webmin/quota/uninstall.pl.dpkg-new` vs `/usr/share/webmin/gray-theme/samba/images/askepass.gif.dpkg-new`.

Lets see how this files meta data look like:
```
stat /usr/share/webmin/gray-theme/samba/images/askepass.gif.dpkg-new
```

Does `dmesg` report any error? Having random files missing and random changing access issues could mean some drive I/O issues, although I see it's a VM 🤔. However I tried it on all my VMs and didn't face any issue any of them (neither with old manual deb install, nor with new repo install), so its a system specific issue, not a general/method one.

Also force some file system check on reboot, just in case. Is this VirtualBox, VMware or something else?

MichaIng, 2019-10-18:
I close this issue due to missing reply. Feel free to reopen if required, although it indeed looks like something went wrong with download/extraction/storing of files, probably disk corruption.

closed 2019-10-18 by MichaIng