MichaIng/DietPi #3097: myMPD source build

jcorporation, 2019-09-03:
Hi,
I am the developer of myMPD and I plan to release a new myMPD release which will break your source build in the future. 

1)
The mkrelease.sh script is deprecated in my devel branch.
Please consider replacing ``./mkrelease.sh`` with ``./build.sh releaseinstall``. For now I added a ``./mkrelease.sh`` wrapper in devel and a ``./build.sh`` wrapper in master to prevent breakage of your source build.

2)
I see also, that you change the group of the mympd user to dietpi in your systemd service. Why is this needed and is it not better to change the primary usergroup to dietpi? Changing ``usermod -a -G dietpi mympd`` to ``usermod -g dietpi mympd`` should have the same effect?
In the new version myMPD creates the directory /var/lib/mympd ans it subdirs on startup with a umask of 0007.

3)
The symlink to music/libary is no more needed. myMPD handles this internally.

MichaIng, 2019-09-04:
@jcorporation 
Many thanks for your info! Since it is still compatible, I will apply the changes with v6.27. We need to get v6.26 released soon.

About user/group:
- Are you aware of any downsides when applying dietpi group via systemd unit only? It is a failsafe method, that avoids to touch additional groups, the user was added to, so e.g. on reinstall, if the "user" changed permissions or group memberships for some reason, those are not overwritten. Also we can do this as a general method for all software installs, where required, without needing to check and perserve explicitely existing memberships.
- Btw we use dietpi group with all software titles, that write files to media or download dirs, so other media and download software can access those files without any additional configuration needed, and this with all downloaders/sonarr/radarr/../media players etc we offer. But actually myMPD does not write media files, does it? It should be sufficient to add dietpi group as additional group only (so it has access to media files), but it can still run as mympd group?
- I will have a look at config directories and symlinks. When do you expect release where these changes need to be done?

jcorporation, 2019-09-11:
myMPD don't write files to media or download dirs. It writes only internal states to /var/lib/mympd. So  it should be sufficient to add dietpi group as additional group only.

The revised install script now detects old installations and run following actions:
- removes the /usr/share/mympd directory entirely (assets are now embedded in the binary)
- moves /etc/mympd/mympd.conf to /etc/mympd.conf
- removes /etc/mympd directory
- removes /var/lib/mympd/tmp directory
- moves the systemd unit from /usr/lib/systemd/system to /lib/systemd/system (right place according to lintian)

I hope this changes enables a smooth change for the dietpi users. Feel free to propose changes to the installation / upgrade workflow for myMPD.

The new version of myMPD should be released in this month.

jcorporation, 2019-09-13:
myMPD 5.6.0 is now released. It should not break your source build.

MichaIng, 2019-09-15:
@jcorporation 
Okay I updated our installer as well: https://github.com/MichaIng/DietPi/pull/3110

Btw I saw: https://github.com/jcorporation/myMPD/blob/master/build.sh#L184-L188
[rmdir](https://linux.die.net/man/1/rmdir) has the option `--ignore-fail-on-non-empty` which allows it to omit and succeed on non-empty dirs silently.

jcorporation, 2019-09-15:
thanks for the hint with rmdir

closed 2019-10-15 by MichaIng