MichaIng/DietPi #3098: Ethernet issue with u-boot in odroid-n2 beta image

cmessin1, 2019-09-04:
Hopefully this is the correct place for this.

I have found that the u-boot that ships with the current beta odroid-n2 image on the website to be an issue.  The on board ethernet has massive packet loss when connected at 1G, though it is functional at 100mb.

I upgraded to the latest u-boot from the the odroid git and this resolved the issue.

I would suggest you update the u-boot in the posted image.

MichaIng, 2019-09-04:
@cmessin1 
You should be able to install the latest kernel from APT repo via `apt full-upgrade`.

What we should do is applying this full-upgrade on first boot automatically. I am not keen to do this on every software install and regular updates, since in rare cases, e.g. some conflicting repos and such, when forced/automated, it can purge packages that you actually wanted/need, even required system packages by times.

MichaIng, 2019-09-15:
Firmware package upgrades are now done automatically on first run installs: https://github.com/MichaIng/DietPi/commit/b81bee4c90bd8ed960679db49f50d6ac18e289fa

closed 2019-10-15 by MichaIng

oskhen, 2023-01-11:
Apologies in advance for the necro, but I encountered some weird network-issues on a fresh install of DietPi_RPi-ARMv8-Bullseye (installed from [here](https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bullseye.7z)) on my Raspberry Pi 4 4GB regarding packet-loss and extremely slow network-speeds. I found this issue and after setting my link-speed to 100 the network-issues were gone and I could do a `apt full-upgrade` without issues, which then permanently fixed my issue. 

Since this is a 3 year old issue it seems very strange, but perhaps something else is going on here? If it is unrelated, feel free to close the issue again.

Joulinar, 2023-01-11:
where does the RPi 4 connects to? A switch or a router? Did you tried to switch Ethernet cable? Or use a different port?

oskhen, 2023-01-13:
Well, this is extremely awkward. It was a network issue. Apologies