MichaIng/DietPi #3100: Wireguard - User management setting in dietpi-config or dietpi-xyz

Vansinnet, 2019-09-05:
#### Describe the solution you'd like:
I'd like to be able to add and remove clients/users more easily. Currently you have to do a lot of work to add a client to Wireguard server.

Would be cool to add something like this to dietpi-config or dietpi-xyz:

https://github.com/adrianmihalko/raspberrypiwireguard/wiki/User-management-with-Wireguard-User-Management-script

MichaIng, 2019-09-05:
@Vansinnet
Many thanks for your request.

We plan a dedicated dietpi-vpn program from where you can install and manage VPN servers and clients with WireGuard and OpenVPN, add clients, servers to connect to etc. Will be a bid more work, will see if a start can be done with v6.27.

Vansinnet, 2019-09-06:
@MichaIng 

That is great! The new RPi4 has the required hardware to get decent speeds over VPN; so I think a lot of people will want "dietpi-VPN" in the coming years.

csgabe, 2019-09-08:
An easier way is adding wg-dashboard:
https://github.com/wg-dashboard/wg-dashboard

Vansinnet, 2019-09-08:
> An easier way is adding wg-dashboard:
> https://github.com/wg-dashboard/wg-dashboard

That's perfect! I'll use this instead of wg_config (which is terminal only). 

@MichaIng 
Couldn't wg-dashboard be used in the meantime until you have everything in place for a dietpi-vpn-module? Maybe incorporate wg-dashboard into dietpi with the 6.26 release?

MichaIng, 2019-09-08:
@Vansinnet 
Most probably nothing to implement "meanwhile" but additionally. Our own scripts will always be console and whiptail-based only, wg-dashboard is a nice addition if you want to monitor/configure your WireGuard remotely via web UI.

Vansinnet, 2019-09-09:
@MichaIng 
Does this mean wg-dashboard isn't something to add to Dietpi until you add a Dietpi-VPN-module? It's no worries, wg-config tool is plenty until then!

MichaIng, 2019-09-09:
@Vansinnet 
> Does this mean wg-dashboard isn't something to add to Dietpi until you add a Dietpi-VPN-module?

IMO both features are unrelated:
- DietPi-VPN is something I aim for a longer time, so is kinda on my ToDo, although a larger project indeed.
- wg-dashboard is WireGuard only, but with web UI (means a Node.js-based webserver is running), so offers just a different set of features, only partly overlapping with DietPi-VPN. Not much coding required to implement it, at least as long as it works with newer Node.js versions as well 😉.

However you might add this to our [FeatHub](https://feathub.com/MichaIng/DietPi) to raise some attention. Perhaps we change the title of this issue to be a `wg-dashboard` software request, since DietPi-VPN is on my list anyway?

Vansinnet, 2019-09-09:
@MichaIng 
I'm content with your reply. It's not something I really, really need since I use `wg-config` to manage users for my Wireguard server. And since you're already planning to implement Dietpi-VPN, I can wait for that. :)

I'm not adding `wg-dashboard` to Feathub since you're working on Dietpi-VPN. It sounds great and would render `wg-dashboard` obsolete!

Many thanks for your feedback! And you're doing a great job. :)

HyperCriSiS, 2022-05-09:
This one seems also good:
https://github.com/ngoduykhanh/wireguard-ui

MichaIng, 2022-05-09:
A pure slim CLI has benefits as well, but I agree that providing an install option for the existing web interface is probably the better first step.

Joulinar, 2022-05-09:
It looks nice and is easy to run (just a single executable) but it did not recognise existing installations/configuration. It's generating an own key pair. Means whole existing configuration for server and clients would need to be migrated somehow. https://github.com/ngoduykhanh/wireguard-ui/issues/177

As well login user/pass seems to be written into a config file without encryption https://github.com/ngoduykhanh/wireguard-ui/issues/45

Joulinar, 2026-09-26:
Picking this up again, since managing WireGuard clients is still manual work for users, and the web UIs mentioned above do not work with an existing setup (they generate their own keys and config, see my comment above). Here is a concept for a whiptail-based `dietpi-wireguard`. I'd like your opinion before I start coding.

## Main idea: the existing config is the only source of truth

The tool works on whatever server config is there. It does not rely on DietPi defaults and has no database of its own. So it also works for WireGuard servers that were not installed via `dietpi-software`, and it does not force a config layout on users like PiVPN does.

| What | Where it comes from |
|---|---|
| Server interfaces | `/etc/wireguard/*.conf` with `ListenPort` (client configs are skipped) |
| Server public key | derived via `wg pubkey` from the `PrivateKey` in the config (`server_public.key` may not exist) |
| Clients | `[Peer]` blocks of the server config |
| Client names | comment line directly above `[Peer]` (matches our current `# Client 1`), optionally PiVPN `### begin <name> ###` markers, fallback: short public key |
| Full client config (for QR) | client config files, matched to peers by the public key derived from their `PrivateKey`. Our existing `wg0-client.conf` is picked up as "Client 1" this way, without moving anything |
| Next free client IP | subnet(s) from `[Interface] Address` minus all peer `AllowedIPs` (IPv4, plus IPv6 if configured) |
| Endpoint/DNS/AllowedIPs for new clients | taken from an existing client config as template, fallback `dietpi.txt` (`SOFTWARE_PUBLIC_DOMAIN_NAME`, `SOFTWARE_WIREGUARD_PORT`), else ask |

Limitation: for peers created outside the tool, the client private key is not on the server, so they get status/edit/remove but no QR code. The menu would explain this.

## Menu draft

```
DietPi-WireGuard ── wg0 (active, port 51820, 10.9.0.1/24)
  Interface      : [wg0]                  <- only if more than one server config exists
  ●─ Clients
  Client 1       : ● online  10.9.0.2  42s ago  rx 1.2 GiB tx 80 MiB
  Phone          : ○ offline 10.9.0.3  3 days ago
  <a1B2c3...>    : ○ never   10.9.0.4  (no name/no client config)
  Add client     : Create a new client
  ●─ Server
  Settings       : Endpoint, port, defaults for new clients
  Edit config    : Open wg0.conf in editor (validated before applying)
  Service        : [active] restart/stop
```

- Per client: show QR code, show config, rename, AllowedIPs (full tunnel / server LAN only / server only, same as the options in our current client template), DNS, keepalive, add PresharedKey, disable, remove.
- Status from `wg show <if> dump`: last handshake, transfer, endpoint.
- CLI for scripting and the docs: `dietpi-wireguard list|status|add <name>|remove <name>|qr <name>`, similar to `dietpi-ddns apply`.

## Technical notes

- Peer changes are applied with `wg syncconf wg0 <(wg-quick strip wg0)`, so other clients stay connected. Only `[Interface]` changes restart `wg-quick@wg0`.
- The config is never regenerated. Only single `[Peer]` blocks are added, removed or changed; unknown keys and comments stay. Backup before each write, validation with `wg-quick strip`, rollback on error.
- If the config has `SaveConfig = true`, wg-quick overwrites it on shutdown and drops the comments (names). The tool detects this and warns.
- Disabled clients are moved to e.g. `wg0.disabled` instead of commenting them out, which is easier to parse reliably.
- QR code via `qrencode -t ansiutf8` outside whiptail, with a terminal size check.
- New client files with `umask 077`/`0600`.

## Open questions

1. **Name/scope:** a separate `dietpi-wireguard`, or a server part in `dietpi-vpn`? I'd go with a separate tool: `dietpi-vpn` is a client tool, and `dietpi-banner` uses its `status` output.
2. **Where to store new client configs:** `/etc/wireguard/clients/<name>.conf`, or flat `wg0-<name>.conf` next to the existing `wg0-client.conf`? The flat layout looks like interface configs to wg-quick, which does no harm but is less clean.
3. **dietpi-software integration:** later, the install could create the first client through the tool, so client configs are created in only one place. I'd leave that for a second step.

What do you think?

MichaIng, 2026-09-26:
I like the concept to just work with the existing configs. Might be good to list unused client configs (`*.conf` with no `ListenPort` and not in any server config) separately as well, to allow some cleanup.

I'd just use filenames as client names, as there is no standard for this, and the first (comment) line might show something entirely different. If `SaveConfig = true` causes comments to be removed (I didn't know that setting), this is another argument. Also, it is common to strip comments when sharing a client config with QR code (otherwise phones have a harder time scanning it), so that "name" never arrives at the client either.

I'd generally try to avoid the "Settings" submenu, and instead add the settings to the main menu, to save one navigation step. If it becomes too long/crowded, we can reconsider.

Definitely a separate `dietpi-wireguard` tool. `dietpi-vpn` is not only a client-only tool, but also an OpenVPN-only tool.

Actually, might be an idea to allow managing client configs for external servers as well in `dietpi-wireguard`? The interface menu could show above mentioned client configs with no local matching server config, and treat them as client configs for external servers, which this host can connect to. This would also allow setups where the host acts as VPN server and client at the same time. But maybe this is a follow-up task, after the base server management features have been battle-tested.

Unsure about whether to use a subdir for client configs. At least the tool would need to detect those in the parent dir as well, and maybe transparently move them to the subdir on change, similar to how `dietpi-network` migrates interfaces from `/etc/network/interfaces` to `interfaces.d`, when saving settings for that interface.

I'd remove any client config generation in `dietpi-software` then, and just show a prompt that `dietpi-wireguard` can be used to create and manage them, after the installation went through.

EDIT: And a fully-featured CLI would be nice!

Joulinar, 2026-09-26:
Thanks for the feedback! Here is an updated concept based on your points.

## Config types in `/etc/wireguard`

With external connections in mind, besides server configs there are three kinds of configs without `ListenPort`. All of them can be told apart by keys alone, without any database of our own:

| Type | Detection | Meaning |
|---|---|---|
| **Server** | `[Interface]` with `ListenPort` | local WireGuard server |
| **Client of a local server** | `[Peer] PublicKey` matches a local server, and its own public key is a peer there | config for a phone, laptop, ... |
| **Unused client config** | `[Peer] PublicKey` matches a local server, but its own public key is not (or no longer) a peer there | listed separately for cleanup or re-adding |
| **External connection** | `[Peer] PublicKey` does not match any local server | this host connects as client to another server (follow-up) |

This distinction matters for the cleanup list: without it, configs for external VPN providers would show up as unused.

## Client config location and names

- Client configs for local servers go to `/etc/wireguard/clients/<interface>-<name>.conf`, e.g. `clients/wg0-phone.conf`. The prefix shows which server interface they belong to, and the `clients/` subdir keeps them apart from configs this host brings up itself, so `wg-quick up wg0-phone` cannot happen by accident. Servers and external connections stay in `/etc/wireguard/`, where `wg-quick@<name>` expects them.
- The client name is the filename without the interface prefix and `.conf`, as you suggested. No names are parsed from comments.
- The server a client belongs to is still detected by key, not by the prefix. So configs with other names are found as well.

## Migration like `dietpi-network`

- Client configs are read from `/etc/wireguard/clients/` first, then from `/etc/wireguard/`.
- Showing the QR code or the config does not move anything.
- On the first change (rename, AllowedIPs, DNS, keepalive, PSK), the file is moved unchanged to `clients/`, then only the affected lines are edited. The client submenu title shows a hint that this will happen.
- The file is moved, not regenerated, so the user's own keys and comments stay.
- Our existing `wg0-client.conf` becomes `clients/wg0-client.conf` (name "client") on its first change. No patch in `.update/patches` is needed.
- Removing a client deletes the file, wherever it is, and its `[Peer]` block in the server config.

## Main menu without a settings submenu

"Defaults for new clients" are dropped completely, as they would be a hidden database. When adding a client, endpoint, DNS and AllowedIPs are prefilled from the most recently changed client config of the same server and can be adjusted right there. So the main menu only holds what is in the server config:

```
DietPi-WireGuard
  Interface          : [wg0]              <- only with more than one server, switches the view
  ●─ Server wg0
  Service            : [active, enabled at boot]
  Port               : [51820]
  Address            : [10.9.0.1/24]
  Edit config        : Open wg0.conf in editor (validated before applying)
  ●─ Clients
  phone              : ● online   10.9.0.3  42s ago  rx 1.2 GiB tx 80 MiB
  laptop             : ○ offline  10.9.0.2  3 days ago
  <a1B2c3...>        : ○ never    10.9.0.4  no local config
  Add client         : Create a new client
  ●─ Unused client configs                 <- only if any exist
  old-tablet         : for wg0, not a peer -> remove / re-add
  ●─ External connections                  <- follow-up
  provider           : [connected] vpn.example.net:51820
```

- About 6 fixed entries; the rest grows with the number of clients. With several servers, only one is shown at a time.
- Changing the port or address breaks all existing client configs, so the tool asks whether to update them as well. Devices then need to scan the new QR code.

Per client (submenu stays): QR code (comments stripped, as you suggested), show config, rename, AllowedIPs (full tunnel / server LAN only / server only), DNS, keepalive, add PresharedKey, disable, remove. Peers without a local client config get status, disable and remove only.

## Cleanup

The "Unused client configs" section (and `dietpi-wireguard cleanup`) also lists leftover key files in `/etc/wireguard`, like `server_*.key` and `client_*.key` from the `dietpi-software` install. They are not needed, since the private keys are in the configs and public keys can be derived with `wg pubkey`. Only key files that match a key in an existing config are listed. They are removed after confirmation, never automatically.

## CLI

```
dietpi-wireguard status [<interface>]
dietpi-wireguard list [<interface>]
dietpi-wireguard add <name> [-i <if>] [--ip <ip>] [--dns <dns>] [--allowed-ips <cidr>] [--endpoint <host:port>] [--keepalive <s>] [--psk]
dietpi-wireguard remove|disable|enable|qr|show <name> [-i <if>]
dietpi-wireguard rename <name> <new>
dietpi-wireguard set <name> <key>=<value>
dietpi-wireguard server <if> port=<port>
dietpi-wireguard cleanup
```

## dietpi-software

As you suggested: no client config generation anymore, just a note after install that clients are created and managed with `dietpi-wireguard`.

## External connections (follow-up)

The config types above are already designed for this: up/down, start at boot via `wg-quick@<name>`, status, import from file. One caveat for later: an external connection with `AllowedIPs = 0.0.0.0/0` on a host that is also a server routes the replies to its own clients into that tunnel. This will need a warning or policy routing.

MichaIng, 2026-09-26:
Looks good. One remaining point regarding the CLI: Please avoid duplication with the `wg` CLI. If `dietpi-wireguard status wg0` shows not significantly more/different info than `wg show wg0`, then people should rather learn about and use `wg` directly. Same with `dietpi-wireguard list wg0`, if this is intended to show the client list of this server/interface.

Joulinar, 2026-09-26:
Good point. I'd split it like this: `wg` covers the runtime state, `dietpi-wireguard` manages the config files.

- `status` is dropped, `wg show` covers it.
- `show` is dropped as well, it would just be `cat` of the client config.
- `list` stays, but shows only what `wg` does not know: name, address, config file and state (active/disabled/unused). No handshake or transfer data. Disabled peers and unused configs are not visible to `wg` at all.

```
$ dietpi-wireguard list wg0
NAME      ADDRESS     STATE     CONFIG
phone     10.9.0.3    active    /etc/wireguard/clients/wg0-phone.conf
laptop    10.9.0.2    disabled  /etc/wireguard/clients/wg0-laptop.conf
-         10.9.0.4    active    (no local config, public key a1B2c3...)
old-tab   -           unused    /etc/wireguard/clients/wg0-old-tab.conf
```

The other commands stay, as `wg set` only changes the runtime state and not the config files. The menu still shows the online status per client, since that is where it is useful at a glance.

If that works for you, I'll start with the implementation.

closed 2026-10-03 by MichaIng

Dynamic5912, 2026-10-08:
How does this differ from PiVPN?

Joulinar, 2026-10-08:
`dietpi-wireguard` uses existing configurations that the user has created individualy. This is different from PiVPN, which uses its own default settings and ignores existing configurations.

Dynamic5912, 2026-10-08:
Is it a replacement for PiVPN?

Joulinar, 2026-10-08:
nope

MichaIng, 2026-10-08:
Though, you should be able to replace PiVPN with `dietpi-wireguard` without issues, since it just works with the configs at hand. But the other way round, it won't work, since PiVPN creates its own new configs, hence won't take over what you probably created with `dietpi-wireguard`, _unless the naming matches by chance._

Now I remember https://github.com/wg-dashboard/wg-dashboard was the other topic here, but it has been archived in 2020 sadly.
The alternative they link in their readme has been archive in 2023: https://github.com/EmbarkStudios/wg-ui
The other alternative has not been archived, but did not receive any commit since 4 years ☠️: https://github.com/subspacecommunity/subspace
Tailscale is of course well known, and available in `dietpi-software` already. But for server systems which are reachable from the Internet anyway, I prefer an independent WireGuard server, while Tailscale uses and hence bind you to their own infrastructure. Handy for non-server systems, but IMO unnecessary for server systems.