MichaIng/DietPi #3101: WPA_Supplicant in DietPi Buster cannot connect to enterprise networks / eduroam

ImminentFate, 2019-09-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.25.3`
- Distro version | `buster`
- Kernel version | `Linu DietPi 4.19.66+`
- SBC device | `RPi Zero W`
- Power supply used | `Offical RPi 5.1V 2.5A PSU
- SDcard used | SanDisk Ultra A1 1GB

#### Additional Information (if applicable)
- Software title | **wpa_supplicant 2:2.7+git20190128+0c1e29f-6**
- Was the software title installed freshly or updated/migrated? **Fresh Install**
- Can this issue be replicated on a fresh installation of DietPi? **Yes**

#### Steps to reproduce

1. Using a clean install:  
2. Use wpa_supplicant.conf to connect to an enterprise network (eduroam etc) that uses username + password authentication. 

#### Expected behaviour
- Enterprise network will connect after a reboot

#### Actual behaviour
- Network connection does not get established
- Calltrace highlights `Comm: wpa_supplicant Tainted`

#### Extra details
- Does not work even after directly copying a config from another working RPi on Raspbian Stretch

#### Workaround
- Downgrading to wpasupplicant=2:2.4-1+deb9uf from the Raspbian Stretch repo resolves the issue.

MichaIng, 2019-09-08:
@ImminentFate 
Many thanks for your report.

Did you use `dietpi-wifi.txt` to apply the EAP credentials on first boot, or did you manually edit `wpa_supplicant.conf`?
In case of manual edit, can you paste your non-functional `wpa_supplicant.conf` (with erased sensitive content of course), so we can try to identify the incompatible setting.

ImminentFate, 2019-09-10:
I tried both ways actually, but neither worked. 
Here's the `wpa_supplicant.conf` file: 
```
country=AU
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
	ssid="eduroam"
	scan_ssid=1
	key_mgmt=WPA-EAP
	eap=PEAP
	identity="username@edudomain"
	password="pass"
	phase1="peaplabel=0"
	phase2="auth=MSCHAPV2"
}
```

Notably, this exact config file works on Raspbian Buster and Stretch. I also happened to find this reference to what looks like the same bug in the base version of Buster back in June, which may have been patched since: https://bugs.launchpad.net/raspbian/+bug/1834749

MichaIng, 2019-09-10:
@ImminentFate 
Hmm, the wpasupplicant package version has not changed since then: https://packages.debian.org/buster/wpasupplicant
Raspbian Buster matches Debian Buster here, which can be checked through the Packages file or repo pool:
- http://raspbian.raspberrypi.org/raspbian/pool/main/w/wpa/
- http://raspbian.raspberrypi.org/raspbian/dists/buster/main/binary-armhf/Packages

But just to be sure, you might want to try a full APT upgrade that includes firmware, if not yet done: `apt update && apt full-upgrade`
After RPi4 release firmware/kernel upgrades have been quite regular to moreless quickly fix issues with the RPi4 integration.
- You did test non-EAP connection successfully, right?

Also the bug report you linked shows kernel errors, so it is probably not even a wpasupplicant issue. Could you check for kernel errors as well after attempting to connect to the EAP network? `dmesg` should show those error traces if present.

I still have eduroam access, but sadly not mobile RPi that I can quickly test with 🤔.

ImminentFate, 2019-09-11:
I'll flash a fresh card tomorrow and test those steps out :). It's weird that downgrading wpa_supplicant worked though, especially if the version haven't changed between, but good pickup that I haven't tried `apt full-upgrade`. 
A non-EAP connection does successfully work, yes.

MichaIng, 2019-09-11:
@ImminentFate 
The Stretch repo of course ships an older version, but on the Buster repo, the more current v2.7 version has not changed since 06 Jun 2019.

closed 2020-07-05 by MichaIng