MichaIng/DietPi #3102: G_AGUG mariadb error

kirkface, 2019-09-09:
#### Details:
- Date           | Mon  9 Sep 19:34:54 AEST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (index=4)
- Kernel version | #1253 SMP Thu Aug 15 12:02:08 BST 2019
- Distro         | buster (index=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title |

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Setting up mariadb-common (1:10.3.17-0+deb10u1) ...
update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist
dpkg: error processing package mariadb-common (--configure):
 installed mariadb-common package post-installation script subprocess returned error exit status 2
dpkg: dependency problems prevent configuration of libmariadb3:armhf:
 libmariadb3:armhf depends on mariadb-common; however:
  Package mariadb-common is not configured yet.

dpkg: error processing package libmariadb3:armhf (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of libmariadb-dev:
 libmariadb-dev depends on libmariadb3 (= 1:10.3.17-0+deb10u1); however:
  Package libmariadb3:armhf is not configured yet.

dpkg: error processing package libmariadb-dev (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of libmariadbclient-dev:armhf:
 libmariadbclient-dev:armhf depends on libmariadb-dev (= 1:10.3.17-0+deb10u1); however:
  Package libmariadb-dev is not configured yet.

dpkg: error processing package libmariadbclient-dev:armhf (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 mariadb-common
 libmariadb3:armhf
 libmariadb-dev
 libmariadbclient-dev:armhf
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2019-09-09:
@kirkface 
Many thanks for your report.
> update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist

Strange, this file seems to be missing, which is part of the package itself. Please verify it is missing:
```
ls -l /etc/mysql/mariadb.cnf
```
Check prior error output for any reason why unpacking of the package might have failed. Otherwise since it is a config file, if you or anything else removed it manually, AFAIK it will not automatically be re-created. Probably there is a `/etc/mysql/mariadb.cnf.dpkg-new` or with `.ucf-dist` file ending, then you can recover it manually:
```
mv /etc/mysql/mariadb.cnf.dpkg-new /etc/mysql/mariadb.cnf
```
Else create this default file manually:
```
cat << _EOF_ > /etc/mysql/mariadb.cnf
# The MariaDB configuration file
#
# The MariaDB/MySQL tools read configuration files in the following order:
# 1. "/etc/mysql/mariadb.cnf" (this file) to set global defaults,
# 2. "/etc/mysql/conf.d/*.cnf" to set global options.
# 3. "/etc/mysql/mariadb.conf.d/*.cnf" to set MariaDB-only options.
# 4. "~/.my.cnf" to set user-specific options.
#
# If the same option is defined multiple times, the last one will apply.
#
# One can use all long options that the program supports.
# Run program with --help to get a list of available options and with
# --print-defaults to see which it would actually understand and use.

#
# This group is read both both by the client and the server
# use it for options that affect everything
#
[client-server]

# Import all .cnf files from configuration directory
!includedir /etc/mysql/conf.d/
!includedir /etc/mysql/mariadb.conf.d/
_EOF_
```
Then try to configure the package manually: `dpkg --configure mariadb-common`
If this succeeds now, you can finish the other upgrades as well: `apt upgrade`

kirkface, 2019-09-09:
ls -l /etc/mysql/mariadb.cnf

No such file or directory

mv /etc/mysql/mariadb.cnf.dpkg-new /etc/mysql/mariadb.cnf

No such file or directory

cat << _EOF_ > /etc/mysql/mariadb.cnf

apt working now!

MichaIng, 2019-09-09:
@kirkface 
Great, why ever this file was missing. At best, just in case, check that MariaDB is working as intended:
```
systemctl status mariadb
```
respectively the software that uses a MariaDB database.