MichaIng/DietPi #3103: TasmoAdmin

svh1985, 2019-09-10:
### Creating a software request

#### Formal software information
- Software title | TasmoAdmin
- Short description | TasmoAdmin is an administrative Website for Devices flashed with Sonoff-Tasmota.
- GitHub URL (if available) | https://github.com/reloxx13/TasmoAdmin

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- Via dietpi-software is always better :) 

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
1. Install webserver stack
2. Install php-curl php-zip

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
Yes!

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/+79

svh1985, 2019-09-10:
I think I can implement this myself just need some help here and there :)
Especially with the webserver virtual dir config :)

MichaIng, 2019-09-11:
@svh1985 
Sounds interesting. Jep feel free to send a PR, which will speed up implementation of course. Btw regarding your post in the other topic, I added an updated HowTo to the wiki about how to implement software: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

svh1985, 2019-10-05:
@MichaIng So I've been working on this but for some reason my DietPi Fork is a bit messed up. My previous pull requests were already merged for the v2.26, is it safe to delete my Fork on github and fork it again for the TasmoAdmin pull request?

MichaIng, 2019-10-05:
@svh1985 
If you want to be sure that everything has been merged, you could open a test PR.
Indeed if everything has been merged, cleanest would be a fresh fork.

svh1985, 2019-10-06:
@MichaIng TasmoAdmin requires mod_rewrite, what do you suggest to do;
-virtual directory with custom settings and location eg /var/tasmoadmin (better suggestions are welcome)
-place in /var/www/tasmoadmin and change the default config by enabling mod-rewrite

MichaIng, 2019-10-06:
@svh1985 
Modules cannot be enabled per-dir, since those need to be enabled on initial daemon start. There are many other webserver applications which require mod_rewrite, so simply enable it server-wide.

The rewrite rules however should be added to a separate config file, usually this would be:
- `/etc/apache2/sites-available/dietpi-tasmoadmin.conf` on Apache + `a2ensite dietpi-tasmoadmin`
- `/etc/nginx/sites-dietpi/dietpi-tasmoadmin.conf` on Nginx
- `/etc/lighttpd/conf-available/99-dietpi-tasmoadmin.conf` on Lighttpd + `lighttpd-enable-mod dietpi-tasmoadmin`

The rewrite rules then often need to be placed outside of any location/directory block (although depends on if those are internal rewrites when accessing the web UI already or API-kind redirects so that external clients are redirected to the correct endpoints from base domain/URL), other permission-like directives then inside the `/var/www/tasmoadmin` directory, respectively `/tasmoadmin` location/URL block.

Just as an example check out the Nextcloud configs: https://github.com/MichaIng/DietPi/tree/dev/.conf/dps_114

However would be okay if you just add the config for the webserver(s) you are familiar with or have docs from TasmoAdmin, I can translate those for the other webserver(s) then.

svh1985, 2019-10-07:
> @svh1985
> Modules cannot be enabled per-dir, since those need to be enabled on initial daemon start. There are many other webserver applications which require mod_rewrite, so simply enable it [...]

I've pushed the config for Apache.
For lighthttpd I found this: https://github.com/reloxx13/TasmoAdmin/issues/155

MichaIng, 2019-10-07:
@svh1985 
Ah yeah its `.htaccess` based, so Lighttpd + Nginx configs should be based on: https://github.com/reloxx13/TasmoAdmin/blob/master/tasmoadmin/.htaccess
- Largest parts are indeed the rewrites, the issue you posted handles some of them, but not all. Requires some fiddling to get this running, so from my side nothing I can do for v6.26 release anymore, but will definitely start working on it for v6.27. We can then also contribute this to TasmoAdmin project, at least Wiki.
- What I really like, is the robots/search crawler protection. This is something we can add to other web applications as well, where you don't want content to be listed on search engines or simply reduce server load.

MichaIng, 2019-11-05:
Implementation has been merged into development branch: #3143

closed 2020-01-01 by MichaIng