MichaIng/DietPi #3105: Package 'netdata' has no installation candidate

ajouwens, 2019-09-10:
#### Details:
- Date           | Tue 10 Sep 21:12:14 BST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | ARMbian
- SBC device     | Pine A64 (aarch64) (index=40)
- Kernel version | #5.92 SMP Fri Aug 2 00:18:27 CEST 2019
- Distro         | stretch (index=4)
- Command        | G_AGI netdata
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Package 'netdata' has no installation candidate
```

MichaIng, 2019-09-11:
@ajouwens 
Many thanks for your report. Please see for solution: https://github.com/MichaIng/DietPi/issues/3062#issuecomment-524880146 (use `netdata_aarch64.7z` file in your case)
Issue has been solved for v6.26.

ajouwens, 2019-09-11:
Thanks. It's working now.

MichaIng, 2019-09-11:
Perfect, I'll mark this issue as closed then in favour of the other one. Feel free to reopen if required.

closed 2019-09-11 by MichaIng