MichaIng/DietPi #3107: invoke-rc.d: not found"

Superbroom, 2019-09-14:
#### Details:
- Date           | Sat 14 Sep 06:25:48 BST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid XU3/XU4/HC1/HC2 (armv7l) (index=11)
- Kernel version | #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018
- Distro         | stretch (index=4)
- Command        | G_AGI openvpn easy-rsa iptables
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Tried installing openvpn via dietpi-software

#### Expected behaviour:

- ... I would expect openvpn to install

#### Actual behaviour:

- ... openvpn is not installing

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Preparing to unpack .../0-libnfnetlink0_1.0.1-3_armhf.deb ...
Unpacking libnfnetlink0:armhf (1.0.1-3) ...
Selecting previously unselected package libip6tc0:armhf.
Preparing to unpack .../1-libip6tc0_1.6.0+snapshot20161117-6_armhf.deb ...
Unpacking libip6tc0:armhf (1.6.0+snapshot20161117-6) ...
Selecting previously unselected package libiptc0:armhf.
Preparing to unpack .../2-libiptc0_1.6.0+snapshot20161117-6_armhf.deb ...
Unpacking libiptc0:armhf (1.6.0+snapshot20161117-6) ...
Selecting previously unselected package libxtables12:armhf.
Preparing to unpack .../3-libxtables12_1.6.0+snapshot20161117-6_armhf.deb ...
Unpacking libxtables12:armhf (1.6.0+snapshot20161117-6) ...
Selecting previously unselected package libnetfilter-conntrack3:armhf.
Preparing to unpack .../4-libnetfilter-conntrack3_1.0.6-2_armhf.deb ...
Unpacking libnetfilter-conntrack3:armhf (1.0.6-2) ...
Selecting previously unselected package iptables.
Preparing to unpack .../5-iptables_1.6.0+snapshot20161117-6_armhf.deb ...
Unpacking iptables (1.6.0+snapshot20161117-6) ...
Selecting previously unselected package liblzo2-2:armhf.
Preparing to unpack .../6-liblzo2-2_2.08-1.2+b2_armhf.deb ...
Unpacking liblzo2-2:armhf (2.08-1.2+b2) ...
Selecting previously unselected package libpkcs11-helper1:armhf.
Preparing to unpack .../7-libpkcs11-helper1_1.21-1_armhf.deb ...
Unpacking libpkcs11-helper1:armhf (1.21-1) ...
Selecting previously unselected package openvpn.
Preparing to unpack .../8-openvpn_2.4.0-6+deb9u3_armhf.deb ...
Unpacking openvpn (2.4.0-6+deb9u3) ...
Selecting previously unselected package easy-rsa.
Preparing to unpack .../9-easy-rsa_2.2.2-2_all.deb ...
Unpacking easy-rsa (2.2.2-2) ...
Setting up libpkcs11-helper1:armhf (1.21-1) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Setting up libxtables12:armhf (1.6.0+snapshot20161117-6) ...
Processing triggers for systemd (232-25+deb9u12) ...
Setting up easy-rsa (2.2.2-2) ...
Setting up libnfnetlink0:armhf (1.0.1-3) ...
Setting up libip6tc0:armhf (1.6.0+snapshot20161117-6) ...
Setting up liblzo2-2:armhf (2.08-1.2+b2) ...
Setting up libnetfilter-conntrack3:armhf (1.0.6-2) ...
Setting up libiptc0:armhf (1.6.0+snapshot20161117-6) ...
Setting up openvpn (2.4.0-6+deb9u3) ...
/var/lib/dpkg/info/openvpn.postinst: 38: /var/lib/dpkg/info/openvpn.postinst: invoke-rc.d: not found
/var/lib/dpkg/info/openvpn.postinst: 38: /var/lib/dpkg/info/openvpn.postinst: invoke-rc.d: not found
dpkg: error processing package openvpn (--configure):
 subprocess installed post-installation script returned error exit status 127
Setting up iptables (1.6.0+snapshot20161117-6) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Processing triggers for systemd (232-25+deb9u12) ...
Errors were encountered while processing:
 openvpn
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2019-09-14:
@Superbroom 
Many thanks for your report.

Very strange, the missing binary is from the essential [init-system-helpers](https://packages.debian.org/buster/all/init-system-helpers/filelist) package, which is a MUST on all Debian-based systems and pre-installed on all DietPi images as well of course. I just verified that it is present on our Odroid XU4 image.

So no idea how, but something removed this file or you have some data loss due to disk corruption or similar.

First of all to re-create the binary, try: `apt install --reinstall init-system-helpers`
To check for any kernel errors that might be related, please check as well for red (error) messages: `dmesg`

Superbroom, 2019-09-14:
Is there any way to stop openvpn from trying to install while running the command `apt install --reinstall init-system-helpers`? 

It's throwing the same error when using this command.

`root@DietPi:~# apt install --reinstall init-system-helpers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 0 B/44.0 kB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 49811 files and directories currently installed.)
Preparing to unpack .../init-system-helpers_1.48_all.deb ...
Unpacking init-system-helpers (1.48) over (1.48) ...
Setting up init-system-helpers (1.48) ...
Setting up openvpn (2.4.0-6+deb9u3) ...
/var/lib/dpkg/info/openvpn.postinst: 38: /var/lib/dpkg/info/openvpn.postinst: invoke-rc.d: not found
/var/lib/dpkg/info/openvpn.postinst: 38: /var/lib/dpkg/info/openvpn.postinst: invoke-rc.d: not found
dpkg: error processing package openvpn (--configure):
 subprocess installed post-installation script returned error exit status 127
Errors were encountered while processing:
 openvpn
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

closed 2019-09-14 by Superbroom

MichaIng, 2019-09-15:
@Superbroom 
Ah yeah this is APT preventing install/reinstall of any packages to fix a failed package install/config state, since it wants to solve any non-finished installs first to have a consistent state.

Sadly so far I could not find e.g. an APT option for something like "ignore any pending/unfinished/failed install/config states and do ONLY the one install/reinstall I instructed": https://manpages.debian.org/apt-get
Purging the failed package (openvpn in your case) would have worked. Not sure if reinstalling `dpkg` itself somehow omits or resets pending install states, but I am glad it worked with this.

derw, 2025-03-04:
Try adding PATH=$PATH:/sbin/ to .bashrc