MichaIng/DietPi #3109: Sonnar fails to start

ezekini, 2019-09-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | Stretch 9.11
- Kernel version | Linux DietPi-rpi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | Sonarr
- Was the software title installed freshly or updated/migrated? installed
- Can this issue be replicated on a fresh installation of DietPi? Don't know

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...Sonarr starting normally

#### Actual behaviour

Sonarr service fails to start. I tried reinstalling but no luck.
When running dietpi-software status Sonarr returns:
[FAILED] DietPi-Services | ● sonarr.service - Sonarr (NzbDrone) Daemon (DietPi)
   Loaded: loaded (/etc/systemd/system/sonarr.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-09-14 14:12:25 -03; 6s ago
  Process: 9555 ExecStart=/usr/bin/mono -O=-aot /opt/NzbDrone/NzbDrone.exe -nobrowser -data=/mnt/dietpi_userdata/sonarr (code=exited, status=1/FAILURE)
 Main PID: 9555 (code=exited, status=1/FAILURE)

Sep 14 14:12:24 DietPi-rpi mono[9555]:   at NzbDrone.Core.Datastore.DbFactory.RegisterDatabase (NzbDrone.Common.Composition.IContainer container) [0x00006] in <96f52d5e5b95442a89ae3bbf62e59664>:0
Sep 14 14:12:24 DietPi-rpi mono[9555]:   at NzbDrone.Host.NzbDroneServiceFactory.Start () [0x00037] in <83751627b1fd4bb1b4a2a3075c439f56>:0
Sep 14 14:12:24 DietPi-rpi mono[9555]:   at NzbDrone.Host.Router.Route (NzbDrone.Host.ApplicationModes applicationModes) [0x0007b] in <83751627b1fd4bb1b4a2a3075c439f56>:0
Sep 14 14:12:24 DietPi-rpi mono[9555]:   at NzbDrone.Host.Bootstrap.Start (NzbDrone.Host.ApplicationModes applicationModes, NzbDrone.Common.EnvironmentInfo.StartupContext startupContext) [0x00047] in <83751627b1fd4bb1b4a2a3075c439f56>:0
Sep 14 14:12:24 DietPi-rpi mono[9555]:   at NzbDrone.Host.Bootstrap.Start (NzbDrone.Common.EnvironmentInfo.StartupContext startupContext, NzbDrone.Host.IUserAlert userAlert, System.Action`1[T] startCallback) [0x00075] in <83751627b1fd4bb1b4a2a3075c439f56>:0
Sep 14 14:12:24 DietPi-rpi mono[9555]:   at NzbDrone.Console.ConsoleApp.Main (System.String[] args) [0x0002f] in <bd424479d84b43e2be7be110e677f257>:0
Sep 14 14:12:24 DietPi-rpi mono[9555]: Press enter to exit...
Sep 14 14:12:25 DietPi-rpi systemd[1]: sonarr.service: Main process exited, code=exited, status=1/FAILURE
Sep 14 14:12:25 DietPi-rpi systemd[1]: sonarr.service: Unit entered failed state.
Sep 14 14:12:25 DietPi-rpi systemd[1]: sonarr.service: Failed with result 'exit-code'.

#### Extra details

- ...

MichaIng, 2019-09-15:
@ezekini 
There is a known issue with the Sonarr install that is fixed for v6.26: https://github.com/MichaIng/DietPi/issues/3029

Could you please try the following first:
```
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
G_AGUP
G_AGI nzbdrone
systemctl restart sonarr
systemctl status sonarr
```
If this still fails, please paste the full error log that matches the timestamps after doing the service restart above:
```
journalctl -u sonarr
```

ezekini, 2019-09-15:
I tried the steps described but the service doesn't start. This is the error log as requested. Thank you

```
sep 15 16:47:39 DietPi-rpi systemd[1]: Started Sonarr (NzbDrone) Daemon (DietPi).
sep 15 16:47:44 DietPi-rpi mono[3234]: [Info] Bootstrap: Starting Sonarr - /opt/NzbDrone/NzbDrone.exe - Version 2.0.0.5337
sep 15 16:47:47 DietPi-rpi mono[3234]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/sonarr]
sep 15 16:47:48 DietPi-rpi mono[3234]: [Debug] ProcessProvider: Found 0 processes with the name: NzbDrone.Console
sep 15 16:47:48 DietPi-rpi mono[3234]: [Debug] ProcessProvider: Found 1 processes with the name: NzbDrone
sep 15 16:47:48 DietPi-rpi mono[3234]: [Debug] ProcessProvider:  - [3234] NzbDrone
sep 15 16:47:48 DietPi-rpi mono[3234]: [Info] Router: Application mode: Interactive
sep 15 16:47:48 DietPi-rpi mono[3234]: [Debug] Router: Tray selected
sep 15 16:47:50 DietPi-rpi mono[3234]: [Info] MigrationLogger: *** Migrating data source=/mnt/dietpi_userdata/sonarr/nzbdrone.db;cache size=-10000;datetimekind=Utc;jo
urnal mode=Wal;pooling=True;version=3 ***
sep 15 16:47:56 DietPi-rpi mono[3234]: [Fatal] ConsoleApp: EPIC FAIL!
sep 15 16:47:56 DietPi-rpi mono[3234]: [v2.0.0.5337] NzbDrone.Core.Datastore.CorruptDatabaseException: Database file: /mnt/dietpi_userdata/sonarr/nzbdrone.db is corru
pt, restore from backup if available. See: https://github.com/Sonarr/Sonarr/wiki/FAQ#i-am-getting-an-error-database-disk-image-is-malformed ---> System.Data.SQLite.SQ
LiteException: database disk image is malformed
sep 15 16:47:56 DietPi-rpi mono[3234]: database disk image is malformed
sep 15 16:47:56 DietPi-rpi mono[3234]:   at System.Data.SQLite.SQLite3.Prepare (System.Data.SQLite.SQLiteConnection cnn, System.String strSql, System.Data.SQLite.SQLi
teStatement previous, System.UInt32 timeoutMS, System.String& strRemain) [0x0033c] in <61a20cde294d4a3eb43b9d9f6284613b>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at System.Data.SQLite.SQLiteCommand.BuildNextCommand () [0x000f6] in <61a20cde294d4a3eb43b9d9f6284613b>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at System.Data.SQLite.SQLiteCommand.GetStatement (System.Int32 index) [0x00008] in <61a20cde294d4a3eb43b9d9f6284613b>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at (wrapper remoting-invoke-with-check) System.Data.SQLite.SQLiteCommand.GetStatement(int)
sep 15 16:47:56 DietPi-rpi mono[3234]:   at System.Data.SQLite.SQLiteDataReader.NextResult () [0x0011e] in <61a20cde294d4a3eb43b9d9f6284613b>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at System.Data.SQLite.SQLiteDataReader..ctor (System.Data.SQLite.SQLiteCommand cmd, System.Data.CommandBehavior behave) [0x00
090] in <61a20cde294d4a3eb43b9d9f6284613b>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at (wrapper remoting-invoke-with-check) System.Data.SQLite.SQLiteDataReader..ctor(System.Data.SQLite.SQLiteCommand,System.Dat
a.CommandBehavior)
sep 15 16:47:56 DietPi-rpi mono[3234]:   at System.Data.SQLite.SQLiteCommand.ExecuteReader (System.Data.CommandBehavior behavior) [0x0000c] in <61a20cde294d4a3eb43b9d
9f6284613b>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at System.Data.SQLite.SQLiteCommand.ExecuteNonQuery (System.Data.CommandBehavior behavior) [0x00006] in <61a20cde294d4a3eb43b
9d9f6284613b>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at System.Data.SQLite.SQLiteCommand.ExecuteNonQuery () [0x00006] in <61a20cde294d4a3eb43b9d9f6284613b>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at System.Data.SQLite.SQLiteConnection.Open () [0x00959] in <61a20cde294d4a3eb43b9d9f6284613b>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at FluentMigrator.Runner.Processors.GenericProcessorBase.EnsureConnectionIsOpen () [0x00014] in <137fb96feee44f379d6a8fba4e17
2d1c>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at FluentMigrator.Runner.Processors.SQLite.SQLiteProcessor.Exists (System.String template, System.Object[] args) [0x00000] in
 <137fb96feee44f379d6a8fba4e172d1c>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at FluentMigrator.Runner.Processors.SQLite.SQLiteProcessor.TableExists (System.String schemaName, System.String tableName) [0
x00000] in <137fb96feee44f379d6a8fba4e172d1c>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at FluentMigrator.Runner.VersionLoader.get_AlreadyCreatedVersionTable () [0x0001c] in <137fb96feee44f379d6a8fba4e172d1c>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at FluentMigrator.Runner.VersionLoader.LoadVersionInfo () [0x00028] in <137fb96feee44f379d6a8fba4e172d1c>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at FluentMigrator.Runner.VersionLoader..ctor (FluentMigrator.Runner.IMigrationRunner runner, FluentMigrator.Infrastructure.IA
ssemblyCollection assemblies, FluentMigrator.IMigrationConventions conventions) [0x00077] in <137fb96feee44f379d6a8fba4e172d1c>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at FluentMigrator.Runner.MigrationRunner..ctor (FluentMigrator.Infrastructure.IAssemblyCollection assemblies, FluentMigrator.
Runner.Initialization.IRunnerContext runnerContext, FluentMigrator.IMigrationProcessor processor) [0x00175] in <137fb96feee44f379d6a8fba4e172d1c>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at FluentMigrator.Runner.MigrationRunner..ctor (System.Reflection.Assembly assembly, FluentMigrator.Runner.Initialization.IRu
nnerContext runnerContext, FluentMigrator.IMigrationProcessor processor) [0x00007] in <137fb96feee44f379d6a8fba4e172d1c>:0
[... cut ...]
rAlert, System.Action`1[T] startCallback) [0x00075] in <83751627b1fd4bb1b4a2a3075c439f56>:0
sep 15 16:47:56 DietPi-rpi mono[3234]:   at NzbDrone.Console.ConsoleApp.Main (System.String[] args) [0x0002f] in <bd424479d84b43e2be7be110e677f257>:0
sep 15 16:47:56 DietPi-rpi mono[3234]: Press enter to exit...
sep 15 16:47:57 DietPi-rpi systemd[1]: sonarr.service: Main process exited, code=exited, status=1/FAILURE
sep 15 16:47:57 DietPi-rpi systemd[1]: sonarr.service: Unit entered failed state.
sep 15 16:47:57 DietPi-rpi systemd[1]: sonarr.service: Failed with result 'exit-code'.
```

MichaIng, 2019-09-16:
@ezekini 
```
sep 15 16:47:56 DietPi-rpi mono[3234]: [v2.0.0.5337] NzbDrone.Core.Datastore.CorruptDatabaseException: Database file: /mnt/dietpi_userdata/sonarr/nzbdrone.db is corrupt, restore from backup if available. See: https://github.com/Sonarr/Sonarr/wiki/FAQ#i-am-getting-an-error-database-disk-image-is-malformed ---> System.Data.SQLite.SQ
LiteException: database disk image is malformed
```
The database file `/mnt/dietpi_userdata/sonarr/nzbdrone.db` is corrupt. Did you have a system crash or non-graceful shutdown recently?

Any backup available to restore this file from?

There is a low chance to repair by dumping and recreating it:
```
cd /mnt/dietpi_userdata/sonarr
sqlite3 nzbdrone.db ".dump" | sqlite3 new_nzbdrone.db
mv nzbdrone.db old_nzbdrone.db
mv new_nzbdrone.db nzbdrone.db
systemctl restart sonarr
```

Worst case is to remove the file, then it is re-created. But most settings and download lists are lost then, AFAIK.

ezekini, 2019-09-16:
I could restore a backup db. Thank you

closed 2019-09-16 by ezekini