MichaIng/DietPi #3111: Certbot renew failed

Phil1988, 2019-09-16:
#### Required Information
- DietPi version | 
    G_DIETPI_VERSION_CORE=6
    G_DIETPI_VERSION_SUB=25
    G_DIETPI_VERSION_RC=3
    G_GITBRANCH=master
    G_GITOWNER=MichaIng
- Distro version | stretch
- Kernel version | Linux DietPi 4.9.0-8-amd64 #1 SMP Debian 4.9.144-3.1 (2019-02-19) x86_64 GNU/Linux

- SBC device | 
Virtual Machine (x86_64)

#### Description
I am using dietpi on a VM with following programs installed:
pi-hole
nextcloud
letsencrypt
blynk server
LASP (blynk wanted it I think...)
LAMP

**The renewal worked in the past.**
DietPi-LetsEncrypt is set with 
Domain
Email
Redirect [On]
HSTS [Off]
Key Size 4096

I have updated dietpi and some of the programs (I remember at least the Pi-hole update) since then.

Now the renewal isnt working any more.

I am getting:
[image]

My Nextcloud is not (and never have been) accessable on port 80. Only on Port 443.
In the past the port 80 opened the standard apache site.
One thing I discovered is, that after the pi-hole update, I see the Pihole page on port 80.

Maybe not linked to this issue but maybe yes :D 

I have no idea where to search for the source of this problem.
Tell me what I need to provide and I will do my best :)

Thanks a lot!

MichaIng, 2019-09-16:
@Phil1988 
Many thanks for your report.

Certbot requires port 80 to use the `http-01` challenge. Please open/forward port 80 and retry `certbot renew`.

General convention to force HTTPS is to set a HTTP (port 80) => HTTPS (port 443) redirect. If you did not chose this within DietPi-LetsEncrypt before, on Apache do:
```
cat << _EOF_ > /etc/apache2/sites-available/000-default.conf
<VirtualHost *:80>
        Redirect permanent / https://your.domain.de/
</VirtualHost>
_EOF_
a2ensite 000-default
```

Having Pi-hole blocking page available on webroot is default/expected. If this is really serviced via HTTP (instead of expected HTTPS) depends on how you blocked HTTP/configured Apache.

Phil1988, 2019-09-17:
@MichaIng : Thanks a lot for your fast reply.

My port 80 is forwarded / open as I have access from the internet.
Currently I just see the warning page from my browser, saying, that this site uses HTTP Strict Transport Security (HSTS) and that the certificate has expired (Fehlercode: SEC_ERROR_EXPIRED_CERTIFICATE).

(also a quick check if HSTS to [On] on Letsencrypt could solve the issue didnt change anything)

my /etc/apache2/sites-available/000-default.conf looks like this
```
<VirtualHost *:80>
ServerName mydomain.de
        ServerAdmin webmaster@localhost
        DocumentRoot /var/www

        ErrorLog ${APACHE_LOG_DIR}/error.log
        #CustomLog ${APACHE_LOG_DIR}/access.log combined
RewriteEngine on
RewriteCond %{SERVER_NAME} =mydomain.de
RewriteRule ^ https://%{SERVER_NAME}%{REQUEST_URI} [END,QSA,R=permanent]
</VirtualHost>
```

in the letsencrypt logs (/var/log/letsencrypt/letsencrypt.log) I can see:
```
2019-09-17 08:07:39,104:DEBUG:acme.client:Received response:
HTTP 200
Server: nginx
```
I thought it should be apache instead of nginx?

on the renewal via the dietpi-launcher -> Letsencrypt, I can see 
`[  OK  ] DietPi-LetsEncrypt | Apache2 webserver detected`

I am still having no idea why the renewal process got broken. 
I can just imagine that installation of blynk server (which installed LASP) maybe has destroyed some configurations...

MichaIng, 2019-09-17:
@Phil1988 
If the browser states that HSTS is enabled, then changing something in `dietpi-letsencrypt` has no effect anymore. However HSTS is fine as long as you do not want to access with HTTP by times with the same browsers/clients, and HSTS does not break `certbot`, I use it myself with Apache webserver.

> Server: nginx

That is the ACME server, not your webserver 😉.

As long as you can access to your webserver by entering the bare domain (that you used for the certificate) into any web browser with `http://` (+ being redirected) and `https://`, certbot should work as well.

Please verify the following:
- `ls -l /var` lists `www` as writeable by the `www-data` user.
- Apache is running while you try to run `certbot renew`.
- `cat /etc/letsencrypt/renewal/you.domain.de.conf` shows:
```
authenticator = apache
installer = apache
```

Phil1988, 2019-09-17:
@MichaIng  
Thanks for your reply

> > Server: nginx
> 
> That is the ACME server, not your webserver wink.
good to know :D

- `ls -l /var` lists:
`drwxrwxr-x  5 www-data www-data  4096 Jul 20 03:42 www`
- `sudo systemctl status apache2` tells me thats apache is running.
- `cat /etc/letsencrypt/renewal/ourpicloud.spdns.de.conf`
tells me:

```
# renew_before_expiry = 30 days
version = 0.10.2
archive_dir = /etc/letsencrypt/archive/my.domain.de
cert = /etc/letsencrypt/live/my.domain.de/cert.pem
privkey = /etc/letsencrypt/live/my.domain.de/privkey.pem
chain = /etc/letsencrypt/live/my.domain.de/chain.pem
fullchain = /etc/letsencrypt/live/my.domains.de/fullchain.pem

# Options used in the renewal process
[renewalparams]
account = b23c4cacbe12d4244abca4bc5d5837ge
authenticator = webroot
rsa_key_size = 4096
installer = apache
[[webroot_map]]
my.domain.de = /var/www
```
so the authenticator = webroot and not apache...

is that a hit? If yes.. should I change it to apache and try to renew?

MichaIng, 2019-09-17:
@Phil1988 
Ahh, I just recognise that you use a veeery outdated version of certbot and most likely of DietPi then as well?

Please consider to upgrade to current version: `dietpi-update 1`
Otherwise at least update your certbot packages: `apt install python3-certbot-apache`

The webroot authentication method should work as well, we applied this during some time since the Apache+Nginx plugins failed with the message you got as well, which is due to the usage of a protocol (TLS-SNI) that was rendered as insecure and dropped by Let's Encrypt. However on current certbot versions, those webserver specific plugins got updated to again work well.

Phil1988, 2019-09-17:
I will do so and report here back..

manwhile: the version of dietpi:
`DietPi v6.25.3 : 13:04 - Di 17.09.2019
 ─────────────────────────────────────────────────────
 - Device model : Virtual Machine (x86_64)
 - CPU temp : N/A
 - LAN IP : 192.168.1.114 (eth0)
 - MOTD : If G_AGUP fails with "https://apt.sonarr.tv":
          https://github.com/MichaIng/DietPi/issues/3029`

MichaIng, 2019-09-17:
@Phil1988 
Hmm, do you run a Debian Jessie image?

Phil1988, 2019-09-17:
Debian yes... Jessie... I dont know :D
` DietPi Team     : MichaIng (lead), Daniel Knight (founder)
 Image           : DietPi Core Team (pre-image: Debian)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : PINE64 community
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com`

MichaIng, 2019-09-17:
Ah even there v0.28 is present, which already contains the fix. However if your image is old, certbot was probably installed from source and not via APT package that time, so `/etc/certbot_scripts/certbot-auto` would be the binary.

To check your Debian version: `echo $G_DISTRO_NAME`

Phil1988, 2019-09-17:
```
echo $G_DISTRO_NAME
stretch
```

I also did run 
> apt install python3-certbot-apache

but checking `sudo nano /etc/letsencrypt/renewal/ourpicloud.spdns.de.conf`
still gives me the version = 0.10.2

MichaIng, 2019-09-17:
@Phil1988 
Okay, what does `certbot --version` say?

Phil1988, 2019-09-17:
its says certbot 0.28.0

MichaIng, 2019-09-17:
@Phil1988 
Nope, is not required then.
Please try again: `certbot renew`

Phil1988, 2019-09-17:
Still gives me:

```
Saving debug log to /var/log/letsencrypt/letsencrypt.log

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Processing /etc/letsencrypt/renewal/my.domain.de.conf
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Cert is due for renewal, auto-renewing...
Plugins selected: Authenticator webroot, Installer apache
Renewing an existing certificate
Performing the following challenges:
http-01 challenge for my.domain.de
Waiting for verification...
Cleaning up challenges
Attempting to renew cert (my.domain.de) from /etc/letsencrypt/renewal/my.domain.de.conf produced an unexpected error: Failed authorization procedure. my.domain.de (http-01): urn:ietf:params:acme:error:connection :: The server could not connect to the client to verify the domain :: Fetching https://my.domain.de/.well-known/acme-challenge/oe1iyGiGODUdjUQDfnUHKUFo2CQO7xNh2LPCl6BQEOY: Timeout during connect (likely firewall problem). Skipping.
All renewal attempts failed. The following certs could not be renewed:
  /etc/letsencrypt/live/my.domain.de/fullchain.pem (failure)

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -

All renewal attempts failed. The following certs could not be renewed:
  /etc/letsencrypt/live/my.domain.de/fullchain.pem (failure)
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
1 renew failure(s), 0 parse failure(s)

IMPORTANT NOTES:
 - The following errors were reported by the server:

   Domain: my.domain.de
   Type:   connection
   Detail: Fetching
   https://my.domain.de/.well-known/acme-challenge/oe1iyGiGODUdjUQDfnUHKUFo2CQO7xNh2LPCl6BQEOY:
   Timeout during connect (likely firewall problem)

   To fix these errors, please make sure that your domain name was
   entered correctly and the DNS A/AAAA record(s) for that domain
   contain(s) the right IP address. Additionally, please check that
   your computer has a publicly routable IP address and that no
   firewalls are preventing the server from communicating with the
   client. If you're using the webroot plugin, you should also verify
   that you are serving files from the webroot path you provided.
```

MichaIng, 2019-09-17:
@Phil1988 
It tries webroot authentication again. Please try: `certbot renew --apache`

Phil1988, 2019-09-17:
Ohhh my god.. you are godlike... 
so "oh my MichaIng!"
You did it!

I changed 

```
#authenticator = webroot
authenticator = apache
```
in /etc/letsencrypt/renewal/ourpicloud.spdns.de.conf

and then did 

> certbot renew

result: 
```
Congratulations, all renewals succeeded. The following certs have been renewed:
  /etc/letsencrypt/live/ourpicloud.spdns.de/fullchain.pem (success)
```

Thanks a lot dude! You are awesome :)
How can I buy you a beer (or tea if you prefer? ;) )

MichaIng, 2019-09-17:
@Phil1988 
Great, still not sure why webroot authentication did not work, however no issue as long as the native Apache plugin finally does its job 😃.

> How can I buy you a beer (or tea if you prefer? ;) )

Paypal donations are always open, see e.g. "Sponsors" link at the top of GitHub page, currently its mostly coffee 😉. However I'm always happy to help and in many cases there is some positive outcome for DietPi as well:
- With v6.26 patch, assure that all users are on current certbot packages and for Nginx+Apache users, if currently webroot authentication is in use, offer to switch to Nginx/Apache plugin authentication automatically.

MichaIng, 2019-09-17:
I reopen this issue to not forget implementing the above mentioned steps.

MichaIng, 2019-09-18:
Done: https://github.com/MichaIng/DietPi/commit/eafdd3b7671fff2b4cfdc48d0949ee2cd434fe97

Phil1988, 2019-09-19:
Well done dude !

MichaIng, 2019-09-19:
@Phil1988 
I will do a final test (the migration prompt) then close.

closed 2019-10-15 by MichaIng