MichaIng/DietPi #3113: Software request: AlgoVPN

xmijo, 2019-09-17:
### Creating a software request

#### Formal software information
- Software title | Algo VPN
- Short description | Algo VPN simplifies the setup of a personal IPSEC and/or Wireguard VPN. It uses the most secure defaults available, works with common cloud providers, and does not require client software on most devices. Basically, Algo automatically creates a Wireguard VPN server on your VPS, on-the-fly. 

- Official URL (if available) | https://github.com/trailofbits/algo
- GitHub URL (if available) | https://github.com/trailofbits/algo
- (Official) install documentation | https://github.com/trailofbits/algo

#### Are there similar/alternative software titles available with DietPi-Software?
NordVPN seems to be available in DietPi software but it is a 3rd party VPN provider, and not really the same as Algo at all.

#### What makes your requested software better than the above solutions, if available?
It is fully open source, and allows you to not have to rely on 3rd party VPN providers. 3rd party VPN providers may state they do not log their users' traffic but it is impossible to verify their claim. Algo allows you to deploy a Wireguard server to a VPS and works with many popular VPS providers such as DigitalOcean or Vultr, or basically other any other VPS running Debian. It is very simple to install.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
Perhaps making sure that Python, virtualenv, and other dependencies are installed before running the Algo scripts. The Readme on Algo's Github page says it needs to be Python2 but I think Python3 works nowadays and the Readme just hasn't been updated yet.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
The instructions on their github readme outlines the steps and commands required to deploy Algo to your VPS.

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/
Added on Feathub: https://feathub.com/MichaIng/DietPi/+120

MichaIng, 2019-09-17:
@xmijo 
Many thanks for your request.

Looks nice, however Python + virtualenv, uff heavy metal, not to forget the the WireGuard and IPSec server install that is still required aside of it. It takes care about the client configs only, as far as I could see.

There has been another suggestion for `wg-dashboard`: https://github.com/MichaIng/DietPi/issues/3100#issuecomment-529179023
It provides WireGuard client setup via web interface and seems to be much more lightweight. At least something that could be compared, also in combination with the WireGuard install that we already provide.

xmijo, 2019-09-17:
@MichaIng 
You're right and I agree with you-- Python is indeed a pretty bulky dependency to have installed just to make the Algo script work. I hadn't considered that because I tend to have Python on my systems, but for those that don't it could be a bit heavy.

But the Wireguard server installs to your VPS, not to your Dietpi system. Allow me to clarify if I was a bit unclear-- What Algo does when you launch the script locally is to deploy itself to a VPS server of your choice (e.g. DigitalOcean) and allow you to use it as a VPN, on-the-fly. It is a very simple automated process and you basically just provide Algo with your server API key. After server deployment you use the Wireguard client files with your local Wireguard client software and can connect to your server. You then have your own personal VPN server and do not have to rely on 3rd party VPN providers. The advantages to using a 3rd-party VPN server are, besides better privacy, it offers flexibility in terms of providers (many VPS providers out there), and is often cheaper (not to mention some of the larger VPS providers offer discount coupons).

It is of course possible to deploy Algo locally too, creating a home VPN server out of your Dietpi installation which can be accessed when you're away from home.

But the more I think about it now I'm not sure if it would be a good addition to the Dietpi software list. While it is very user-friendly and simple to install it, it isn't 100% automatic and requires a few user choices during configuration (which are simple choices for me and you but maybe not for everyone).

So I'll just go ahead and close this software request but feel free to re-open it later if there's further interest.

MichaIng, 2019-09-17:
@xmijo 
Thanks for clarification. I reopen the issue since with a guided server install at least it becomes quite interesting. It would be not the only software we offer that requires manual user input, especially for a VPN, so no issue with that.

A VPS makes much sense in many cases in comparison to a local VPN server that relies on network/bandwidth access/limitations and will always reveal your home IP. So a home VPN is basically another use case compared to a 3rd party VPN provider and a server on a VPS with public/shared IP ranges on different location.

Since you added it to FeatHub, lets see how other user interest is, IMO at least it would make the VPN use case offer complete. Only thing is that I have no otherwise unused VPS to test with.