MichaIng/DietPi #3115: Chromium Browser cannot be installed for Kiosk mode.

RaveGun, 2019-09-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.25.3`
- Distro version | `Buster/sid`
- Kernel version | `Linux Dash 4.19.66-v7+`
- SBC device | `Unknown Device (armv71)` / RPi2
- Power supply used | 5V 2A Generic
- SDcard used | SanDisk ultra 

#### Additional Information (if applicable)
- Software title | Chromium Browser
- Was the software title installed freshly or updated/migrated? Updated
- Can this issue be replicated on a fresh installation of DietPi? I think so.

#### Steps to reproduce

1. Setup Autostart in Kiosk mode with chromium.
2. Wait for chromium to install.

#### Expected behaviour

- Chomium browser shall be installed.
- The device shall restart and open Chromium.

#### Actual behaviour

- Installing of the chromium-browser package fails due to dependencies.

#### Extra details

- The chromium-browser package requires `libstdc++6` >7
- There is only one chromium available verision: `74.0.3729.157-rpt5 armf`
- And there is only the libstdc++6 version: `6.3.0-18+rpi1+deb9u1 armf`

MichaIng, 2019-09-18:
@RaveGun 
Many thanks for your report.

We will investigate.

RaveGun, 2019-09-18:
Hi,

Hmm, I installed this long time ago and since then it was hanging on a wall displaying some webpage (I hat it running in Kiosk mode). 

I ran now and then the dietpie-update. But I don't remember that the last time I run the update to have any issues, but I did this remotely on my phone and not really paying attention. Actually I don't think I restarted or powered off the board since then.

Now that I restarted it I got the an error while it was starting the X server complaining about the unknown --kiosk-mode. Digging through the start script I found out that chromium-browser was not present in the system.

Here is the output of that apt policy.

```
root@Dash:~# apt policy chromium-browser
chromium-browser:
  Installed: (none)
  Candidate: 74.0.3729.157-rpt5
  Version table:
     74.0.3729.157-rpt5 500
        500 https://archive.raspberrypi.org/debian buster/main armhf Packages
```

Also:
```
root@Dash:~# apt policy libstdc++6
libstdc++6:
  Installed: 6.3.0-18+rpi1+deb9u1
  Candidate: 6.3.0-18+rpi1+deb9u1
  Version table:
     9.2.1-8 -1
         -1 https://deb.debian.org/debian sid/main armhf Packages
 *** 6.3.0-18+rpi1+deb9u1 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status
```

RaveGun, 2019-09-18:
Hi,

I also listed this:
```
root@Dash:~# apt-cache policy
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 https://archive.raspberrypi.org/debian buster/main armhf Packages
     release o=Raspberry Pi Foundation,a=testing,n=buster,l=Raspberry Pi Foundation,c=main,b=armhf
     origin archive.raspberrypi.org
 500 https://repos.influxdata.com/debian stretch/stable armhf Packages
     release o=InfluxDB,n=stretch,l=InfluxDB,c=stable,b=armhf
     origin repos.influxdata.com
 500 https://packages.grafana.com/oss/deb stable/main armhf Packages
     release o=grafana stable,a=stable,n=stable,l=grafana stable,c=main,b=armhf
     origin packages.grafana.com
 500 https://download.docker.com/linux/raspbian stretch/edge armhf Packages
     release o=Docker,a=stretch,l=Docker CE,c=edge,b=armhf
     origin download.docker.com
  -1 https://deb.debian.org/debian sid/main armhf Packages
     release o=Debian,a=unstable,n=sid,l=Debian,c=main,b=armhf
     origin deb.debian.org
 500 http://raspbian.raspberrypi.org/raspbian stretch/rpi armhf Packages
     release o=Raspbian,a=oldstable,n=stretch,l=Raspbian,c=rpi,b=armhf
     origin raspbian.raspberrypi.org
 500 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages
     release o=Raspbian,a=oldstable,n=stretch,l=Raspbian,c=non-free,b=armhf
     origin raspbian.raspberrypi.org
 500 http://raspbian.raspberrypi.org/raspbian stretch/contrib armhf Packages
     release o=Raspbian,a=oldstable,n=stretch,l=Raspbian,c=contrib,b=armhf
     origin raspbian.raspberrypi.org
 500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
     release o=Raspbian,a=oldstable,n=stretch,l=Raspbian,c=main,b=armhf
     origin raspbian.raspberrypi.org
Pinned packages:
     wireguard-dkms -> 0.0.20190905-1 with priority 100
     wireguard-tools -> 0.0.20190905-1 with priority 100
     wireguard -> 0.0.20190905-1 with priority 100
     docker-ce -> 5:18.09.0~3-0~raspbian-stretch with priority -1
```

MichaIng, 2019-09-19:
@RaveGun 
Found the issue:
- Raspbian Stretch on the one hand.
- But the RPi-specific firmware repo is Buster.
- This leads to Chromium install failure because it requires Buster packages while only Raspbian Stretch is available.
- There is Debian Sid as well, but this is only for WireGuard, and besides those three packages, the whole repo is marked with priority `-1` as intended, so nothing else can be installed from there.

Failsafe: To assure which base packages are in use (since you pasted `Buster/sid` above):
```
cat /etc/debian_version
apt policy base-files
```
- Actually this cannot be something else then Stretch.

Then adjust the RPi-specific firmware repo to be Stretch instead of Buster and retry the Chromium install:
```
sed -i 's/buster/stretch/g' /etc/apt/sources.list.d/raspi.list
G_AGUP
apt install chromium-browser
```

RaveGun, 2019-09-19:
Hi,
Thank you for your support.

This fixed the Chormium dependencies issue, but before I opened the issue I broke something in the process. I am getting now error related to missing `locale`.

Thanks again.

RaveGun, 2019-09-20:
Hi,
Thanks. the --reinstall did the trick.
Thank you.

closed 2019-09-20 by RaveGun