MichaIng/DietPi #3120: DietPi-Software | not required HTTP server

AlexanderShniperson, 2019-09-20:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
Yes, when install DietPi i haven't choice to don't install any HTTP server

#### Describe the solution you'd like:
Add option to do not install HTTP server

#### Describe alternatives you've considered:
...

#### Additional context
...

MichaIng, 2019-09-21:
@AlexanderShniperson 
Note that the chosen webserver preference is not installed, as long as you don't choose any other software title that requires a webserver. It is just a preference that will be respected in case required. But yeah, maybe we should make this point clearer right from the main menu? The dialog after selecting the preference however is very clear IMO?
> Webserver Preference Changed:
>
> Apache2 has been selected as your webserver preference.
>
> When you select any software for install that requires a webserver, DietPi will automatically install your preferred choice (Apache2).

AlexanderShniperson, 2019-09-23:
Hi, looks like i somehow missed this note, yes it is clear

MichaIng, 2019-09-23:
@AlexanderShniperson 
We'll add a desktop preference as well soon, which will behave similar then. When doing this we'll assure that the result/handling of this selection is very clear right from the main menu.

MichaIng, 2019-09-30:
Okay I mark this as closed then. Feel free to reopen if required.

closed 2019-09-30 by MichaIng