MichaIng/DietPi #3121: DietPi-Config | Impossible to configure Hotspot

AlexanderShniperson, 2019-09-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | CORE-6  SUB-25  RC-3
- Distro version | 10.1
- Kernel version | 4.19.66-v7+ Aug 15 11:49:46
- SBC device | RPi 3 Model A+
- Power supply used | 5V 1A RAVpower
- SDcard used | Transcend 8Gb

#### Additional Information (if applicable)
- Software title | WiFi Hotspot setup
- Was the software title installed freshly or updated/migrated? - Fresh install
- Can this issue be replicated on a fresh installation of DietPi? - Yes

#### Steps to reproduce
1. at installation step try to setup Hotspot

#### Expected behaviour
- Hotspot setup success

#### Actual behaviour
- impossible to setup because there is no Ethernet port to connect to cable

#### Extra details
- just need to setup Hotspot without internet connection

MichaIng, 2019-09-21:
@AlexanderShniperson 
Many thanks for your request.

Jep this is a known limitation of our WiFi Hotspot setup currently. This is especially true since DietPi does not really work well without any internet access, thus your setup will lead to several error messages and hangs on boot currently, without some manual adjustments. An offline mode is planned, however, as well as a rework of our network config feature to allow more flexibility.

MichaIng, 2021-06-02:
Marking this as closed in favour of #2737, which covers exactly the request and has much information collected already.

closed 2021-06-02 by MichaIng