MichaIng/DietPi #3123: Dietpi-software | Sonarr | Issue Installing Sonarr via dietpi-software

paulmceneaney, 2019-09-21:
### Creating a bug report/issue

**Linux novice so please be gentle**

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | 10.1
- Kernel version | Linux DietPi 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux
- SBC device | RPi4 4Gb
- Power supply used | 5V 2.5A UK power supply
- SDcard used | SanDisk ultra 32Gb

#### Additional Information (if applicable)
- Software title | Sonarr
- Was the software title installed freshly or updated/migrated? - Fresh install
- Can this issue be replicated on a fresh installation of DietPi? - Yes
< 97dfdac8-0dc5-477b-9530-14210379e391>
- Bug report ID | reference code: 97dfdac8-0dc5-477b-9530-14210379e391

#### Steps to reproduce

1. ... Using the `dietpi-software` command i tried to install Sonarr but i failed to download due to an unrecognised pub key
2. ... Shrugged it off and tried to reinstall OctoPrint due to no stream coming from webcam
3. ... So, Using the `dietpi-software` command, select OctoPrint and get the previous error relating to not being able to install Sonar due to an unrecognised pub key
4. ... As far as I can tell, OctoPrint failed to reinstall and the `dietpi-software` launcher closed back to the terminal

#### Expected behaviour

- ...  Sonarr should have installed
- ... Octoprint should have reinstalled and my webcam stream would have appeared successfully

#### Actual behaviour

- ... Trying to dowload any new software using the `dietpi-software` command seems to only produce an error relating to not being able to install Sonarr due to an unrecognised pub key

#### Extra details

- ... ¯\_(ツ)_/¯

MichaIng, 2019-09-21:
@paulmceneaney 
Many thanks for your report. Please see here for solution: https://github.com/MichaIng/DietPi/issues/3029

paulmceneaney, 2019-09-21:
Many thanks. Solution fixed problem.

closed 2019-09-21 by paulmceneaney