MichaIng/DietPi #3128: Dietpi-Software | Docker does not start after install

Sp0rTB4cK, 2019-09-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | 

> G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version |

>  buster

- Kernel version | 

> Linux DietPi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux

- SBC device | 

> RPi B (armv6l)

- Power supply used | 

> 5V 2.1A

- SDcard used | 

> SanDisk ultra 32GB

#### Additional Information (if applicable)
- Software title | 

> Docker

- Was the software title installed freshly or updated/migrated?

> freshly installed on a fresh DietPi installation

- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce
1. Install (162) Docker using dietpi-software
2. execute "docker version" or an other docker command

#### Expected behaviour
- docker command runs correctly

#### Actual behaviour
-with root: "docker version" shows some information AND that the docker daemon is not running.
-with dietpi: "docker version" shows some information AND 

> Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get http://%2Fvar%2Frun%2Fdocker.sock/v1.40/version: dial unix /var/run/docker.sock: connect: permission denied

#### Extra details
- systemctl shows containerd.service as failed.
- docker service status:

> docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; disabled; vendor preset: enabled)
  Drop-In: /lib/systemd/system/docker.service.d
           └─dietpi-simple.conf
   Active: inactive (dead) since Sat 2019-09-28 13:37:58 BST; 27s ago
     Docs: https://docs.docker.com
  Process: 1051 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=killed, signal=TERM)
 Main PID: 1051 (code=killed, signal=TERM)

Sep 28 13:37:57 DietPi systemd[1]: Started Docker Application Container Engine.
Sep 28 13:37:58 DietPi systemd[1]: Stopping Docker Application Container Engine...
Sep 28 13:37:58 DietPi systemd[1]: docker.service: Main process exited, code=killed, status=15/TERM
Sep 28 13:37:58 DietPi systemd[1]: docker.service: Succeeded.
Sep 28 13:37:58 DietPi systemd[1]: Stopped Docker Application Container Engine.

- containerd service status:

> containerd.service - containerd container runtime
   Loaded: loaded (/lib/systemd/system/containerd.service; enabled; vendor preset: enabled)
   Active: failed (Result: core-dump) since Sat 2019-09-28 13:38:17 BST; 56s ago
     Docs: https://containerd.io
  Process: 1074 ExecStartPre=/sbin/modprobe overlay (code=exited, status=0/SUCCESS)
  Process: 1075 ExecStart=/usr/bin/containerd (code=dumped, signal=SEGV)
 Main PID: 1075 (code=dumped, signal=SEGV)

Sep 28 13:38:17 DietPi systemd[1]: Starting containerd container runtime...
Sep 28 13:38:17 DietPi systemd[1]: Started containerd container runtime.
Sep 28 13:38:17 DietPi systemd[1]: containerd.service: Main process exited, code=dumped, status=11/SEGV
Sep 28 13:38:17 DietPi systemd[1]: containerd.service: Failed with result 'core-dump'.

MichaIng, 2019-09-28:
@Sp0rTB4cK
Many thanks for your report. 

Please see the following issue with solution: https://github.com/MichaIng/DietPi/issues/3126

Sp0rTB4cK, 2019-09-28:
Thank you for your reply. 

In issue #3126 , a restart fixed the problem. Unfortunately a restart does not bring any success for me. The problem persists.

For me it looks as if less docker is the problem but containerd.

MichaIng, 2019-09-28:
Hmm, there was an issue that the Docker repo did not support ARMv6 in between (indeed making containerd fail), but AFAIK this was solved before the version for Raspbian Buster became stable.
Here is the thread around the Raspbian Buster build: https://github.com/docker/for-linux/issues/709

Did you install Docker directly on first boot or separately after first run setup and one reboot was finished? In first case I am not sure first why the install even succeeded or if it actually failed and if it succeeded, which kernel version/modules it was built against.

What is at least worth to try is to uninstall and reinstall it again one time to assure it is build against the currently loaded kernel version.

With v6.26 we changed it a way that the install is skipped if loaded and installed kernel do not match, which should work around the issue.

Sp0rTB4cK, 2019-09-28:
I tried both ways. Both the first boot variant and the subsequent installation have the same result.

I uninstalled Docker again using dietpi software and reinstalled it. I rebooted between each step. Unfortunately I could not achieve any improvement.

I read the comments in the issue you linked and found a working solution:
https://github.com/docker/for-linux/issues/709#issuecomment-516496192

MichaIng, 2019-09-29:
@Sp0rTB4cK 
Indeed it seems that it is not yet solved for ARMv6. See here the workaround with a 3rd-party containerd.io package: https://github.com/docker/for-linux/issues/709#issuecomment-536295224

Hmm, if this is not solved until v6.26 release, I will add this workaround to DietPi-Software as well, but lets hope for the best...

MichaIng, 2019-10-10:
@Sp0rTB4cK 
Now the issue seems to have been fixed: https://github.com/docker/for-linux/issues/709#issuecomment-540656802

Sp0rTB4cK, 2019-10-11:
I'll verify that in the next days.

closed 2019-10-15 by MichaIng

Sp0rTB4cK, 2019-11-22:
I know, its late, but I've found the time to test it again. Docker works perfectly now.

MichaIng, 2019-11-22:
@Sp0rTB4cK 
Many thanks for verifying!