MichaIng/DietPi #3130: Radarr crush on start

Dr0bac, 2019-09-28:
Radarr fail to start, crush after few seconds after `service radarr start`.

when i enter `dietpi-services status` i get

```
root@DietPi:~# dietpi-services status
[  OK  ] DietPi-Services | Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | nmbd                 active (running) since Sat 2019-09-28 20:38:12 CEST; 6min ago
[  OK  ] DietPi-Services | smbd                 active (running) since Sat 2019-09-28 20:38:13 CEST; 6min ago
[  OK  ] DietPi-Services | transmission-daemon  active (running) since Sat 2019-09-28 20:38:13 CEST; 6min ago
[  OK  ] DietPi-Services | plexmediaserver      active (running) since Sat 2019-09-28 20:38:13 CEST; 6min ago
[FAILED] DietPi-Services | ● radarr.service - Radarr Daemon (DietPi)
   Loaded: loaded (/etc/systemd/system/radarr.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-09-28 20:43:39 CEST; 48s ago
  Process: 1018 ExecStart=/usr/bin/mono -O=-aot /opt/Radarr/Radarr.exe -nobrowser -data=/mnt/dietpi_userdata/radarr (code=exited, status=1/FAILURE)
 Main PID: 1018 (code=exited, status=1/FAILURE)

Sep 28 20:43:38 DietPi mono[1018]:   at NzbDrone.Core.Datastore.DbFactory.Create (NzbDrone.Core.Datastore.Migration.Framework.MigrationType migrationType) [0x00010] in <28071e62f2f449e682962e05af20a3d7>:0
Sep 28 20:43:38 DietPi mono[1018]:   at NzbDrone.Core.Datastore.DbFactory.RegisterDatabase (NzbDrone.Common.Composition.IContainer container) [0x00006] in <28071e62f2f449e682962e05af20a3d7>:0
Sep 28 20:43:38 DietPi mono[1018]:   at Radarr.Host.NzbDroneServiceFactory.Start () [0x00012] in <0c461a949a3646a7a450363f6bc292ce>:0
Sep 28 20:43:38 DietPi mono[1018]:   at Radarr.Host.Router.Route (Radarr.Host.ApplicationModes applicationModes) [0x00067] in <0c461a949a3646a7a450363f6bc292ce>:0
Sep 28 20:43:38 DietPi mono[1018]:   at Radarr.Host.Bootstrap.Start (Radarr.Host.ApplicationModes applicationModes, NzbDrone.Common.EnvironmentInfo.StartupContext startupContext) [0x00047] in <0c461a949a3646a7a450363f6bc292ce>:0
Sep 28 20:43:38 DietPi mono[1018]:   at Radarr.Host.Bootstrap.Start (NzbDrone.Common.EnvironmentInfo.StartupContext startupContext, Radarr.Host.IUserAlert userAlert, System.Action`1[T] startCallback) [0x00075] in <0c461a949a3646a7a450363f6bc292ce>:0
Sep 28 20:43:38 DietPi mono[1018]:   at NzbDrone.Console.ConsoleApp.Main (System.String[] args) [0x0002f] in <105dfd66454c4410adbe72890a95c7cb>:0
Sep 28 20:43:38 DietPi mono[1018]: Press enter to exit...
Sep 28 20:43:39 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=1/FAILURE
Sep 28 20:43:39 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
[  OK  ] DietPi-Services | cron                 active (running) since Sat 2019-09-28 20:38:13 CEST; 6min ago
[  OK  ] DietPi-Services | dropbear             active (running) since Sat 2019-09-28 20:38:11 CEST; 6min ago
[  OK  ] DietPi-Services | dietpi-ramdisk       active (exited) since Sat 2019-09-28 20:37:51 CEST; 6min ago
[  OK  ] DietPi-Services | dietpi-ramlog        active (exited) since Sat 2019-09-28 20:37:51 CEST; 6min ago
[  OK  ] DietPi-Services | dietpi-preboot       active (exited) since Sat 2019-09-28 20:37:53 CEST; 6min ago
[  OK  ] DietPi-Services | dietpi-boot          active (exited) since Sat 2019-09-28 20:38:11 CEST; 6min ago
[  OK  ] DietPi-Services | dietpi-postboot      active (exited) since Sat 2019-09-28 20:38:11 CEST; 6min ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor  inactive (dead)
```

Dr0bac, 2019-09-28:
Problem was with new installation of radarr and old radarr database..

closed 2019-09-28 by Dr0bac

MichaIng, 2019-10-02:
@Dr0bac 
How did you solve it, simply removing the old database? Does Radarr not do internal database migration when being updated or have a script for this?

Dr0bac, 2019-10-02:
Yes, just deleted radarr folder and started radarr again.. Problem was i think, that on old system i got micro sd corrupted and it didn't bring back database files from ram on shutdown.. so i didn't had database files anymore just symlinks in radarr folder.. i had .bak files in folder, maybe renaming them should fix the problem, but nevermind i wanted clean radarr install..

MichaIng, 2019-10-03:
@Dr0bac 
The .bak files are btw restored automatically when dietpi-arr_to_RAM starts and does not find any existing database files to link to RAM.