MichaIng/DietPi #3132: DietPi-Software / Wireguard does not install

guineapig69, 2019-10-01:
--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Tue  1 Oct 16:15:22 BST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1270 SMP Tue Sep 24 18:45:11 BST 2019
- Distro         | buster (index=5)
- Command        | G_AGI wireguard iptables qrencode
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

Use dietpi-software to install wireguard

#### Expected behaviour:

install wireguard

#### Actual behaviour:
the script is interrupted
- ...

#### Extra details:
see below

#### Additional logs:
```
Log file contents:
Setting up wireguard-dkms (0.0.20190913-1) ...
Removing old wireguard-0.0.20190913 DKMS files...

------------------------------
Deleting module version: 0.0.20190913
completely from the DKMS tree.
------------------------------
Done.
Loading new wireguard-0.0.20190913 DKMS files...
It is likely that 4.19.75-v7+ belongs to a chroot's host
Building for 4.19.75+, 4.19.75-v7+, 4.19.75-v7l+ and 4.19.75-v8+
Building initial module for 4.19.75+
Error! Bad return status for module build on kernel: 4.19.75+ (armv7l)
Consult /var/lib/dkms/wireguard/0.0.20190913/build/make.log for more information.
dpkg: error processing package wireguard-dkms (--configure):
 installed wireguard-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of wireguard:
 wireguard depends on wireguard-dkms (= 0.0.20190913-1) | wireguard-modules (= 0.0.20190913); however:
  Package wireguard-dkms is not configured yet.
  Package wireguard-modules is not installed.

dpkg: error processing package wireguard (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 wireguard-dkms
 wireguard
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2019-10-01:
@guineapig69 
Many thanks for your report.

Is there any reason why you updated kernel (via `rpi-update`?) to unstable v4.19.75? It is unlikely, but might be probably the reason for your issue.

Please paste the log as suggested by the error output:
```
cat /var/lib/dkms/wireguard/0.0.20190913/build/make.log
```

I will also run some test install.

theRAAPster, 2019-10-10:
I get the same error on kernel 4.19.57. My make.log:

`
root@DietPi:~# cat /var/lib/dkms/wireguard/0.0.20190913/build/make.log
DKMS make.log for wireguard-0.0.20190913 for kernel 4.19.75+ (armv7l)
Fri 11 Oct 00:34:53 BST 2019
make: Entering directory '/usr/src/linux-headers-4.19.75+'
  CC [M]  /var/lib/dkms/wireguard/0.0.20190913/build/main.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190913/build/noise.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190913/build/device.o
  CC [M]  /var/lib/dkms/wireguard/0.0.20190913/build/peer.o
/bin/sh: 1: /bin/sh: 1: scripts/basic/fixdep: Exec format errorscripts/basic/fixdep: Exec format error

make[1]: *** [scripts/Makefile.build:303: /var/lib/dkms/wireguard/0.0.20190913/build/main.o] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:303: /var/lib/dkms/wireguard/0.0.20190913/build/peer.o] Error 2
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[1]: *** [scripts/Makefile.build:303: /var/lib/dkms/wireguard/0.0.20190913/build/device.o] Error 2
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[1]: *** [scripts/Makefile.build:303: /var/lib/dkms/wireguard/0.0.20190913/build/noise.o] Error 2
make: *** [Makefile:1519: _module_/var/lib/dkms/wireguard/0.0.20190913/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.19.75+'
`

This is 100% repeatable for me on a brand new install of dietpi on a new Raspberry Pi 4, following the getting started guide here: https://dietpi.com/phpbb/viewtopic.php?p=9#p9 (skipping optional steps), and selecting only wireguard to install.

Unfortunately this is the same error I get when trying to install wireguard on plain raspbian, following a guide like [this](https://www.reddit.com/r/pihole/comments/bnihyz/guide_how_to_install_wireguard_on_a_raspberry_pi/), so don't think this is a dietpi issue necessarily, but also not finding a lot of help elsewhere yet.

MichaIng, 2019-10-11:
@theRAAPster 
Could you guys try to reboot first and and run `dpkg-reconfigure wireguard-dkms` afterwards?

WireGuard requires a kernel module to be built. Probably right after last kernel upgrade this does not work when the old kernel is still loaded.

Another try would be to use the new Raspbian Bullseye repo version of the packages, which are available now and used with v6.26. You might want to apply to the beta for this:
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=beta' /DietPi/dietpi.txt
dietpi-update
```

theRAAPster, 2019-10-11:
After rebooting:
```
$ sudo dpkg-reconfigure wireguard-dkms
/usr/sbin/dpkg-reconfigure: wireguard-dkms is broken or not fully installed
```

I'll give the bullseye repo a try later today.

MichaIng, 2019-10-11:
@theRAAPster 
Ahh `dpkg --configure wireguard-dkms` or `apt -f install` would have been it. `dpkg-reconfigure` only works if the install has fully finished, sorry for pasting wrong command 😉.

theRAAPster, 2019-10-12:
@MichaIng thanks for the suggestion, wireguard installed successfully using the Bullseye repo on v6.26.

I was already down the path of trying 6.26 before I saw your reply about fixing the install. I'm going to tinker with wireguard for a few days and will probably do a reinstall after that to make sure I have everything documented, will give fixing a try then.

closed 2019-12-11 by MichaIng