MichaIng/DietPi #3133: Radarr logs are not moved to ram on fresh dietpi install.

Dr0bac, 2019-10-02:
On Dietpi v6.18 Fourdee moved log files to ram, and i saw that on my system, now when i reinstalled Dietpi (clean install) logs are not linked anymore.

```- DietPi-Software | Radarr/Sonarr/Lidarr: logs (both .txt and .db*) have been moved to DietPi-RAMlog: https://github.com/MichaIng/DietPi/issues/2223```

MichaIng, 2019-10-02:
@Dr0bac 
Many thanks for your report.

Linking logs to RAM is still done: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L11836-L11842
However now stat you are reporting this, I remember that I found on Jessie systems that shm and wal files are removed and recreated due to a cron job, removing the symlinks as well: https://github.com/MichaIng/DietPi/issues/2689#issuecomment-487307295

However this did not affect the logs dir and db files and was only present on Jessie, not Stretch or Buster. But probably this changed a bid.

Please try to recreate the symlinks manually and check back once directly after Radarr start, once after some time (cron job) and once after reboot if those are still in place:
```
systemctl stop radarr
rm -Rf $G_FP_DIETPI_USERDATA/radarr/logs*
mkdir -p /var/log/radarr
ln -s /var/log/radarr $G_FP_DIETPI_USERDATA/radarr/logs
ln -s /var/log/radarr/logs.db $G_FP_DIETPI_USERDATA/radarr/logs.db
ln -s /var/log/radarr/logs.db-shm $G_FP_DIETPI_USERDATA/radarr/logs.db-shm
ln -s /var/log/radarr/logs.db-wal $G_FP_DIETPI_USERDATA/radarr/logs.db-wal
systemctl start radarr
```
I hope I find some time later as well to test it.

MichaIng, 2020-01-19:
@Dr0bac 
Ah lol just saw the it's you as well like on the other issue. So I rechecked and retested up and down and symlink are always created here and stay. I mark this as closed, however feel free to reopen if required, or we discuss this in the new issue: #3347

closed 2020-01-19 by MichaIng