MichaIng/DietPi #3135: dietpi-ramdisk failing to initialize

nnovaes, 2019-10-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | 10.1
- Kernel version | Linux DietPi 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5+deb10u2 (2019-08-08) x86_64 GNU/Linux
- SBC device | VM
- Power supply used | N/A
- SDcard used | N/A

#### Steps to reproduce
1. happens randomly after restartint the server

#### Expected behaviour
1. dietpi-ramdisk.service should be active 

#### Actual behaviour

Dietpi-ramdisk is failing to start

root@DietPi:~# systemctl status dietpi-ramdisk
● dietpi-ramdisk.service - DietPi-RAMdisk
   Loaded: loaded (/etc/systemd/system/dietpi-ramdisk.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-10-02 14:17:13 BST; 9min ago
  Process: 324 ExecStartPre=/bin/mkdir -p /var/tmp/dietpi/logs (code=exited, status=0/SUCCESS)
  Process: 410 ExecStart=/bin/dash -c /boot/dietpi/func/dietpi-ramdisk 0 2>&1 >> /var/tmp/dietpi/logs/dietpi-ramdisk.log (code=exited, status=2)
 Main PID: 410 (code=exited, status=2)

Oct 02 14:17:09 DietPi systemd[1]: Starting DietPi-RAMdisk...
Oct 02 14:17:13 DietPi dash[410]: /bin/dash: 1: cannot create /var/tmp/dietpi/logs/dietpi-ramdisk.log: Read-only file system
Oct 02 14:17:13 DietPi systemd[1]: dietpi-ramdisk.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Oct 02 14:17:13 DietPi systemd[1]: dietpi-ramdisk.service: Failed with result 'exit-code'.
Oct 02 14:17:13 DietPi systemd[1]: Failed to start DietPi-RAMdisk.
root@DietPi:~# cat  /var/tmp/dietpi/logs/dietpi-ramdisk.log
Wed Jul 10 22:09:08 UTC 2019 | DietPi-RAMdisk: Copying DietPi to RAM...
Wed Jul 10 22:09:08 UTC 2019 | DietPi-RAMdisk: Copied DietPi to RAM.
Wed Aug 21 03:48:41 BST 2019 | DietPi-RAMdisk: Storing DietPi to disk...
Wed Aug 21 03:48:41 BST 2019 | DietPi-RAMdisk: Stored DietPi to disk.
Wed Aug 21 03:48:41 BST 2019 | DietPi-RAMdisk: Copying DietPi to RAM...
Wed Aug 21 03:48:43 BST 2019 | DietPi-RAMdisk: Copied DietPi to RAM.
Mon Sep 30 17:56:52 BST 2019 | DietPi-RAMdisk: Storing DietPi to disk...
Mon Sep 30 17:56:52 BST 2019 | DietPi-RAMdisk: Stored DietPi to disk.
Mon Sep 30 17:56:59 BST 2019 | DietPi-RAMdisk: Copying DietPi to RAM...
Mon Sep 30 17:56:59 BST 2019 | DietPi-RAMdisk: Copied DietPi to RAM.
Mon Sep 30 20:16:22 BST 2019 | DietPi-RAMdisk: Storing DietPi to disk...
Mon Sep 30 20:16:22 BST 2019 | DietPi-RAMdisk: Stored DietPi to disk.
Mon Sep 30 20:16:23 BST 2019 | DietPi-RAMdisk: Copying DietPi to RAM...
Mon Sep 30 20:16:26 BST 2019 | DietPi-RAMdisk: Copied DietPi to RAM.
Mon Sep 30 21:12:21 BST 2019 | DietPi-RAMdisk: Storing DietPi to disk...
Mon Sep 30 21:12:22 BST 2019 | DietPi-RAMdisk: Stored DietPi to disk.
Mon Sep 30 21:12:23 BST 2019 | DietPi-RAMdisk: Copying DietPi to RAM...
Mon Sep 30 21:12:27 BST 2019 | DietPi-RAMdisk: Copied DietPi to RAM.
Mon Sep 30 22:44:53 BST 2019 | DietPi-RAMdisk: Storing DietPi to disk...
Mon Sep 30 22:44:54 BST 2019 | DietPi-RAMdisk: Stored DietPi to disk.
Mon Sep 30 22:45:01 BST 2019 | DietPi-RAMdisk: Copying DietPi to RAM...
Mon Sep 30 22:45:02 BST 2019 | DietPi-RAMdisk: Copied DietPi to RAM.
Mon Sep 30 22:54:00 BST 2019 | DietPi-RAMdisk: Storing DietPi to disk...
Mon Sep 30 22:54:01 BST 2019 | DietPi-RAMdisk: Stored DietPi to disk.
Mon Sep 30 22:54:05 BST 2019 | DietPi-RAMdisk: Copying DietPi to RAM...
Mon Sep 30 22:54:06 BST 2019 | DietPi-RAMdisk: Copied DietPi to RAM.
Tue Oct  1 16:36:11 BST 2019 | DietPi-RAMdisk: Storing DietPi to disk...
Tue Oct  1 16:36:12 BST 2019 | DietPi-RAMdisk: Stored DietPi to disk.
Tue Oct  1 16:36:27 BST 2019 | DietPi-RAMdisk: Copying DietPi to RAM...
Tue Oct  1 16:36:32 BST 2019 | DietPi-RAMdisk: Copied DietPi to RAM.
Wed Oct  2 14:17:06 BST 2019 | DietPi-RAMdisk: Storing DietPi to disk...
Wed Oct  2 14:17:07 BST 2019 | DietPi-RAMdisk: Stored DietPi to disk.

#### Extra details

i've rebuilt the server after the same error happened and now it happened again

nnovaes, 2019-10-02:
Also, not sure if helpful but...

root@DietPi:~# cat /proc/mounts
sysfs /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0
proc /proc proc rw,nosuid,nodev,noexec,relatime 0 0
udev /dev devtmpfs rw,nosuid,relatime,size=2003592k,nr_inodes=500898,mode=755 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0
tmpfs /run tmpfs rw,nosuid,noexec,relatime,size=401836k,mode=755 0 0
/dev/sda1 / ext4 ro,relatime 0 0
securityfs /sys/kernel/security securityfs rw,nosuid,nodev,noexec,relatime 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0
tmpfs /run/lock tmpfs rw,nosuid,nodev,noexec,relatime,size=5120k 0 0
tmpfs /sys/fs/cgroup tmpfs ro,nosuid,nodev,noexec,mode=755 0 0
cgroup2 /sys/fs/cgroup/unified cgroup2 rw,nosuid,nodev,noexec,relatime,nsdelegate 0 0
cgroup /sys/fs/cgroup/systemd cgroup rw,nosuid,nodev,noexec,relatime,xattr,name=systemd 0 0
pstore /sys/fs/pstore pstore rw,nosuid,nodev,noexec,relatime 0 0
bpf /sys/fs/bpf bpf rw,nosuid,nodev,noexec,relatime,mode=700 0 0
cgroup /sys/fs/cgroup/freezer cgroup rw,nosuid,nodev,noexec,relatime,freezer 0 0
cgroup /sys/fs/cgroup/rdma cgroup rw,nosuid,nodev,noexec,relatime,rdma 0 0
cgroup /sys/fs/cgroup/perf_event cgroup rw,nosuid,nodev,noexec,relatime,perf_event 0 0
cgroup /sys/fs/cgroup/pids cgroup rw,nosuid,nodev,noexec,relatime,pids 0 0
cgroup /sys/fs/cgroup/memory cgroup rw,nosuid,nodev,noexec,relatime,memory 0 0
cgroup /sys/fs/cgroup/blkio cgroup rw,nosuid,nodev,noexec,relatime,blkio 0 0
cgroup /sys/fs/cgroup/devices cgroup rw,nosuid,nodev,noexec,relatime,devices 0 0
cgroup /sys/fs/cgroup/cpu,cpuacct cgroup rw,nosuid,nodev,noexec,relatime,cpu,cpuacct 0 0
cgroup /sys/fs/cgroup/net_cls,net_prio cgroup rw,nosuid,nodev,noexec,relatime,net_cls,net_prio 0 0
cgroup /sys/fs/cgroup/cpuset cgroup rw,nosuid,nodev,noexec,relatime,cpuset 0 0
systemd-1 /proc/sys/fs/binfmt_misc autofs rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=13353 0 0
debugfs /sys/kernel/debug debugfs rw,relatime 0 0
mqueue /dev/mqueue mqueue rw,relatime 0 0
hugetlbfs /dev/hugepages hugetlbfs rw,relatime,pagesize=2M 0 0
sunrpc /run/rpc_pipefs rpc_pipefs rw,relatime 0 0
tmpfs /DietPi tmpfs rw,nosuid,nodev,noatime,size=10240k 0 0
tmpfs /tmp tmpfs rw,nosuid,nodev,noatime,size=1047552k 0 0
tmpfs /var/log tmpfs rw,nosuid,nodev,noatime,size=51200k 0 0

root@DietPi:~# cat /etc/fstab
# Please use "dietpi-drive_manager" to setup mounts
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------
//10.1.2.17/dietpi /mnt/samba cifs username=dietpi,password=<edited>,iocharset=utf8,uid=dietpi,gid=dietpi,file_mode=0770,dir_mode=0770,vers=3.1.1,_netdev,nofail 0 0

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs /DietPi tmpfs defaults,size=10m,noatime,nodev,nosuid,mode=1777 0 0
tmpfs /var/log tmpfs defaults,size=50m,noatime,nodev,nosuid,mode=1777 0 0

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAPFILE
#----------------------------------------------------------------
/var/swap none swap sw 0 0

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
root@DietPi:~#

MichaIng, 2019-10-02:
@nnovaes 
Many thanks for your report.

Indeed the root partition is mounted read-only:
> /dev/sda1 / ext4 ro,relatime 0 0

Also it is missing in your fstab, which is probably the reason for that.

Does the following work?
`mount -o remount,rw /dev/sda1 /` or
`mount -o rw /dev/sda1 /`

When you manage to mount or boot it correctly with R/W access, run `dietpi-drive_manager` once to have it added to your fstab, or add it manually to fstab. Another possibility is to mount the drive to another system and edit the `/etc/fstab` there, but actually it should be possible to mount to R/W on running system.

If it still fails to mount R/W on reboot, check `journalctl` and/or `dmesg` for related system logs during early boot.

nnovaes, 2019-10-02:
i was able to issue 

> mount -o remount,rw /dev/sda1 /

and then edit the fstab to uncomment the line of my second drive. i then was able to successfully boot and ran dietpi-drive_manager to update fstab. issue resolved. thank you very much!

closed 2019-10-02 by nnovaes