MichaIng/DietPi #3137: running squeezelite result in crashing sparky SBC

seniorgod, 2019-10-03:
I trying to run squeezelite as music player on my sparky.
Currently i use: mpd, upmpcli, shairport-sync and spotifyd without problems. ....ok, ok i still experience kernel traces from usb sound stack while hearing, but the music doesn't stop....

I loaded squeezelite with dietpi-software and configure it for my usb dac. Sometimes it runs for some songs. But sometimes sparky crashes and must be rebooted by detaching/attaching power supply. 

Does anybody experience the same problem?
Is it worth trying to load the "standard" squeezelite from debian?

#### Required Information
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
debian is 9.11
Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
Sparky SBC (armv7l)

seniorgod, 2019-10-04:
i was able to get some log trace while SBC is crashing.... 
I compiled squeezelite by myself to ensure that everything fits to my sparky/usbridge. 
Because of the RAMLog log system which is installed, that "forget" any log after restart, i started squeezelite via ssh and comandline set log level to all=debug and wait what happens. 
After some songs SBC crashes. What i can understand is, that we have a buffer underun in ALSA which "kills" sparky. 

`
[19:34:37.982798] process_strm:274 strm command q
[19:34:37.982939] decode_flush:237 decode flush
[19:34:37.983063] output_flush:435 flush output buffer
[19:34:37.984087] sendSTAT:189 STAT: STMf
[19:34:38.020268] process:521 audg
[19:34:38.020512] process_audg:433 audg gainL: 65536 gainR: 65536 adjust: 0
[19:34:38.020661] set_volume:233 setting internal gain left: 65536 right: 65536
[19:34:38.021725] process:521 strm
[19:34:38.022060] process_strm:274 strm command s
[19:34:38.022201] process_strm:344 strm s autostart: 1 transition period: 10 transition type: 0 codec: f
[19:34:38.022341] sendSTAT:189 STAT: STMf
[19:34:38.022808] codec_open:265 codec open: 'f'
[19:34:38.023054] _tcp_connect:486 connecting to 192.168.178.2:9000
[19:34:38.024434] stream_sock:577 header: GET /stream.mp3?player=00:17:f7:01:06:5b HTTP/1.0
[19:34:38.024689] sendSTAT:189 STAT: STMc
[19:34:38.024864] process_strm:377 set fade mode: 0
[19:34:38.025070] process:521 audg
[19:34:38.025202] process_audg:433 audg gainL: 65536 gainR: 65536 adjust: 0
[19:34:38.025709] set_volume:233 setting internal gain left: 65536 right: 65536
[19:34:38.044071] stream_thread:247 headers: len: 118
HTTP/1.1 200 OK
Server: Logitech Media Server (7.9.2 - 1568726655)
Connection: close
Content-Type: audio/x-flac
[19:34:38.045123] sendRESP:220 RESP
[19:34:38.159810] write_cb:128 setting track_start
[19:34:38.165158] output_thread:687 open output device: plughw:CARD=P20,DEV=0
[19:34:38.165528] alsa_open:351 opening device at: 44100
[19:34:38.265902] alsa_open:422 opened device plughw:CARD=P20,DEV=0 using format: S32_LE sample rate: 44100 mmap: 1
[19:34:38.272546] alsa_open:513 buffer: 40 period: 4 -> buffer size: 1764 period size: 441
[19:34:40.043395] process:521 strm
[19:34:40.043669] process_strm:274 strm command t
[19:34:40.043826] sendSTAT:189 STAT: STMt
[19:34:40.559306] ALSA snd_pcm_hw_prepare:603 SNDRV_PCM_IOCTL_PREPARE failed (-32)
[19:34:40.559822] ALSA snd_pcm_recover:8309 cannot recovery from underrun, prepare failed: Broken pipe
[19:34:40.560082] output_thread:776 pcm wait error: Broken pipe
[19:34:40.560355] output_thread:717 XRUN
[19:34:40.561782] ALSA snd_pcm_hw_prepare:603 SNDRV_PCM_IOCTL_PREPARE failed (-32)
[19:34:40.562089] ALSA snd_pcm_recover:8309 cannot recovery from underrun, prepare failed: Broken pipe
[19:34:40.562285] output_thread:719 XRUN recover failed: Broken pipe
[19:34:40.562440] output_thread:717 XRUN
[19:34:40.563568] ALSA snd_pcm_hw_prepare:603 SNDRV_PCM_IOCTL_PREPARE failed (-32)
[19:34:40.563848] ALSA snd_pcm_recover:8309 cannot recovery from underrun, prepare failed: Broken pipe
[19:34:40.564038] output_thread:719 XRUN recover failed: Broken pipe
[19:34:40.564189] output_thread:717 XRUN
[19:34:40.565407] ALSA snd_pcm_hw_prepare:603 SNDRV_PCM_IOCTL_PREPARE failed (-32)
[19:34:40.565708] ALSA snd_pcm_recover:8309 cannot recovery from underrun, prepare failed: Broken pipe
[19:34:40.565914] output_thread:719 XRUN recover failed: Broken pipe
[19:34:40.566080] output_thread:717 XRUN
[19:34:40.567322] ALSA snd_pcm_hw_prepare:603 SNDRV_PCM_IOCTL_PREPARE failed (-32)
[19:34:40.567627] ALSA snd_pcm_recover:8309 cannot recovery from underrun, prepare failed: Broken pipe
[19:34:40.567830] output_thread:719 XRUN recover failed: Broken pipe
[19:34:40.567993] output_thread:717 XRUN
`

MichaIng, 2019-10-05:
@seniorgod 
Many thanks for your report.

Damn where has my comment gone... Probably I missed to push the "Comment" button...

For reference, the kernel trace error issue: #2610

Jep I what I suggested yesterday (but failed to post) was the idea to disable/uninstall DietPi-RAMlog and make the syslog boot persistent:
```
mkdir -p /var/log/journal && systemctl restart systemd-journald
```
After a crash you could run `journalctl` then to check all system logs from before the crash. However your SSH solution serves as well 🙂.

I was checking the Sparky kernel/firmware repo for any relevant changes, but the last USB driver update has already been applied some DietPi versions ago:
- https://github.com/sparky-sbc/sparky-test
- https://github.com/sparkysbc/Linux
____________
Actually from the logs it is possibly related to your kernel trace error. The affected kernel modules are the same.

> I compiled squeezelite by myself to ensure that everything fits to my sparky/usbridge.

👍 Good idea to rule any binary/build issues out.

Did you as well try to adjust the buffer-related values of the systemd unit we ship: `/etc/systemd/system/squeezelite.service`
> `-a 4096:1024:16:0`

Meaning: **4096** byte buffer size, **1024** byte period size, **16** bit sample rate, **no** mmap (https://manpages.debian.org/buster/squeezelite-pa/squeezelite.1.en.html#OPTIONS)

seniorgod, 2019-10-08:
Today i tried to use the -a option you mentioned.

First i use -a 4096:1024:32:0 because my USB DAC wants to receive 32 bit.
It runs several minutes without problem. 
Because i will not define the bitrate i try -a 4096:1024::0 . After starting that my sparky get to be unusable after some seconds. restart. 

Than i try: 
squeezelite -n "dietpi squeeze"  -a 4096:1024:32:0 -C 1 -o "plughw:1,0" -d "all=debug" -f /var/log/squeezelite.log 

squeeze get defunct: 

root@DietPi:~/squeezelite# ps ax | grep squeeze
 1777 pts/0    Zl+    3:37 [squeezelite] <defunct>
 1901 pts/2    S+     0:00 grep --color=auto squeeze
root@DietPi:~/squeezelite# 

this is the output of journalctl: 
`
Okt 08 23:15:02 DietPi kernel: iso_packets:0, bInterval:1, urb_interval:1, reg_con:0x4
Okt 08 23:15:02 DietPi kernel: iso_packets:0, bInterval:4, urb_interval:8, reg_con:0x4
Okt 08 23:15:02 DietPi kernel: aotg_hub_urb_dequeue, retval:-43, urb not submitted or unlinked , type = 0 
Okt 08 23:15:03 DietPi kernel: timeout: still 1 active urbs on EP #1
Okt 08 23:15:04 DietPi kernel: timeout: still 1 active urbs on EP #1
Okt 08 23:15:04 DietPi kernel: ------------[ cut here ]------------
Okt 08 23:15:04 DietPi kernel: WARNING: at /imxhdd/opt/1301TAG/sparky_volumio/kernel/drivers/usb/core/urb.c:328 usb_submit_urb+0x3d4/0x404()
Okt 08 23:15:04 DietPi kernel: URB de867a00 submitted while active
Okt 08 23:15:04 DietPi kernel: Modules linked in: nls_cp437 snd_usb_audio snd_hwdep snd_usbmidi_lib spidev ethernet atc260x_cap_gauge atc260x_irkeypad autofs4
Okt 08 23:15:04 DietPi kernel: CPU: 1 PID: 1779 Comm: squeezelite Tainted: G        W    3.10.38 #22
Okt 08 23:15:04 DietPi kernel: [<c0015b34>] (unwind_backtrace+0x0/0x138) from [<c001304c>] (show_stack+0x24/0x2c)
Okt 08 23:15:04 DietPi kernel: [<c001304c>] (show_stack+0x24/0x2c) from [<c002beb8>] (warn_slowpath_common+0x4c/0x6c)
Okt 08 23:15:04 DietPi kernel: [<c002beb8>] (warn_slowpath_common+0x4c/0x6c) from [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40)
Okt 08 23:15:04 DietPi kernel: [<c002bf6c>] (warn_slowpath_fmt+0x30/0x40) from [<c03fe604>] (usb_submit_urb+0x3d4/0x404)
Okt 08 23:15:04 DietPi kernel: [<c03fe604>] (usb_submit_urb+0x3d4/0x404) from [<bf04c518>] (snd_usb_endpoint_start+0x15c/0x348 [snd_usb_audio])
Okt 08 23:15:04 DietPi kernel: [<bf04c518>] (snd_usb_endpoint_start+0x15c/0x348 [snd_usb_audio]) from [<bf052950>] (start_endpoints+0xc4/0x140 [snd_usb_audio])
Okt 08 23:15:04 DietPi kernel: [<bf052950>] (start_endpoints+0xc4/0x140 [snd_usb_audio]) from [<bf054428>] (snd_usb_pcm_prepare+0x138/0x3a8 [snd_usb_audio])
Okt 08 23:15:04 DietPi kernel: [<bf054428>] (snd_usb_pcm_prepare+0x138/0x3a8 [snd_usb_audio]) from [<c055384c>] (snd_pcm_do_prepare+0x14/0x2c)
Okt 08 23:15:04 DietPi kernel: [<c055384c>] (snd_pcm_do_prepare+0x14/0x2c) from [<c055322c>] (snd_pcm_action_single+0x38/0x78)
Okt 08 23:15:04 DietPi kernel: [<c055322c>] (snd_pcm_action_single+0x38/0x78) from [<c055354c>] (snd_pcm_action_nonatomic+0x54/0x60)
Okt 08 23:15:04 DietPi kernel: [<c055354c>] (snd_pcm_action_nonatomic+0x54/0x60) from [<c055482c>] (snd_pcm_common_ioctl1+0x828/0xd4c)
Okt 08 23:15:04 DietPi kernel: [<c055482c>] (snd_pcm_common_ioctl1+0x828/0xd4c) from [<c0555420>] (snd_pcm_playback_ioctl1+0x138/0x440)
Okt 08 23:15:04 DietPi kernel: [<c0555420>] (snd_pcm_playback_ioctl1+0x138/0x440) from [<c0121e38>] (do_vfs_ioctl+0x80/0x598)
Okt 08 23:15:04 DietPi kernel: [<c0121e38>] (do_vfs_ioctl+0x80/0x598) from [<c01223bc>] (SyS_ioctl+0x6c/0x7c)
Okt 08 23:15:04 DietPi kernel: [<c01223bc>] (SyS_ioctl+0x6c/0x7c) from [<c000f340>] (ret_fast_syscall+0x0/0x30)
Okt 08 23:15:04 DietPi kernel: ---[ end trace 928922720e8eef52 ]---
Okt 08 23:15:04 DietPi kernel: cannot submit urb 7, error -16: unknown error
Okt 08 23:15:04 DietPi kernel: cannot submit urb 7, error -16: unknown error
Okt 08 23:15:04 DietPi kernel: cannot submit urb 7, error -16: unknown error
Okt 08 23:15:04 DietPi kernel: cannot submit urb 7, error -16: unknown error
Okt 08 23:15:04 DietPi kernel: aotg_ring_enqueue_isoc_td:927, dbg!
Okt 08 23:15:04 DietPi kernel: cannot submit urb 3, error -1: unknown error
Okt 08 23:15:04 DietPi kernel: cannot submit urb 7, error -16: unknown error
Okt 08 23:15:04 DietPi kernel: aotg_hub_urb_dequeue, retval:-43, urb not submitted or unlinked , type = 0 
Okt 08 23:15:04 DietPi kernel: aotg_hub_urb_dequeue, retval:-43, urb not submitted or unlinked , type = 0 
Okt 08 23:15:04 DietPi kernel: aotg_hub_urb_dequeue, retval:-43, urb not submitted or unlinked , type = 0 
Okt 08 23:16:05 DietPi kernel: INFO: rcu_preempt detected stalls on CPUs/tasks: { 1} (detected by 2, t=6002 jiffies, g=1098, c=1097, q=326)
Okt 08 23:16:05 DietPi kernel: Backtrace for cpu 2 (current):
Okt 08 23:16:05 DietPi kernel: CPU: 2 PID: 28 Comm: kworker/2:1 Tainted: G        W    3.10.38 #22
Okt 08 23:16:05 DietPi kernel: Workqueue: events cs_dbs_timer
Okt 08 23:16:05 DietPi kernel: [<c0015b34>] (unwind_backtrace+0x0/0x138) from [<c001304c>] (show_stack+0x24/0x2c)
Okt 08 23:16:05 DietPi kernel: [<c001304c>] (show_stack+0x24/0x2c) from [<c00148a4>] (smp_send_all_cpu_backtrace+0x58/0xc4)
Okt 08 23:16:05 DietPi kernel: [<c00148a4>] (smp_send_all_cpu_backtrace+0x58/0xc4) from [<c00a49d8>] (rcu_check_callbacks+0x664/0x7e4)
Okt 08 23:16:05 DietPi kernel: [<c00a49d8>] (rcu_check_callbacks+0x664/0x7e4) from [<c003b7a8>] (update_process_times+0x3c/0x68)
[... cut ...]
Okt 08 23:16:05 DietPi kernel: 7f80: c0c76000 c0d29dec c0c89ed4 414fc091 c07c6920 c0d295bd 00000000 c0c77fb0
Okt 08 23:16:05 DietPi kernel: 7fa0: c0010050 c0010048 600e0013 ffffffff
Okt 08 23:16:05 DietPi kernel: [<c000ef40>] (__irq_svc+0x40/0x70) from [<c0010048>] (arch_cpu_idle+0x28/0x38)
Okt 08 23:16:05 DietPi kernel: [<c0010048>] (arch_cpu_idle+0x28/0x38) from [<c00736b8>] (cpu_startup_entry+0x68/0x24c)
Okt 08 23:16:05 DietPi kernel: [<c00736b8>] (cpu_startup_entry+0x68/0x24c) from [<c0c00a34>] (start_kernel+0x2c4/0x320)

`

Try more tomorrow......

MichaIng, 2019-10-09:
@seniorgod 
Okay from the error trace nothing that I would understand. Not sure if there was some special reason for us to force 16bit (on all installs/devices), it is just the only thing that might play a role and which can be easily adjusted/tested.

MichaIng, 2020-01-01:
I mark this issue as closed since it has been a while. Feel free to reopen if required.

closed 2020-01-01 by MichaIng