MichaIng/DietPi #3149: esp01215e LCD Driver support for N2

moatazsalemVF, 2019-10-09:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...
i would be happy to get my esp01215e elecrow 7 inch touch display working with Odroid n2

#### Describe the solution you'd like:

- ...
add the option is display settings
#### Describe alternatives you've considered:

- ...
normal PC screen
#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2019-10-11:
@moatazsalemVF 
Many thanks for your request. Gathering any information/guides about how to do that would be helpful as I also need to start from scratch by doing some web searches 😉.

MichaIng, 2022-01-29:
@moatazsalemVF 
Is it [this](https://www.elecrow.com/7-inch-1024-600-hdmi-lcd-display-with-touch-screen.html) LCD?
It says all drivers are included with the DVD, but it also lists a small number of supported SBCs, not including Odroid N2 🤔.

Since it is a HDMI LCD, EDID auto-detection may apply correct resolution automatically, at last on RPi only the resolution needs to be set: https://www.elecrow.com/wiki/index.php?title=7_Inch_1024*600_HDMI_LCD_Display_with_Touch_Screen

But instructions about how to get touch input running is entirely missing. Probably the DVD shipped with the display provides info about that?

MichaIng, 2022-02-13:
Marking it as closed until another ping or request states that it doesn't work it your new mainline kernel image either.

closed 2022-02-13 by MichaIng