MichaIng/DietPi #3150: SSH fail on the first and second time, success on 3rd try

tindecken, 2019-10-10:
**Steps:**
SSH to the Pi
Login with correct username/password

**Expected:** After first time, it login success
**Actual result**: It return access denied on the fist and second time.

[image]

MichaIng, 2019-10-10:
@tindecken 
Many thanks for your report.

If it worked well the 3rd time, you must have entered the password wrong on first attempts. There is really no chance that within the Linux core authentication mechanism, the same password is denied first and accepted afterwards, as long as one does not change it on another session meanwhile 😉.

tindecken, 2019-10-11:
Thank you, but i'm sure i'm typing correct password for all times.
You can try ssh to my pi for the check.
tindecken.ddns.net
root/rivaldo

MichaIng, 2019-10-11:
@tindecken 
I just recognised the "`Send automatic password`" in your KiTTY output when access is denied. On successful login this does not appear.
I tested on my (PuTTY) and when typing password wrong or right, this does not show up:
```
login as: root
root@192.168.178.30's password:
Access denied
root@192.168.178.30's password:
Access denied
root@192.168.178.30's password:
 ─────────────────────────────────────────────────────
 DietPi v6.26.0 (dev) : Update available
 ─────────────────────────────────────────────────────
 - FQDN/hostname : VM-Buster
 - LAN IP : 192.168.178.30 (eth0)
 - Freespace (RootFS) : 6.9G
 - MOTD : Open Beta v6.26, please help testing: https://github.com/MichaIng/DietPi/issues/3131
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder)
 Image           : DietPi Core Team (pre-image: Debian mini.iso)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : PINE64 community | oct8l
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com

 dietpi-update   : Run now to update DietPi from v6.26.0 (dev) to v6.26.1.

root@VM-Buster:~#
```
I checked PuTTY authentication options and can't find some auto-password feature, just the Pagent method when having PuTTY agent running on client system. Not sure whether KiTTY has some additional option to send some password automatically? Although strange then that you still need to enter some first 🤔.

tindecken, 2019-10-11:
yes, i'm using `Send automatic password` in Kitty SSH. When I turn it off, and manually type user/pass, it doesn't happen.
Thank you for the support. :)

MichaIng, 2019-10-13:
@tindecken 
Okay great we identified it. What I also recognised that in your image after first login attempt there is an empty line before `Send automatic password`, on second attempt not. Not sure if this means something but it is strange to me that outputs are not exactly the same, like if there was a new line character in the first sent password or such, which would then greatly explain the error. Not sure how this feature works but probably something to have the KiTTY dev(s) have a look at: https://github.com/cyd01/KiTTY/issues

Just found an issue that the automatic password sent happens when the server expects something else (e.g. 2FA): https://github.com/cyd01/KiTTY/issues/24
One idea is that e.g. you actually need to accept the servers host key first when either entering username the first time, but KiTTY overrides this request by sending the password, then when the actual password request comes, it sends it a second time, leading to a two lined entry with doubled password. However just a guess that came to my mind 😉.

MichaIng, 2019-10-13:
Just ran some tests. And indeed I can verify:
- It has nothing to do with host key accepting. This is asked fore before even username is asked for.
- Also it has nothing to do with auto username, the below happens when setting it and when entering it manually as well.
- When defining auto password, sometimes it works, but mostly not (`Access denied`) without changing anything. Sometimes even the session crashes.
- Not sure how this can be, probably some timing issue, but this is especially kind dangerous since fail2ban or similar might lock you out due to this... Definitely should be reported to KiTTY dev(s).

MichaIng, 2019-10-13:
```
Incoming packet #0x4, type 51 / 0x33 (SSH2_MSG_USERAUTH_FAILURE)
  00000000  00 00 00 12 70 75 62 6c 69 63 6b 65 79 2c 70 61  ....publickey,pa
  00000010  73 73 77 6f 72 64 00                             ssword.
Event Log: Send automatic password
Event Log: Sent password
Outgoing packet #0x5, type 2 / 0x02 (SSH2_MSG_IGNORE)
  00000000  00 00 00 73 be 9c bc b2 37 26 78 97 06 0d ed e0  ...s....7&x.....
  00000010  81 9a e3 0c 04 3e 67 62 b4 01 d3 d4 cd 3d 3a 1a  .....>gb.....=:.
  00000020  c0 b8 3c 1f cb e8 f4 f3 33 ba e1 56 bf 82 57 2b  ..<.....3..V..W+
  00000030  ac 1e 78 2e 75 fd 78 6a 9a 7e d0 05 5b 4a 6d 73  ..x.u.xj.~..[Jms
  00000040  3b 49 e3 ba cf 34 dc 5f 3a bb c5 cc 4e 18 0f b0  ;I...4._:...N...
  00000050  36 10 36 f8 05 f5 52 be 29 d8 46 a4 c1 70 22 8b  6.6...R.).F..p".
  00000060  2e 5b 7a 81 13 12 51 6f 3f 03 94 30 5a b4 45 70  .[z...Qo?..0Z.Ep
  00000070  bb 79 cc e9 f2 63 c9                             .y...c.
Outgoing packet #0x6, type 50 / 0x32 (SSH2_MSG_USERAUTH_REQUEST)
  00000000  00 00 00 04 72 6f 6f 74 00 00 00 0e 73 73 68 2d  ....root....ssh-
  00000010  63 6f 6e 6e 65 63 74 69 6f 6e 00 00 00 08 70 61  connection....pa
  00000020  73 73 77 6f 72 64 00 00 00 00 06 XX XX XX XX XX  ssword.....XXXXX
  00000030  XX                                               X
Outgoing raw data at 2019-10-13 12:45:27
  00000000  44 bb 18 71 19 a6 34 8e b9 5a 37 54 ad f3 d8 8a  D..q..4..Z7T....
  00000010  db a9 09 a7 d4 40 d5 d3 bd db 6f df f2 23 fb 79  .....@....o..#.y
  00000020  4b a8 be 6b e7 47 30 27 18 f1 12 e0 e1 28 d0 10  K..k.G0'.....(..
  00000030  e2 40 ed 41 e6 19 95 39 51 5c 75 65 c8 d9 f5 ac  .@.A...9Q\ue....
  00000040  c0 a8 88 52 2f b6 6b 28 b1 e4 a0 ee 29 7b 87 8d  ...R/.k(....){..
  00000050  f7 52 11 53 80 b3 f2 dc 54 54 99 be 9e 03 20 51  .R.S....TT.... Q
  00000060  b2 80 d3 8a 0c 2c a3 84 9f bc 99 08 5f 35 9c 48  .....,......_5.H
  00000070  e1 37 5a ea ba 16 46 86 1e b8 14 18 75 6b 8d 01  .7Z...F.....uk..
  00000080  f3 8e b3 d7 f6 d9 37 a2 44 02 4e b0 89 89 f9 31  ......7.D.N....1
  00000090  47 08 53 cf 8e 69 0f 95 bf 51 6a 66 4c 6d 11 da  G.S..i...QjfLm..
  000000a0  91 43 c9 99 1a a4 10 02 17 cf 62 98 b4 c6 12 73  .C........b....s
  000000b0  93 3c 89 53 86 a2 15 ac 57 ae 2d 1c 73 82 4b 7b  .<.S....W.-.s.K{
  000000c0  52 cf 72 f9 be 86 11 d2 a2 ff 41 c5 fc df e0 96  R.r.......A.....
  000000d0  30 b6 8b da 6c 4c 2f d2 ef 92 6f 3d f4 93 b0 9a  0...lL/...o=....
  000000e0  b8 f0 cb 74 69 06 d8 77 95 dc b0 d4 1f 6f dc bf  ...ti..w.....o..
  000000f0  5d db f6 5f 68 af 6e 24 f5 d8 cf 5d e3 31 14 c2  ].._h.n$...].1..
  00000100  04 ff a8 7d 9c 68 3f bb de 3b 4e 8c 52 ff 2a 11  ...}.h?..;N.R.*.
Incoming raw data at 2019-10-13 12:45:28
  00000000  a4 72 00 ff 0f 5e 73 ad d6 41 3f d7 33 36 f3 d4  .r...^s..A?.36..
  00000010  83 69 c2 37 7a 71 cf 59 bb 2d cb 1f f5 6b 34 6c  .i.7zq.Y.-...k4l
  00000020  56 9a 4d 3d 75 07 5a 31 ba 73 82 39 a5 a0 82 7e  V.M=u.Z1.s.9...~
  00000030  f5 e3 af 0c f2 cb f2 c9 2e 19 fd f5 43 3d b4 72  ............C=.r
  00000040  54 51 c5 94 86 80 ba 37 93 57 8d 4c 0b 08 f3 72  TQ.....7.W.L...r
Incoming packet #0x5, type 51 / 0x33 (SSH2_MSG_USERAUTH_FAILURE)
  00000000  00 00 00 12 70 75 62 6c 69 63 6b 65 79 2c 70 61  ....publickey,pa
  00000010  73 73 77 6f 72 64 00                             ssword.
Event Log: Password authentication failed
```
The SSH2_MSG_USERAUTH_FAILURE in first line is btw due to KiTTY (every SSH client) tries to authenticate without any password first, then with public key (if configured), last with password. But the SSH2_MSG_USERAUTH_FAILURE in last incoming packet is what should not happen. Something must go wrong sometimes, sometimes not, when sending the auto password.

MichaIng, 2019-10-13:
Just reported the issue: https://github.com/cyd01/KiTTY/issues/59

tindecken, 2019-10-14:
Thank you the awesome investigate, and support.

MichaIng, 2020-10-23:
The issue has been solved with the newest KiTTY version: https://github.com/cyd01/KiTTY/issues/59#issuecomment-703289454
Marking as closed.

closed 2020-10-23 by MichaIng