MichaIng/DietPi #3151: DietPi- Config | Set ARM Temp Limit for RPI3 and RPI4 to 75c. Currently set at 65c

Trunkzeh, 2019-10-10:
Hey @MichaIng ;

Following on from the previous conversation within the RPI4 temperature bug report; this could probably be done via board ID or HW ID of some description so that the ARM throttling limit is increased to 75c instead of the 65c thats currently being set.

Thoughts on the matter?

MichaIng, 2019-10-11:
@Trunkzeh 
Many thanks for your request. Indeed this has already been done for v6.26: https://github.com/MichaIng/DietPi/blob/dev/rootfs/var/lib/dietpi/services/dietpi-firstboot.bash#L79

Trunkzeh, 2019-10-11:
Hey MichaIng/DietPi ; this was pushed for 6.26.1 or later? It didn’t apply
on my RPI2 on 6.26.1; this was a fresh reinstall.

MichaIng, 2019-10-12:
@Trunkzeh 
This was with v6.26.0 already, but it does not apply with v6.25 images since this code runs early on very first boot before an update is applied.

Do you think we should apply the change with v6.26 for all RPi4 models (where currently 65°C is set)?

Trunkzeh, 2019-10-14:
@MichaIng Both would be favourable; but certainly RPI4 as these units run considerably hotter than previous generations of raspberrypis. I've kept mine on a fan shim and currently am running it as a testbed for dietpi so its not even running production environment software yet and it will still reach 65c regularly.

MichaIng, 2019-10-14:
Done: https://github.com/MichaIng/DietPi/commit/95fc1dbf7b30cb0802ee43348426f1b478952d23

closed 2019-10-15 by MichaIng