MichaIng/DietPi #3153: RP2 crashing at 55C on Dietpi, stable at 66C on Raspbian.

PokeyPoke, 2019-10-11:
### Creating a bug report/issue

#### Required Information
- DietPi version |
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | 10.1
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- SBC device | RPi 2 Model B (armv7l)
- Power supply used | 5V 3A NorthPada (Tried several power supplies)
- SDcard used | PNY Elite 64gb, and two generic 8gb sd cards

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? Yes, on three different SD cards

#### Steps to reproduce
1. Finish fresh install of dietpi
2. Install/Run Stress test
3. Run stress test with CPU/RAM for 1min or more
4. During the stress test, once the temp reaches 54 or 55C the system stops responding.

#### Expected behaviour
Should be able to push rpi up to 80C without crashing.

#### Actual behaviour
Pi stops responding once temp reaches 54-55C

#### Extra details
I have tested the same pi, with the same SD card, and same power supply with Raspbian and Stress test and was able to get the pi up to 66C without any issues. Once I tried Dietpi again it would consistently stop responding at 54-55C.

MichaIng, 2019-10-11:
@PokeyPoke 
Many thanks for your report.

Indeed strange, since I push my RPi2 regularly to higher temps temps and of course the RPi should survive 100°C without issues, besides you get your fingers burn when touching and might smell some melting plastic cover 😉.

Have you compared the `/boot/config.txt` for differences between Raspbian and your DietPi?
Also you could disable/uninstall DietPi-RAMlog and make syslog persistent: `mkdir /var/log/journal`
After crash you are then able to check `journalctl` for error messages.

PokeyPoke, 2019-10-12:
Hi Michalng!

Thank you so much for the reply.

I was taking at look at the two config files and they are a bit different. Nothing stood out to me in either one that would cause this issue. If it helps, here is a wall of text that contains both config files, dietpi first, raspbian second. In the meantime I will set syslog persistent. Thank you again for you attention to my problem.

Dietpi config.txt
# IMPORTANT:
# - Modifications to /boot/config.txt will not be preserved on reboot.
# - Please ensure you edit from the DietPi-RAMdisk location: /DietPi/config.txt

#-------Display---------
# If you get no picture, set the following to "1" to apply most compatible HDMI settings.
#hdmi_safe=1

# Uncomment to adjust the HDMI signal strength if you have interferences, blanking, or no display.
# - Values from "0" to "11" are allowed, use values above "7" only if required, e.g. with very long HDMI cable. 
# - Default on first RPi1 is "2" and on RPi2 and above: "5"
#config_hdmi_boost=5

# Uncomment if HDMI display is not detected and composite is being outputted.
#hdmi_force_hotplug=1

# Set the following two settings to "1" to disable video output and framebuffer completely.
# NB: This will lead to some error messages on boot, which can be safely ignored.
#hdmi_ignore_hotplug=1
#hdmi_ignore_composite=1

# Uncomment to force a console size. By default it will be display's size minus overscan.
#framebuffer_width=1280
#framebuffer_height=720

# SDTV_MODES
#sdtv_mode=0

# Uncomment to force a specific HDMI mode (this will force VGA).
#hdmi_group=1
#hdmi_mode=1

# Uncomment to force an HDMI mode rather than DVI. This can make audio work in DMT (computer monitor) modes.
#hdmi_drive=2

# Set "hdmi_blanking=1" to allow the display going into standby after 10 minutes without input.
# With default value "0", the display shows a blank screen instead, but will not go into standby.
# NB: With "1" some applications (e.g. Kodi, OMXPlayer) cannot prevent display standby due to missing DPMS signal.
#hdmi_blanking=1

# Set to "1" if your display has a black border of unused pixels visible.
disable_overscan=1

# Uncomment the following to adjust overscan. Use positive numbers if console goes off screen, and negative if there is too much border.
#overscan_left=16
#overscan_right=16
#overscan_top=16
# SDTV_MODES
#sdtv_mode=0

# Uncomment to force a specific HDMI mode (this will force VGA).
#hdmi_group=1
#hdmi_mode=1

# Uncomment to force an HDMI mode rather than DVI. This can make audio work in DMT (computer monitor) modes.
#hdmi_drive=2

# Set "hdmi_blanking=1" to allow the display going into standby after 10 minutes without input.
# With default value "0", the display shows a blank screen instead, but will not go into standby.
# NB: With "1" some applications (e.g. Kodi, OMXPlayer) cannot prevent display standby due to missing DPMS signal.
#hdmi_blanking=1

# Set to "1" if your display has a black border of unused pixels visible.
[... 122 lines cut ...]

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

PokeyPoke, 2019-10-12:
For some reason, when I went to install ramlog, there were some updates happening and now my system can get to 60C before crashing. I checked journalctl, but I'm a complete noobie. It gave me thousands of lines and I skimmed them. There didn't seem to be anything strange happening. Sorry if I'm not good at helping to troubleshoot. Please let me know if there is anything else I can do or share to help diagnose the problem.

PokeyPoke, 2019-10-12:
The last item I can find before the pi2 stops responding is:

> at NzbDrone.Core.Download.TrackedDownloads.DownloadMonitoringService.ProcessClientDownloads (NzbDrone.Core.Download.IDownloadClient downloadClient) [0x0000c]
 in <28071e62f2f449e682962e05af20a3d7>:0

I'm going to test again with a completely clean install in hopes of finding something else.

PokeyPoke, 2019-10-12:
The last item I can find before the pi2 stops responding is:

at NzbDrone.Core.Download.TrackedDownloads.DownloadMonitoringService.ProcessClientDownloads (NzbDrone.Core.Download.IDownloadClient downloadClient) [0x0000c]
in <LOT OF NUMBERS, not sure what they are>:0

I'm going to test again with a completely clean install in hopes of finding something else.

PokeyPoke, 2019-10-12:
Here is the last bit of the log before crashing on a clean install of dietpi:
`Oct 12 02:32:15 DietPi kernel: usb-storage 1-1.5.2:1.0: USB Mass Storage device detected
Oct 12 02:32:15 DietPi kernel: scsi host0: usb-storage 1-1.5.2:1.0
Oct 12 02:32:15 DietPi kernel: usbcore: registered new interface driver uas`

I had the same usb drive connected via powered hub with the raspbian test that went very well.

MichaIng, 2019-10-13:
@PokeyPoke 
You should btw stop all other services when running a stress test, as you don't want to corrupt e.g. downloads currently done by NzbDrone 😉.

About the second log, does the timestamp really fit to the time of the crash? Looks like a regular last boot log. The UAS driver has been enabled by default with one of the last firmware updates, but actually it is not fully supported by RPi. See two other logs about this (my RPi2):
```
[    4.354458] usb 1-1.2: The driver for the USB controller dwc_otg_hcd does not support scatter-gather which is
[    4.354470] usb 1-1.2: required by the UAS driver. Please try an other USB controller if you wish to use UAS.
```
If I remember the discussion correctly it is just about in some circumstances the presence of this driver is required, even that it is not actively in use for attached USB drives. I have it hence blacklisted on my system `blacklist uas`, however it should not cause any issues when loaded.

MichaIng, 2020-01-01:
I mark this issue as closed since it has been a while. Feel free to reopen if required.

closed 2020-01-01 by MichaIng