MichaIng/DietPi #3157: dietpi initial setup fails due to redis server

AaronDavidSchneider, 2019-10-14:
#### Details:
- Date           | Mon 14 Oct 08:16:15 BST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (index=4)
- Kernel version | #1244 SMP Thu Jul 4 18:48:07 BST 2019
- Distro         | buster (index=5)
- Command        | systemctl start redis-server
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

- fresh install of dietpi with Dietpi optimised software: nextcloud, pihole, pivpn, gitea
- I used static IP address settings with cloudflare dns
- I used a mounted drive for dietpi_userdata

#### Expected behaviour:

- dietpi should complete install

#### Actual behaviour:

- fails the installation 

#### Extra details:

#### Additional logs:
```
Log file contents:
Job for redis-server.service failed because the control process exited with error code.
See "systemctl status redis-server.service" and "journalctl -xe" for details.
```

MichaIng, 2019-10-14:
@AaronDavidSchneider 
Please paste:
```
journalctl -u redis-server
cat /var/log/redis/redis-server.log
```

AaronDavidSchneider, 2019-10-14:
I am sorry, but I moved on with my installation. It worked when I installed dietpi first and additional software afterwards.

MichaIng, 2019-10-14:
@AaronDavidSchneider 
Okay, I will keep an eye on it. Will mark this issue as closed then. Feel free to reopen if required.

closed 2019-10-14 by MichaIng