MichaIng/DietPi #3158: installation jackett

yopyop2019, 2019-10-14:
#### Details:
- Date           | Mon 14 Oct 14:43:04 BST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1270 SMP Tue Sep 24 18:45:11 BST 2019
- Distro         | buster (index=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

Hello, I have installed sonarr, transmission and jackett, everything worked "OK" until 3 days ago that the jackett stopped working for what I reinstalled, to solve the problem but it was not solved by what I uninstalled and I could never reinstall again, I get the following error:

[image]

yopyop2019, 2019-10-14:
i resolve this with:

cd /var/lib/dpkg/info

find -iname "*kodi*" -exec rm -r {} \;

closed 2019-10-14 by yopyop2019

MichaIng, 2019-10-14:
@yopyop2019 
The hammer method is not the best solution. Please see, I bet it is due to this issue: https://github.com/MichaIng/DietPi/issues/3031#issuecomment-540477241
Solution:
```
mkdir -p /etc/polkit-1/localauthority/50-local.d
apt install kodi
```