MichaIng/DietPi #3159: Issues on automatic install (i2c, start up, desktop, ...)

Gokujo, 2019-10-15:
#### Required Information
- DietPi version | 6.25.3
- Distro version | 9.11
- Kernel version | `Linux JAS-TT 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`
- SBC device | RPi 3 Model B+ (armv7l)
- Power supply used | 5.1V 2.5A
- SDcard used | SanDisk / Intenso

#### Additional Information (if applicable)
I use automatic install and config on first start up. Therefore I used version 6.21. So I had to rollback.
This is the dietpi.txt
```
# IMPORTANT:
# - This is intended for advanced users, unless you know what you are doing, do not edit this file. Please use the DietPi programs instead.
# - Modifications to /boot/dietpi.txt will not be preserved on reboot.
# - Please ensure you edit from the DietPi-RAMdisk location: /DietPi/dietpi.txt
# - Do not remove uncommented lines, as the items are scraped by DietPi programs, on demand.

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# DietPi-Automation settings, applied on first boot of DietPi only, ONCE!
#------------------------------------------------------------------------------------------------------
##### Language/Regional Options #####
# Locale: eg: "en_GB.UTF-8" / "de_DE.UTF-8" | One entry and UTF-8 ONLY!
AUTO_SETUP_LOCALE=de_DE.UTF-8

# Keyboard Layout eg: "gb" / "us" / "de" / "fr"
AUTO_SETUP_KEYBOARD_LAYOUT=de

# Timezone eg: "Europe/London" / "America/New_York" | Full list (TZ*): https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
AUTO_SETUP_TIMEZONE=Europe/Berlin

##### Networking Options #####
# Enable Ethernet or WiFi adapter: 1=enable | 0=disable
# - If both Ethernet and WiFi are enabled, WiFi will take priority and Ethernet will be disabled. 
# - If using WiFi, please edit dietpi-wifi.txt to pre-enter credentials.
AUTO_SETUP_NET_ETHERNET_ENABLED=1
AUTO_SETUP_NET_WIFI_ENABLED=0

# Force ethernet speeds: 0=automatic speed | 10 = 10Mbit | 100 = 100Mbit etc
# - This is mainly aimed at Pine A64 which may have an HW issue that causes unstable 1Gbit link.
AUTO_SETUP_NET_ETH_FORCE_SPEED=0

# Enter your Static Network details below, if applicable.
AUTO_SETUP_NET_USESTATIC=0
AUTO_SETUP_NET_STATIC_IP=192.168.0.100
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.0.1
AUTO_SETUP_NET_STATIC_DNS=8.8.8.8

# Hostname
AUTO_SETUP_NET_HOSTNAME=JAS-TT

##### Misc Options #####
# Swapfile size to generate: 0=disable | 1=auto (2GB-RAM = size) | 2+=manual (MB)
AUTO_SETUP_SWAPFILE_SIZE=1
# Swapfile location
AUTO_SETUP_SWAPFILE_LOCATION=/var/swap

# Disable HDMI output (and GPU/VPU where supported) for supported devices: RPi, Odroid C1, Odroid C2
AUTO_SETUP_HEADLESS=0

# Unmask (enable) systemd-logind service, which is masked by default on DietPi
AUTO_UNMASK_LOGIND=0

# Custom Script (pre-networking and pre-DietPi install)
# - Allows you to automatically execute a custom script before network is up on first boot.
# - Copy your script to /boot/Automation_Custom_PreScript.sh and it will be executed automatically.
# - Executed script log: /var/tmp/dietpi/logs/dietpi-automation_custom_prescript.log

[... 182 lines cut ...]
# Chromium options
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=720
SOFTWARE_CHROMIUM_AUTOSTART_URL=https://dietpi.com

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Dev settings
#------------------------------------------------------------------------------------------------------
DEV_GITBRANCH=master
DEV_GITOWNER=MichaIng

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Settings, automatically added by dietpi-update
#------------------------------------------------------------------------------------------------------
```

#### Issues
- After installation I have to connect hardware keyboard to RPi to enter to next screen because of agreement screen (I have to agree with Putty and reboot then)
- All icons on desktop (Mate) disappeared (I tried to create new one but nothing happend)
- After reboot device stucks on i2c /dev entries... (reboot doesn't help) or it timed out to connect to TT0 devices (it takes to long to start) and sometimes on Apache server
- I noticed that menu with hardware clock disappered
- when I activate wifi in autoinstall (set it to 1) it can't connect to network
- during installation config file ignores CONFIG_AUTO_DIETPI_UPDATES, like I said I used older version (6.21.1) and it did an autoupdate which caused fails during install (but issue with desktop dissapear)

MichaIng, 2019-10-15:
@Gokujo 
Many thanks for your report.

>  Therefore I used version 6.21. So I had to rollback.

Why you needed to use v6.21? Please always use the current most current image, which is currently v6.25.3. If anything goes wrong with this, we'll going to investigate, but I don't gonna check first boot issues with older versions if there is a newer image available, please understand that.

Gokujo, 2019-10-15:
> Why you needed to use v6.21? 

because of issues. I had configs for that version and an image

> Note that we change those settings by times

I noticed that. it were a lot of config text so I didn't copied old config here. the new one was made extra for the actual version

MichaIng, 2019-10-15:
@Gokujo 
> because of issues. I had configs for that version and an image

Please report the issues you faced with the current version then, that is more productive then investing time in outdated versions that are not relevant/present on current version anymore 😉.

Gokujo, 2019-10-15:
> Please report the issues you faced with the current version then, that is more productive then investing time in outdated versions that are not relevant/present on current version anymore 

- After installation I have to connect hardware keyboard to RPi to enter to next screen because of agreement screen (I have to agree with Putty and reboot then)
- All icons on desktop (Mate) disappeared (I tried to create new one but nothing happend)
- After reboot device stucks on i2c /dev entries... (reboot doesn't help) or it timed out to connect to TT0(tty) devices (it takes to long to start) and sometimes on Apache server
- I noticed that menu with hardware clock disappered
- when I activate wifi in autoinstall (set it to 1) it can't connect to network
- during installation config file ignores CONFIG_AUTO_DIETPI_UPDATES, like I said I used older version (6.21.1) and it did an autoupdate which caused fails during install (but issue with desktop dissapear)
- keyboard settings don't apply

MichaIng, 2019-10-15:
@Gokujo 
> After installation I have to connect hardware keyboard to RPi to enter to next screen because of agreement screen (I have to agree with Putty and reboot then)

Jep this is expected. When doing automated first run installs, the license agreement is skipped and delayed until the first login. I see now that you enabled desktop auto login and of course it is not 100% automatic then on very first login because of agreement. However a desktop login most likely is for interactive use anyway? Or did you setup some automation that starts something from the desktop on login so that you don't need a keyboard anymore?

> All icons on desktop (Mate) disappeared (I tried to create new one but nothing happend)

Do you login as root or with another user? Will try to replicate, however in case of custom user it is expected, however not great indeed. Perhaps we can automatically create desktop items for non-root users that are chosen for auto login.

> After reboot device stucks on i2c /dev entries... (reboot doesn't help) or it timed out to connect to TT0(tty) devices (it takes to long to start) and sometimes on Apache server

Please paste either some picture of the screen, or if boot finally succeeds, paste the related syslog entries: `journalctl`

> I noticed that menu with hardware clock disappered

Jep that is hidden now if the device does not even have an RTC connector: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-config#L1310
The RPi simply requires `fake-hwclock` to not have 1970s system clock on boot, so no need to allow removing it from our menus. If you somehow manage to attach one via GPIO interfaces or such, then `hwclock` command must succeed for the system to use it, and then you will also see the menu entry again in `dietpi-config`.

> when I activate wifi in autoinstall (set it to 1) it can't connect to network

Did you verify that `dietpi-wifi.txt` contains all required credentials? Does it work well if you set it up after first boot via `dietpi-config`?

> during installation config file ignores CONFIG_AUTO_DIETPI_UPDATES, like I said I used older version (6.21.1) and it did an autoupdate which caused fails during install (but issue with desktop dissapear)

Note that the current image is as well the most current stable version. So there is no update available, hence non applied on first boot 😉. As well note that `CONFIG_AUTO_DIETPI_UPDATES` is not required to have an update on first boot. This is always done. This setting is to automatically apply updates, if available, during daily update check via cron job. However I am no fans of this setting, actually planned to remove it, since our updates by times contain interactive components that I simply want users to recognise.

> keyboard settings don't apply

Jep known [Debian bug](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=818065), workaround implemented with v6.26. Please run `setupcon --save` after changing keyboard layout. However I will as well push v6.26 release today.

Gokujo, 2019-10-15:
> Or did you setup some automation that starts something from the desktop on login so that you don't need a keyboard anymore?

it's the point, I just insert a burned imaged and let it install. then with help of git I install my app

> Do you login as root or with another user?

as root

> Did you verify that dietpi-wifi.txt contains all required credentials? Does it work well if you set it up after first boot via dietpi-config?

yes, this file have data for connection. after install I have to activate wifi module

> actually planned to remove it

if you do this how can I get the information what changed in dietpi.txt, config.txt and dietpi-wifi.txt?

Gokujo, 2019-10-15:
journalctl you can find [here](https://pastebin.com/qg4unFef)
https://pastebin.com/qg4unFef

MichaIng, 2019-10-15:
@Gokujo 
> Okt 15 16:09:24 JAS-TT3 systemd[1]: Starting Show Plymouth Boot Screen...

But this is from an old image, isn't it? Because the new image does not come with plymouth pre-installed as it is known to cause strange boot hangs and "wait for..." messages for some reason. Or did you manually install and enable plymouth splash screen?

> Okt 15 16:09:27 JAS-TT3 systemd[1]: Started ifup for eth0.
Okt 15 16:09:27 JAS-TT3 systemd[1]: Starting Raise network interfaces...
Okt 15 16:09:27 JAS-TT3 systemd[1]: Started ifup for wlan0.

Both, Ethernet and WiFi are enabled. Without having a manual created routing table (or having WiFi Hotspot installed, which creates a routing table as well), this is known to cause issues since the system does not know to which device to send network requests. If you want to use WiFi for www access, please disable Ethernet. Before first boot, this would be:
```
AUTO_SETUP_NET_ETHERNET_ENABLED=0
AUTO_SETUP_NET_WIFI_ENABLED=1
```
However WiFi connection seems to establish successfully:
> Okt 15 16:09:33 JAS-TT3 wpa_supplicant[658]: wlan0: CTRL-EVENT-CONNECTED - Connection to b4:fb:e4:21:30:9d completed [id=0 id_str=]

I see you installed "ntp" for network time sync. Did you change time sync mode in dietpi-config to "Custom"? Otherwise it will wait for 2 minutes (changed to 1 minute with v6.26) for systemd-timesyncd to complete, which fails due to ntpd doing the job. And please only install ntp if your are absolutely sure that you require it. In 95% of all cases it is simply an overkill of time precision and therefore unnecessary overhead for the system.
And ntpd reports:
> Okt 15 16:09:52 JAS-TT3 ntpd[730]: receive: Unexpected origin timestamp 0xe15051a6.3e156476 does not match aorg 0000000000.00000000 from server@176.96.138.245 xmt 0xe15051b0.3c635ea2

So please check your ntpd config as well.

XRDP is btw known to have issues on Stretch. Another reason to use the new image, which is based on Raspbian Buster. However with v6.26 we work around the known XRDP issues by installing the packages from backports: `apt install xrdp/stretch-backports`
But generally I recommend to use VNC which is MUCH faster and there are simple portable clients available for Windows. We also did some fixes/enhancements with v6.26 for general VNC connection.

And about your I2C issues, I bet it is due to serial consoles being enabled. Please go to dietpi-config > Advanced Options > Serial/UART and disable all serial consoles and the ttyS0 device as well.
__________
Btw I verified the missing desktop icons on MATE desktop, when you login as root. When logging in via different user via lightdm, it works. This is a known security "feature", since on MATE, the caja window manager is responsible to show desktop icons and has some security vulnerabilities why it should not do that with root permissions. On other desktops, the icons and desktop context menu and such are separate from window manager, if I understood correctly, hence it is no issue there: https://github.com/mate-desktop/caja/issues/871

I successfully enabled desktop icons+feature by this:
```
mkdir -p /root/.config/autostart
cat << _EOF_ > /root/.config/autostart/caja.desktop
[Desktop Entry]
Type=Application
Exec=caja --force-desktop
Hidden=false
X-MATE-Autostart-enabled=true
Name=root-desktop
_EOF_
```
- Logout + in again in case.
- This root-only autostart entry forces caja desktop.

However I will not implement this for now due to the security aspects. We should keep this as known limitation with MATE and instead allow easier setup and enhanced default experience for non-root user login on desktop. Btw `AUTO_SETUP_AUTOSTART_TARGET_INDEX=2` + `AUTO_SETUP_AUTOSTART_LOGIN_USER=dietpi` works well already, but desktop icons need to be added manually since the dietpi-software install only adds them for root user. As well in case of DietPi scripts, those need to be edited to be run with `sudo`.

MichaIng, 2019-10-15:
I opened a separate issue about missing MATE desktop icons + context menu: #3160

Gokujo, 2019-10-16:
> But this is from an old image, isn't it?

yes it was an update from old version to new one, because the new one had issues with mate desktop

> However WiFi connection seems to establish successfully

yes, because I activated module in dietpi-config. I will disable lan.

> I see you installed "ntp" for network time sync.

yes I had issues with wrong time. RPi ntp was modified to local server. [ntp.conf](https://pastebin.com/2HtK7eQd). I tried to change in dietpi-config but it didn't helped

> XRDP is btw known to have issues on Stretch.

I will remove it from auto install. I had problems to connect through VNC without this.

> AUTO_SETUP_AUTOSTART_LOGIN_USER=dietpi

If its the case, so I will change to user

> If so, we would at least need to update all images so that users can use their current dietpi.txt on every image that can be downloaded on dietpi.com.

It would be great

MichaIng, 2019-10-16:
@Gokujo 
> yes it was an update from old version to new one, because the new one had issues with mate desktop

Yeah as said, old images contain reasonably more issues then newer ones. So please always use the most current image and report issues you face with them.
The MATE issue btw will happen on an old image as well since this is by design wanted, as mentioned above. Not sure if/why this was working in the past, but it must not due to security reasons.

> yes I had issues with wrong time. RPi ntp was modified to local server. ntp.conf. I tried to change in dietpi-config but it didn't helped

You setup multiple servers now with multiple directives. If you want to use a single local server, try to remove all the `pool` directives below. If this fails, but you can `ping 192.168.0.10`, then I guess this host does not serve NTP protocol?

As said you can use ntpd, but to not conflict with DietPi trying to check systemd-timesyncd status, you should set time sync state to "Custom" via `dietpi-config`. However, systemd-timesyncd is so simple, if it fails, ntpd most likely fails for the same reason, so you should then check the host (e.g. try with another machine) or network.

> I will remove it from auto install. I had problems to connect through VNC without this.

Please try it after updating to v6.26 or using a fresh image (which is updated on first boot), since we made some rework of the VNC install code. This mostly affected TigerVNC, but RealVNC at least partly as well.

XRDP should work very well now, as well on Stretch images, however slow compared to VNC 😉.

Gokujo, 2019-10-23:
I'm sorry. I am working on my project for exam. Only after 11.11. I can test new image. I didn't had time even to inform.

MichaIng, 2019-10-23:
@Gokujo 
Exam should have priority. Wish you all the best for that! 🙂

Gokujo, 2019-12-02:
[Journal](https://pastebin.com/Uev4rKef)
So, hey. Finally I got some time to test rp. I downloaded image from the website (it was the same as before ;)) and started an install with configuration in first post. 
It worked well! Fast install, fast startup! But, I forgot to change the login user for automatic desktop.

MichaIng, 2019-12-04:
@Gokujo 
Okay great. Quick side note, run: `systemctl disable -now serial-getty@ttyS0`
Since the mini UART seems to be disabled or used for other purpose, this TTY is doomed to fail and just spams the boot log :wink:.

I adjusted RPi firstboot behaviour a bid according this to have obsolete serial TTYs disabled automatically. New images are done after v6.27 release.

Gokujo, 2019-12-05:
> @Gokujo
> Okay great. Quick side note, run: `systemctl disable -now serial-getty@ttyS0`
> Since the mini UART seems to be disabled or used for other purpose, this TTY is doomed to fail and just spams the boot log 😉.
> 
> I adjusted RPi firstboot behaviour a bid according this to have obsolete serial TTYs disabled automatically. New images are done after v6.27 release.

Hi,
I get this: `Failed to parse lines 'ow'`

MichaIng, 2019-12-05:
@Gokujo 
Ah sorry, a missing dash, it should have been:
```
systemctl disable --now serial-getty@ttyS0
```

Gokujo, 2019-12-05:
Thanks,

I close this issue! It works fine now!

closed 2019-12-05 by Gokujo