MichaIng/DietPi #3164: DietPi-Software | Kodi: Install requires polkit-1

glogiotatidis, 2019-10-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.25.3
- Distro version | 10.1
- Kernel version | `uname -a`
- SBC device | RPi 3 A
- Power supply used | 5V 2A
- SDcard used | Sandisk

#### Additional Information (if applicable)
- Software title | Kodi
- Was the software title installed freshly or updated/migrated? Fresh
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Select from `dietpi-software` to install Kodi
2. Kodi install fails

#### Expected behaviour
- Kodi should install

#### Actual behaviour

- `dpkg` configure cannot complete install b/c `/etc/polkit-1/` directories, references by `kodi.postinst` do not exist

#### Extra details

- Installing `polkit-1` debian package fixes the issue

glogiotatidis, 2019-10-16:
And just upgraded to 6.26.3 to see `Pre-patch 17` that fixes the issue. Thanks for the quick fix :smile:

closed 2019-10-16 by glogiotatidis

MichaIng, 2019-10-16:
@glogiotatidis 
Jep, it is nice that Kodi provides policy kit info, but it is kinda bug that it requires this path to exist even that the package does not list `policykit1` as dependency. And it shouldn't 😉.