MichaIng/DietPi #3165: Update from v6.25.3 to v6.26.3 on Odroid C2 fails

BerndKohl, 2019-10-16:
#### Details:
- Date           | Wed 16 Oct 20:45:25 CEST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (index=12)
- Kernel version | #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Ign:1 https://downloads.plex.tv/repo/deb public InRelease
Hit:2 https://downloads.plex.tv/repo/deb public Release
Ign:3 https://cdn-aws.deb.debian.org/debian stretch InRelease
Get:4 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease [91.0 kB]
Get:5 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease [94.3 kB]
Get:6 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease [91.8 kB]
Hit:7 https://cdn-aws.deb.debian.org/debian stretch Release
Get:9 https://cdn-aws.deb.debian.org/debian-security stretch/updates/main arm64 Packages [482 kB]
Ign:11 http://fuzon.co.uk/meveric all InRelease
Ign:12 http://fuzon.co.uk/meveric stretch InRelease
Err:13 http://fuzon.co.uk/meveric all Release
  Unable to connect to fuzon.co.uk:http:
Err:14 http://fuzon.co.uk/meveric stretch Release
  Unable to connect to fuzon.co.uk:http:
Reading package lists...
E: The repository 'http://fuzon.co.uk/meveric all Release' does no longer have a Release file.
E: The repository 'http://fuzon.co.uk/meveric stretch Release' does no longer have a Release file.
```

keeno1982, 2019-10-17:
Looks like fuzon.co.uk down again

MichaIng, 2019-10-17:
@BerndKohl @keeno1982 
Many thanks for your report. Indeed fuzon.co.uk is down, I just informed @Fourdee .

Meanwhile you can switch over to the other mirror:
```
sed -i 's|http://fuzon.co.uk/https://oph.mdrjr.net/g' /etc/apt/sources.list.d/*.list
G_AGUP
```

BerndKohl, 2019-10-17:
But although the dietpi-update process did finish with an error dietpi now seems to believe it is already running v.6.26.3...
How can I make dietpi believe the update did not succeed to be able to restart it?

MichaIng, 2019-10-17:
@BerndKohl 
Ah yeah thats true, there was an issue in v6.25 that prevented the script being exited correctly: https://github.com/MichaIng/DietPi/commit/1e52d1348a678b422959f3cb12db30a6065f430b

However in this case it is actually fine. The update does not depend on the updated Odroid APT repo, hence no need to redo the update. If you want to keep up with Odroid repo updates, kernel, firmware etc:
```
apt update
apt full-upgrade
```

If you however really want to rerun the dietpi-update:
```
dietpi-update -1
```
- This internally reduces the subversion string by 1, hence the last update is repeated.

cloneman, 2020-01-16:
Server appears to be down.

Muggedadscher, 2020-01-16:
The server is down again.

Joulinar, 2020-01-16:
@cloneman @Muggedadscher 
Many thanks for your report. Pls can you check if the suggested workaround is still working 

`sed -i 's|http://fuzon.co.uk/https://oph.mdrjr.net/g' /etc/apt/sources.list.d/*.list
G_AGUP`

MichaIng, 2020-01-16:
Forgotten invoice. I just paid it, server is back online.

fuzon.co.uk is as very old server hidden behind an old VPS interface, hence outside our regular view. We'll migrate the APT server into dietpi.com, once MVS allows to raise drive space separately. Then this regular server down every 3 months has an end 😅.

closed 2020-02-27 by MichaIng